jsk-ros-pkg/jsk_robot #1238: [app_manager] リモートで実験を使っているときのロックに使えないか．

k-okada, 2020-08-03:
リモートからロボットに入って実験したいんだけど，そのときのmutexの取り方があるといいとおもっています．
で，思いついたのはapp_manager でダミーのようなアプリを立ち上げて，その間に作業して，終わったら，アプリを停止する．みたいなことが出来たりしないかな．

- app_manager が立ち上がったら，毎朝の10時のプログラムは停止されなかったりするかな？それともcronだから関係ない？
- app_manager でアプリが呼ばれたらtwitterに流すようにすると，それを見ていると誰かが動かしたことが共有できないかな．

Cc: @ishiguroJSK @knorth55 @708yamaguchi

knorth55, 2020-08-03:
app_managerはapp_schedulerというので時間起動しています。
https://github.com/knorth55/app_manager_utils

app_managerは今はあとからタスクが来た場合には上書きされて、あとのものが実行されます。
app_managerで何かしらが行ったらtweetするのはapp_managerのプラグインで実装できます。
https://github.com/knorth55/app_manager_utils

knorth55, 2020-10-13:
- [x] app_managerでappをoverwriteしないようにする (https://github.com/PR2/app_manager/pull/26)
- [ ] rwt_app_chooserから`/action/parameters`を設定できるようにする
- [ ] appがない状態でロボットが動いた場合にはwarningを出す
- [ ] app_manager_utilsのような関数をsmachにも用意する