jsk-ros-pkg/jsk_robot #1259: fetchの/move_baseによる移動がカクつく

mqcmd196, 2020-10-16:
状況:
====
fetchが廊下を移動するときに，コストマップを踏んでいないにもかかわらず，速度が0.1m/sぐらいでカクつきながら移動していた．

解決策:
====
Rviz上でのfetchのモデル，点群，コストマップなどを非表示にすると同じ条件下で，max_vel_x(=0.5 m/s)に近い値で移動するようになった．

原因:
====
1. cpuがhigh load
1. cmd_velのhz小によりodomから調べた速度が0になることがある 
1. このときローカルプランナはx方向+0.1m/sくらいの指令をだす
1. このループから抜け出せない

確認方法
====
Rvizでのモデル，点群，コストマップ表示時と非表示時で様々な値を比較しました
- htopでの直近1分のcpu load値が表示時で35，非表示時で16
- `rostopic hz /cmd_vel -w 10` が表示時で3，非表示時で7
- 表示時に/move_baseが以下のWARNINGを出力していた
```
[ WARN] [1602853054.981716233] [/move_base:ros.move_base]: Control loop missed its desired rate of 10.0000Hz... the loop actually took 0.2950 seconds
``` 

Cc:@708yamaguchi @knorth55

knorth55, 2020-10-17:
`vel_type_mux`で`/navigation/cmd_vel`のHzが低い時にすぐに`/teleop/cmd_vel`に切り替わってしまう問題については， #1260 で解決しました．
CPUロードが大きく`move_base`のHzがでていない問題については，マザーボードごとアップグレードするしかない気がします．
外部PCを設置できる拡張性がFetchにはあまりないので・・・

mqcmd196, 2020-10-19:
@knorth55 
ありがとうございます！

mqcmd196, 2021-01-04:
再びこの問題が発生したらreopenします．

closed 2021-01-04 by mqcmd196