jsk-ros-pkg/jsk_robot #1273: fetch1075のps3joyコントローラによる操作が不安定

mqcmd196, 2020-11-13:
症状:
/joy topic が来ないときがある．/var/log/syslogを見てもbluetoothが接続してすぐにtimeoutしてしまう

解決法?:
fetchとjoyをusbで接続した状態で
```bash
sudo sixpair
```
抜いて
```bash
sudo sixad --stop
sudo sixad -s  # sudo sixad --start でも可
```

Cc: @708yamaguchi @sktometometo

708yamaguchi, 2020-11-13:
ちなみに、PS3コントローラのペアリングをした際の
```
tail -f /var/log/syslog
```
の結果は以下のようなかんじです。

```
Nov 12 20:28:54 fetch1075 dhclient: bound to 133.11.216.145 -- renewal in 256 seconds.
Nov 12 20:28:54 fetch1075 NetworkManager[15406]: <info> (wlan0): DHCPv4 state changed renew -> renew
Nov 12 20:28:54 fetch1075 NetworkManager[15406]: <info>   address 133.11.216.145
Nov 12 20:28:54 fetch1075 NetworkManager[15406]: <info>   prefix 24 (255.255.255.0)
Nov 12 20:28:54 fetch1075 NetworkManager[15406]: <info>   gateway 133.11.216.254
Nov 12 20:28:54 fetch1075 NetworkManager[15406]: <info>   nameserver '133.11.216.240'
Nov 12 20:28:54 fetch1075 NetworkManager[15406]: <info>   nameserver '8.8.8.8'
Nov 12 20:28:54 fetch1075 NetworkManager[15406]: <info>   domain name 'jsk.imi.i.u-tokyo.ac.jp'
Nov 12 20:28:54 fetch1075 dbus[698]: [system] Activating service name='org.freedesktop.nm_dispatcher' (using servicehelper)
Nov 12 20:28:54 fetch1075 dbus[698]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Nov 12 20:29:58 fetch1075 sixad-sixaxis[1504]: started
Nov 12 20:29:58 fetch1075 kernel: [72416.926553] input: PLAYSTATION(R)3 Controller (AC:7A:4D:28:FC:DA) as /devices/virtual/input/input12
Nov 12 20:29:59 fetch1075 sixad-sixaxis[1504]: Connected 'PLAYSTATION(R)3 Controller (AC:7A:4D:28:FC:DA)' [Battery 05]
Nov 12 20:30:00 fetch1075 sixad-sixaxis[1504]: Controller connection timed out, retrying.
Nov 12 20:30:00 fetch1075 wpa_supplicant[1123]: wlan0: CTRL-EVENT-SCAN-STARTED 
Nov 12 20:30:05 fetch1075 sixad-sixaxis[1504]: Controller connection timed out, retrying.
Nov 12 20:30:41 fetch1075 sixad-sixaxis[1504]: message repeated 7 times: [ Controller connection timed out, retrying.]
Nov 12 20:30:53 fetch1075 sixad-sixaxis[1504]: Failed to read from device, exiting.
Nov 12 20:30:53 fetch1075 sixad-sixaxis[1504]: Force disconnect of "AC:7A:4D:28:FC:DA"
```

708yamaguchi, 2020-11-16:
@k-okada 先生にデバッグしていただきました．ありがとうございました．

結果的には，bluetoothドングルを追加で付けてコントローラをペアリングすることで，（今のところ数時間くらいは）良い感じに/joyトピックが見えています．
bluetoothドングルをつけている限り，リブートしても良い状態は続いているように見えます．

## ペアリング手順
1. 最初に，fetch1075に備え付けのbluetoothデバイス（`hci0: BD Address: A0:A8:CD:91:60:26`）しか見えていないことを確認する．
```
$ hciconfig dev
hci0:	Type: BR/EDR  Bus: USB
	BD Address: A0:A8:CD:91:60:26  ACL MTU: 1021:5  SCO MTU: 96:5
	UP RUNNING PSCAN 
	RX bytes:5853052 acl:100980 sco:0 events:319 errors:0
	TX bytes:9328 acl:226 sco:0 commands:63 errors:0
```
2. bluetoothドングル（`hci1: BD Address: 00:1B:DC:0D:D0:AD`）を指す．
3. hci1（bluetoothドングル）が`DOWN`，hci0（備え付けbluetoothデバイス）が`UP RUNNING `となっていることを確認する．
```
$ hciconfig  -a
hci1:	Type: BR/EDR  Bus: USB
	BD Address: 00:1B:DC:0D:D0:AD  ACL MTU: 310:10  SCO MTU: 64:8
	DOWN 
	RX bytes:564 acl:0 sco:0 events:29 errors:0
	TX bytes:358 acl:0 sco:0 commands:29 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 

hci0:	Type: BR/EDR  Bus: USB
	BD Address: A0:A8:CD:91:60:26  ACL MTU: 1021:5  SCO MTU: 96:5
	UP RUNNING PSCAN 
	RX bytes:5872211 acl:101296 sco:0 events:328 errors:0
	TX bytes:9355 acl:226 sco:0 commands:72 errors:0
	Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'ubuntu-0'
	Class: 0x6c0100
	Service Classes: Rendering, Capturing, Audio, Telephony
	Device Class: Computer, Uncategorized
	HCI Version: 4.0 (0x6)  Revision: 0x500
	LMP Version: 4.0 (0x6)  Subversion: 0x500
	Manufacturer: Intel Corp. (2)
```
4. hci1をアクティブにし，hci0を非アクティブにする．
```
$ hciconfig hci1 up
$ hciconfig hci0 down
```
5. ここで，hci1（bluetoothドングル）が`UP RUNNING `，hci0（備え付けbluetoothデバイス）が`DOWN`となっていることを確認する．
```
$ hciconfig
hci1:	Type: BR/EDR  Bus: USB
	BD Address: 00:1B:DC:0D:D0:AD  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:1128 acl:0 sco:0 events:58 errors:0
	TX bytes:716 acl:0 sco:0 commands:58 errors:0

hci0:	Type: BR/EDR  Bus: USB
	BD Address: A0:A8:CD:91:60:26  ACL MTU: 1021:5  SCO MTU: 96:5
	DOWN 
	RX bytes:5872488 acl:101296 sco:0 events:331 errors:0
	TX bytes:9364 acl:226 sco:0 commands:75 errors:0
```
6. コントローラとbluetoothドングルをペアリングする．上の段と下の段が異なっているので，まだダメ．
``` 
$ rosrun ps3joy sixpair
Current Bluetooth master: a0:a8:cd:91:60:26
[... 162 lines cut ...]
	BD Address: 00:1B:DC:0D:D0:AD  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:1128 acl:0 sco:0 events:58 errors:0
	TX bytes:716 acl:0 sco:0 commands:58 errors:0

hci0:	Type: BR/EDR  Bus: USB
	BD Address: A0:A8:CD:91:60:26  ACL MTU: 1021:5  SCO MTU: 96:5
	DOWN 
	RX bytes:5872488 acl:101296 sco:0 events:331 errors:0
	TX bytes:9364 acl:226 sco:0 commands:75 errors:0

root@fetch1075:/home/fetch/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/scripts# rosrun ps3joy 
ps3joy_node.py  ps3joy.py       ps3joysim.py    sixpair         
root@fetch1075:/home/fetch/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/scripts# rosrun ps3joy 
ps3joy_node.py  ps3joy.py       ps3joysim.py    sixpair         
root@fetch1075:/home/fetch/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/scripts# rosrun ps3joy sixpair 
Current Bluetooth master: a0:a8:cd:91:60:26
Setting master bd_addr to 00:1b:dc:0d:d0:ad
root@fetch1075:/home/fetch/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/scripts# hciconfig hci1 reset
root@fetch1075:/home/fetch/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/scripts# hciconfig hci0 reset
root@fetch1075:/home/fetch/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/scripts# rosrun ps3joy sixpair 
Current Bluetooth master: 00:1b:dc:0d:d0:ad
Setting master bd_addr to 00:1b:dc:0d:d0:ad
root@fetch1075:/home/fetch/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/scripts# 
```

mqcmd196, 2020-11-18:
ドングル接続による方法でもう一度不安定になったらこのissueをreopenします．

closed 2020-11-18 by mqcmd196

pazeshun, 2020-11-23:
@mqcmd196 @708yamaguchi 
本日登校したら、fetch1075のコントローラが定期的に振動する状態になっていました。周りの人にも話を聞くと、少なくとも午前10時にはこの状態だったようです。
一応報告しておきます。現在部屋にいる人に実害はないです。

708yamaguchi, 2020-11-23:
@pazeshun 
報告ありがとうございます。ご迷惑をおかけしております。
僕もその現象には気づいていたのですが、原因が分からず放置していました。
たしか、fetch1075とペアリングしたら定期的に振動する状態は解除されたような気がします。

sktometometo, 2021-02-28:
本日, fetch1075において, connection timeout が頻発する現象を確認しました.
hciconfig を見たところ, 組み込みのbluetoothデバイスが使用されている模様でした.

```bash
fetch@fetch1075:~$ sudo hciconfig
[sudo] password for fetch: 
hci1:   Type: BR/EDR  Bus: USB
        BD Address: 00:1B:DC:0D:D0:AD  ACL MTU: 310:10  SCO MTU: 64:8
        DOWN 
        RX bytes:564 acl:0 sco:0 events:29 errors:0
        TX bytes:358 acl:0 sco:0 commands:29 errors:0

hci0:   Type: BR/EDR  Bus: USB
        BD Address: A0:A8:CD:91:60:26  ACL MTU: 1021:5  SCO MTU: 96:5
        UP RUNNING PSCAN 
        RX bytes:2456214 acl:42380 sco:0 events:313 errors:0
        TX bytes:8568 acl:202 sco:0 commands:89 errors:0
```

上記手順にしたがって, A0:A8:CD:91:60:26 を downし, 00:1B:DC:0D:D0:AD を upし, 最ペアリングしたところ, connection timeout が起こらなくなりました,

ただ, その後再起動すると, connection timeout にはならないものの,hci0が upしておりました.
デフォルトで A0:A8:CD:91:60:26 が DOWNするようにできればしたほうがいいかもしれません.