jsk-ros-pkg/jsk_robot #1304: 実機のグリッパの状態を確認する方法はありますか？

k-okada, 2020-12-17:
@tkmtnt7000

```
質問なのですが、
(send *fetch* :rarm :gripper :angle-vector)
とするとIRT viewerでのFetchのグリッパの状態が確認できるように
(send *ri* ......)
で実機のグリッパの状態を確認する方法はありますか？

:gripperがメソッドにあることは確認できたのですが、いろいろと試してみてもわからなくて...
```

Cc: @yuli-disney, @softyanija, @tenrobo18, @tkmtnt7000, @ketaro-m

k-okada, 2020-12-17:
https://github.com/jsk-ros-pkg/jsk_robot/blob/46614fb2ec61f46c51e7cf602dd48b35cee2e981/jsk_fetch_robot/fetcheus/fetch-interface.l#L197-L202
のコードは実機用で
```
(send *ri* :gripper :position)
```
は中は
```
(send *ri* :state :gripper-position)
```
が呼ばれてグリッパ間の距離が返ってきます．ややこしいのはこれが[m]で返ってくることです．

あるいはKinematics Simlator でしょうか？ですと，
```
11.irteusgl$ send *ri* :start-grasp
t
12.irteusgl$ send *ri* :robot :rarm :gripper :angle-vector
#f(0.0 0.0)
13.irteusgl$ send *ri* :stop-grasp
t
14.irteusgl$ send *ri* :robot :rarm :gripper :angle-vector
#f(50.0 50.0)
15.irteusgl$ 
```
となります．
```
(* 0.001 2 (vmean (send *ri* :robot :rarm :gripper :angle-vector)))
```
とするとグリッパ間の距離[m]となります．

現状の:gripperメソッドはkinematics simlator 対応機能が無いので，
```
(defmethod fetch-interface
  (:gripper (key)
   "get information about gripper                                                                         
Arguments:                                                                                                
 - key (:position :velocity :effort)                                                                      
Example: (send self :gripper :position) => 0.00"
   (if (send self :simulation-modep)
       (let ((pos (* 0.001 2 (vmean (send robot :rarm :gripper :angle-vector)))))
         (case key
               (:position pos)
               (:velocity 0)
               (:effort (+ (* -110 pos) 1)))) ;; effot is -10 on open, 1.2 for close                      
     ;; for real robot                                                                                    
     (send-super :state (intern (format nil "GRIPPER-~A" (string key)) *keyword-package*))))
  )

```
のコードを自分のデモプログラムで`fetchrobot-interface.l`をロードした後に入れておくと実機，kinematics simulator 共に
```
send *ri* :gripper :rarm :position
```
が使えるようになります．

tkmtnt7000, 2020-12-17:
ありがとうございます.
`(send *ri* :gripper :position)`
をKinematicsSimlator上でずっと試していたので、いつもnilが返ってきている状態でした.

`send *ri* :robot :rarm :gripper :angle-vector`
を使ってシミュレータ上で確認することができました.

PR #1305 (2020-12-24): support :gripper in KinematicsSimulator mode
closes #1304
commit a4b862cc (2020-12-24): support :gripper in KinematicsSimulator mode

closes #1304
closed 2020-12-24 by k-okada