jsk-ros-pkg/jsk_robot #1307: ローカルでも実機と同じように干渉回避するにはどのようにしたら良いのでしょうか．

k-okada, 2020-12-17:
@ketaro-m 
```
ドアやubereatsボックスを衝突回避しようとしてcollision objectを追加した状態で腕を動かしているのですが，ローカルで試した場合と実機で動かした場合で腕の動きが異なります．おそらくはローカルで動かした場合は干渉回避が正しく行われておらずcollision objectを追加しない状態と同じ動きをするのですが，実機の場合は正しく干渉回避をするので，別の動きになってしまうのだと思います．
ローカルでも実機と同じように干渉回避するにはどのようにしたら良いのでしょうか．

ちなみに，ローカルで試す場合は下のlaunchファイルを立ち上げてからcollision objectを追加しています．

$ roslaunch fetch_moveit_config demo.launch
```
@yuli-disney, @softyanija, @tenrobo18, @tkmtnt7000, @tkmtnt7000

https://github.com/jsk-ros-pkg/jsk_demos/issues/1309#issuecomment-746110082

k-okada, 2020-12-17:
１）一番確実なのはgazeboを実行してからmoveitを動かしてすすめることです．
それぞれ別のターミナルで以下を起ち上げてください．
```
$ roslaunch fetch_gazebo playground.launch 
$ roslaunch fetch_moveit_config move_group.launch 
$ roslaunch moveit_rviz.launch config:=true
```
これで基本は実機と同じ構成になります．あとはroseusを起ち上げて以下のプログラムを実行します．
```
(load "package://fetcheus/fetch-interface.l")
(if (not (boundp '*ri*))
    (fetch-init))

(setq *door-collision-model* (make-cube 10 1000 2000 :pos #f(600 0 1000)))
(objects (list *fetch* *door-collision-model*))
(when (boundp '*co*)
  (send *co* :add-object *door-collision-model* :frame-id "base_link" :object-id "door"))

;; reset-pose                                                                                                                
(send *fetch* :angle-vector #f(0 90 0 0 0 0 0 0 0 0))
(send *ri* :angle-vector (send *fetch* :angle-vector))
(send *ri* :wait-interpolation)

;; reach-pose                                                                                                                
(send *fetch* :rarm :inverse-kinematics (make-coords :pos #f(500 0 1200) :rpy (float-vector -pi/2 0 pi/2)))
(send *ri* :angle-vector (send *fetch* :angle-vector))
(send *ri* :wait-interpolation)
```
するとちゃんと障害物を避けている事がわかります．

[image]
[image]

２）gazeboは重くなるのでmoveitだけ使いたい場合ですが，こんどは`demo.launch`を起ち上げた後，`/joint_state_publisher`を止めます．
```
$ roslaunch fetch_moveit_config demo.launch
$ rosnode kill /joint_state_publisher
```
あとはroseusを起ち上げて上と同じプログラムを実行すると，
[image]

と壁をすり抜けてしまいます（困っているのはこの状況でしょうか？こういう風に何をどうして，どうなって，その何が困っているか，と書けるように成ると良いです）

で，その場合ですが`:angle-vector`の代わりに
```
(send *ri* :angle-vector-motion-plan (send *fetch* :angle-vector) :move-arm :rarm)
```
とすると
[image]
のようになりますが，これで期待通りでしょうか？

ちなみに実機でもmoveitが存在する場合でも同じコードを動かすためには
#1306 を使うと良いです．

ketaro-m, 2020-12-18:
ご丁寧にお返事ありがとうございます．今日ラボに行く予定なので，実機とともに試してみたいと思います．