jsk-ros-pkg/jsk_robot #1324: [fetcheus]  fetch1075で send *ri* angle-vectorしたときにservice call error が起こる

tenrobo18, 2021-03-01:
以下のようにfetch1075でshoulder-rを回転させようとしたところ、エラーを吐き、*fetch*は動いたが、実機は動きませんでした。
```lisp
[http://fetch1075:11311][133.11.216.175] temma@p51s:~/semi_ws/src/jsk_demos/jsk_2020_10_semi/euslisp$ roseus
configuring by "/opt/ros/melodic/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; extnum ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:2
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph ___time ___pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtstl irtwrl irtpointcloud eusbullet bullet irtcollision irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x560fe6156690[16374] --> 0x560fe65dc230[32748] top=3c74
irtgl irtglc irtviewer 
EusLisp 9.27( 1.2.2) for Linux64 created on ip-172-30-1-129(Sat Oct 17 05:56:37 UTC 2020)
roseus ;; loading roseus("1.7.4") on euslisp((9.27 ip-172-30-1-129 Sat Oct 17 05:56:37 UTC 2020  1.2.2))
eustf roseus_c_util 1.irteusgl$ load "temma.l"

;; extending gcstack 0x560fe65dc230[32738] --> 0x560fe8fbe3a0[65476] top=7fb3
Color name (:silver) not found
Color name (:silver) not found
Color name (:silver) not found
Color name (:silver) not found
transparent: body #<scene-model #X560fe90b2bf0 71a1  -1010.0 -37800.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :dodgerblue
transparent: body #<scene-model #X560fe910f210 71b  1010.0 -37800.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :dodgerblue
transparent: body #<scene-model #X560fe912f868 71c2  34329.0 -30750.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :dodgerblue
transparent: body #<scene-model #X560fe929c2c0 72a1  -1010.0 -17100.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :dodgerblue
transparent: body #<scene-model #X560fe9af66c0 72b1  1010.0 -17100.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :dodgerblue
transparent: body #<scene-model #X560fe9c1d5b0 72b2  1010.0 -10200.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :dodgerblue
transparent: body #<scene-model #X560fe93e7550 72d1  36349.0 -20550.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :dodgerblue
transparent: body #<body #X560ff09f6ab8 corridor1 (:prism (#f(0.0 0.0 0.0) #f(-1000.0 6.123234e-14 0.0) #f(-1000.0 14800.0 0.0) #f(1000.0 14800.0 0.0) #f(1000.0 -6.123234e-14 0.0) #f(0.0 0.0 0.0)) 1) 0.0 0.0 0.0 / 1.571 0.0 0.0> param 0.3 face-color :orange
transparent: body #<body #X560ff0af8318 corridor2 (:prism (#f(0.0 -6600.0 0.0) #f(-1000.0 -6600.0 0.0) #f(-1000.0 -20700.0 0.0) #f(1000.0 -20700.0 0.0) #f(1000.0 -6600.0 0.0) #f(0.0 -6600.0 0.0)) 1) 0.0 -6600.0 0.0 / 1.571 0.0 0.0> param 0.3 face-color :orange
transparent: body #<body #X560ff0aea380 corridor3 (:prism (#f(0.0 -27300.0 0.0) #f(-1000.0 -27300.0 0.0) #f(-1000.0 -42100.0 0.0) #f(1000.0 -42100.0 0.0) #f(1000.0 -27300.0 0.0) #f(0.0 -27300.0 0.0)) 1) 0.0 -27300.0 0.0 / 1.571 0.0 0.0> param 0.3 face-color :orange
transparent: body #<body #X560ff0ae10f8 corridor4 (:prism (#f(1010.0 -3300.0 0.0) #f(1010.0 -1950.0 0.0) #f(34329.0 -1950.0 0.0) #f(34329.0 -4650.0 0.0) #f(1010.0 -4650.0 0.0) #f(1010.0 -3300.0 0.0)) 1) 1010.0 -3300.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :orange
transparent: body #<body #X560ff0ad4f90 corridor5 (:prism (#f(35339.0 0.0 0.0) #f(34339.0 6.123234e-14 0.0) #f(34339.0 14800.0 0.0) #f(36339.0 14800.0 0.0) #f(36339.0 -6.123234e-14 0.0) #f(35339.0 0.0 0.0)) 1) 35339.0 0.0 0.0 / 1.571 0.0 0.0> param 0.3 face-color :orange
transparent: body #<body #X560ff0ac8900 corridor6 (:prism (#f(35339.0 -6600.0 0.0) #f(34339.0 -6600.0 0.0) #f(34339.0 -20700.0 0.0) #f(36339.0 -20700.0 0.0) #f(36339.0 -6600.0 0.0) #f(35339.0 -6600.0 0.0)) 1) 35339.0 -6600.0 0.0 / 1.571 0.0 0.0> param 0.3 face-color :orange
transparent: body #<body #X560ff0abc5a0 corridor7 (:prism (#f(35339.0 -27300.0 0.0) #f(34339.0 -27300.0 0.0) #f(34339.0 -42100.0 0.0) #f(36339.0 -42100.0 0.0) #f(36339.0 -27300.0 0.0) #f(35339.0 -27300.0 0.0)) 1) 35339.0 -27300.0 0.0 / 1.571 0.0 0.0> param 0.3 face-color :orange
transparent: body #<body #X560ff0aae608 corridor8 (:prism (#f(1010.0 -24110.0 0.0) #f(1010.0 -22760.0 0.0) #f(34329.0 -22760.0 0.0) #f(34329.0 -25460.0 0.0) #f(1010.0 -25460.0 0.0) #f(1010.0 -24110.0 0.0)) 1) 1010.0 -24110.0 0.0 / 0.0 0.0 0.0> param 0.3 face-color :orange

;; extending gcstack 0x560fe8fbe3a0[65466] --> 0x560ffd595af0[130932] top=ffb9
[ WARN] [1614586719.456894312]: ;; collision-object-publisher wait for "apply_planning_scene" service for 5 sec

;;2 face-to-face contact(s) found.
;;2 face-to-face contact(s) found.
;; (make-irtviewer) executed
Detect very large/small objects(3.96799), change perspective
t
2.irteusgl$  (send *fetch* :rarm :shoulder-r :joint-angle -60.0)
-60.0
3.irteusgl$  (send *ri* :angle-vector (send *fetch* :angle-vector))
[ERROR] [1614586786.123296211]: attempted to call service  /plan_kinematic_path, but failed 
;; motion plan error at nil
nil
```

環境は以下の通りです
```bash
[http://fetch1075:11311][133.11.216.175] temma@p51s:~/semi_ws/src/jsk_demos/jsk_2020_10_semi/euslisp$ wstool info
warning: redirecting to https://github.com/tenrobo18/jsk_demos/
warning: redirecting to https://github.com/tenrobo18/jsk_demos/
Warning: version 'jsk_2020_10_semi' not found for 'jsk_demos'
workspace: /home/temma/semi_ws/src

 Localname  S  SCM Version (Spec)         UID  (Spec)                 URI  (Spec) [http(s)://...]
 ---------  -  --- --------------         -----------                 ---------------------------
 jsk_robot  MV git master  (4714d299e1a3) 532cdb6130c9 (4714d299e1a3) github.com/jsk-ros-pkg/jsk_robot
 jsk_pr2eus    git master  (a228cc7d3bb5) a228cc7d3bb5                github.com/jsk-ros-pkg/jsk_pr2eus
 jsk_demos  MV git jsk_2020_10_semi       f5b6e7acfa9e ("jsk_2020_10) github.com/tenrobo18/jsk_demos
```

tenrobo18, 2021-03-01:
@sktometometo 
修正を確認しました
ありがとうございました

sktometometo, 2021-03-01:
この問題ですが, fetch体内で使用している fetch15ブランチにおいて, fetchの個体によらずbase_camera_mount(fetch15のベースについているカメラマウント)を collision object としてついかしているためでした.
https://github.com/knorth55/jsk_robot/pull/73 によりこの問題は起きなくなるはずです.

closed 2021-03-01 by sktometometo