jsk-ros-pkg/jsk_robot #1336: add camera interface for denso-interrface.l

k-okada, 2021-05-24:
``````
Cobottaのカメラですが、デフォルトのキャノンのカメラも動かせるようです
中身はsocket通信して撮像コマンドを送ると、画像が返ってくるだけのようですが、プロトコルがDenso謹製のプロトコル(bCap)になっていて、キャノンがDensoようにインターフェースを作ってあげたような経緯がありそうです。
```
https://github.com/DENSORobot/denso_robot_ros/issues/38

k-okada, 2021-05-24:
@ykawamura96 君の言うように
https://github.com/k-okada/jsk_robot/tree/add_cobotta/jsk_denso_robot/jsk_cobotta_startup
に画像を取って，publishするノードを書くのがいいか 

ROSでは，http://wiki.ros.org/polled_camera　というノードがあって，これに対応する形にするのが一つかな．

k-kimura, 2021-05-24:
@ykawamura96 にもききました．
キャノンのカメラ（N10-W02）でまずはサンプル（BcapClient）を試してみます．
作業としては来週くらいになりそうですが．．
卵デモでも見れるようにしたいとおもってます．

k-kimura, 2021-06-07:
先週，@ykawamura96 と作業してこちらのカメラを見れるようにしました．
Related: https://github.com/k-okada/jsk_robot/pull/17

トラブルの主な原因としては，カメラに振られてた固定IPがDHCP帯になってて別の機器に割り当てられてたことです（情報理工NW内でIPが重複してる状況）．
カメラの固定IPを変更したことで解決しました． 

From: @k-okada comment
> カメラに割り振られているIPを知る方法はあるのかな．あれば，READMEにかきましょう．Issueに書かないと他の人は分からないので書きましょう 

@ykawamura96
カメラのIPの確認・変更については，cobottaの専用のUI（タッチパネルから操作）から行っていたと記憶していますが，
その手順などが書かれたマニュアル（説明書？などのURL）等がありましたら，共有（もしくはREADME）してもらえるとうれしいです．

ykawamura96, 2021-06-07:
@k-kimura https://www.denso-wave.com/ja/robot/download/manual/
特にまとまったものはないので、マニュアルを読む必要が有るかと思います。