jsk-ros-pkg/jsk_robot #1390: [pr1040] /kinect_head/rgb/image_raw is transmitted from c1 to c2 in raw format

knorth55, 2021-09-24:
```
$ rostopic info /kinect_head/rgb/image_raw                                                                   
Type: sensor_msgs/Image

Publishers: 
 * /kinect_head/kinect_head_nodelet_manager (http://pr1040:43267/)

Subscribers: 
 * /kinect_head/kinect_head_c2_nodelet_manager (http://pr1040s:44699/)
```

```
$ rostopic info /kinect_head/rgb/image_raw/compressed
Type: sensor_msgs/CompressedImage

Publishers: 
 * /kinect_head/kinect_head_nodelet_manager (http://pr1040:43267/)

Subscribers: None
```
cc. @Affonso-Gui @tkmtnt7000

knorth55, 2021-09-24:
depth is also the same, i think

Affonso-Gui, 2021-09-24:
I'll take a look if we can pass the compressed node between c1 and c2 communications on the rectify node.