jsk-ros-pkg/jsk_robot #1391: [pr1040] network between c1 and c2 is slow these days

knorth55, 2021-09-24:
these days, network between c1 and c2 is quite slow...

Even when we run `web_video_server`, network connection got much slower, which cause launch failure with machine tag.

cc. @708yamaguchi @tkmtnt7000

k-okada, 2021-09-25:
@708yamaguchi @tkmtnt7000 @shmpwk @Kanazawanaoaki @HiroIshida 

please check email named
`JAXON_RED修理・不具合報告（jaxonredvisionの通信が遅い問題）` 
and check what is happening there.
```
1)　各PCのネットワークカード自体のスピードを確認する
https://askubuntu.com/questions/431911/how-can-i-verify-the-speed-of-my-nic-in-ubuntu
ethtool eth0
2) 実際にどれだけのスピードが出ているか確認する
https://askubuntu.com/questions/104755/how-to-check-internet-speed-via-terminal
ipref
ping でもよい．
```

708yamaguchi, 2021-09-25:
@k-okada 

Thank you for your help.

In c1:
```
$ ethtool lan0
Settings for lan0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: on (auto)
Cannot get wake-on-lan settings: Operation not permitted
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

```
$ ping c2
PING pr1040s (10.68.0.2) 56(84) bytes of data.
64 bytes from pr1040s (10.68.0.2): icmp_seq=1 ttl=64 time=0.161 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=2 ttl=64 time=0.327 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=3 ttl=64 time=0.130 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=4 ttl=64 time=0.098 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=5 ttl=64 time=0.191 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=6 ttl=64 time=0.176 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=7 ttl=64 time=0.359 ms
```

In c2
```
$ ethtool eth0
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
Cannot get wake-on-lan settings: Operation not permitted
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

```
$ ping c1
PING pr1040 (10.68.0.1) 56(84) bytes of data.
64 bytes from pr1040 (10.68.0.1): icmp_seq=1 ttl=64 time=0.146 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=2 ttl=64 time=0.125 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=3 ttl=64 time=0.217 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=4 ttl=64 time=0.129 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=5 ttl=64 time=0.124 ms
```

knorth55, 2021-09-25:
can you check the connection when you run rwt_teleop?

708yamaguchi, 2021-09-25:
OK.

I used `rwt_teleop` app and accessed rwt_teleop control monitor. 

The result is here.
ping sometimes slow.

In c1

```
$ ethtool lan0
Settings for lan0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: on (auto)
Cannot get wake-on-lan settings: Operation not permitted
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

```
$ ping c2
PING pr1040s (10.68.0.2) 56(84) bytes of data.
64 bytes from pr1040s (10.68.0.2): icmp_seq=1 ttl=64 time=0.154 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=2 ttl=64 time=2.32 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=3 ttl=64 time=0.211 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=4 ttl=64 time=2.16 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=5 ttl=64 time=0.185 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=6 ttl=64 time=0.564 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=7 ttl=64 time=2.65 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=8 ttl=64 time=10.8 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=9 ttl=64 time=10.3 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=10 ttl=64 time=2.99 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=11 ttl=64 time=0.143 ms
64 bytes from pr1040s (10.68.0.2): icmp_seq=12 ttl=64 time=0.146 ms
```

In c2

```
$ ethtool eth0
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
Cannot get wake-on-lan settings: Operation not permitted
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

```
$ ping c1
PING pr1040 (10.68.0.1) 56(84) bytes of data.
64 bytes from pr1040 (10.68.0.1): icmp_seq=1 ttl=64 time=0.096 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=2 ttl=64 time=1.05 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=3 ttl=64 time=0.147 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=4 ttl=64 time=0.086 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=5 ttl=64 time=2.07 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=6 ttl=64 time=6.73 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=7 ttl=64 time=3.70 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=8 ttl=64 time=0.349 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=9 ttl=64 time=1.62 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=10 ttl=64 time=2.48 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=11 ttl=64 time=0.136 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=12 ttl=64 time=1.04 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=13 ttl=64 time=0.341 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=14 ttl=64 time=0.228 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=15 ttl=64 time=1.17 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=16 ttl=64 time=0.364 ms
64 bytes from pr1040 (10.68.0.1): icmp_seq=17 ttl=64 time=6.04 ms
```