jsk-ros-pkg/jsk_robot #1407: pr2.l をロードしてからfetch-init するとエラーが出る

k-okada, 2021-11-10:
```
(load "package://pr2eus/pr2.l")

(load "package://fetcheus/fetch.l")
(load "package://fetcheus/fetch-interface.l")

(load "package://pr2eus/pr2.l") ;; comment to fix problem                                                             

(fetch-init)

```
を実行すると以下のエラーになります．`pr2.l`のロードをコメントするとエラーはでません．

```
;; (make-irtviewer) executed
[ WARN] [1636535978.172231168]: ;; collision-object-publisher wait for "apply_planning_scene" service for 5 sec

Call Stack (max depth: 20):
  0: at (setq glvertices gl::aglvertices)
  1: at (send #:inst19525 :init :replace-obj qhull :name "file:///tmp/collada_model_reader_base_link_LJ8bIg.dae")
  2: at (let ((#:inst19525 (instantiate collada-body))) (send #:inst19525 :init :replace-obj qhull :name "file:///tmp/collada_model_reader_base_link_LJ8bIg.dae") #:inst19525)
  3: at (instance collada-body :init :replace-obj qhull :name "file:///tmp/collada_model_reader_base_link_LJ8bIg.dae")
  4: at (setq geom (instance collada-body :init :replace-obj qhull :name "file:///tmp/collada_model_reader_base_link_LJ8bIg.dae"))
  5: at (let (geom glv qhull (local-cds (make-coords :pos (float-vector 0.0 0.0 0.0)))) (setq glv (instance gl::glvertices :init (list (list (list :type :triangles) (list :material (list (list :ambient (float-vector 0.178 0.1805 0.188 1.0)) (list :diffuse (float-vector 0.356 0.361 0.376 1.0)))) (list :indices #i(0 1 2 1 0 3 4 5 0 4 0 2 0 5 606 607 0 606 609 0 607 0 609 3 6 5 4 2 1617 4 6 4 1617 2 1616 1617 1618 6 1617 1617 1616 1627 1618 1617 1627 1615 1616 1 1 1616 2 1627 1616 1615 1627 1615 1632 1615 1 1632 2733 1 3 2733 1632 1 1627 1632 1633 2733 1633 1632
  10: at (apply #'send-message self (class . super) :init args)
  11: at (apply #'send-message self (class . super) :init args)
  12: at (apply #'send-message self (class . super) :init args)
  13: at (send-super* :init args)
  14: at (send #:inst2242 :init)
  15: at (let ((#:inst2242 (instantiate fetch-sensor-robot))) (send #:inst2242 :init) #:inst2242)
  16: at (instance fetch-sensor-robot :init)
  17: at (setq *fetch* (instance fetch-sensor-robot :init))
  18: at (fetch)
  19: at (progn (fetch) (send *fetch* :reset-pose))
  And more...
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 unbound variable gl::aglverticesgl::aglvertices in (setq glvertices gl::aglvertices)E: (exit)
[ INFO] [1636536145.179738257]: cell* ROSEUS_EXIT(context*, int, cell**)
```

mqcmd196, 2021-11-10:
追記:ファイルに書き起こしてload するとバグが発生する

mqcmd196, 2021-11-10:
hoge.lというファイルを作り
```lisp
(load "package://pr2eus/pr2.l")

(load "package://fetcheus/fetch.l")
(load "package://fetcheus/fetch-interface.l")

(load "package://pr2eus/pr2.l") ;; comment to fix problem                                                             

(fetch-init)
```

emacs-shell ではなく，terminal上で
```bash
rlwrap roseus hoge.l
```
とすると，たくさんの数字が表示されたのちに，
```
(when (member :reset-pose (send self :methods)) (send self :reset-pose)) self)
  10: at (apply #'send-message self (class . super) :init args)
  11: at (apply #'send-message self (class . super) :init args)
  12: at (apply #'send-message self (class . super) :init args)
  13: at (send-super* :init args)
  14: at (send #:inst2242 :init)
  15: at (let ((#:inst2242 (instantiate fetch-sensor-robot))) (send #:inst2242 :init) #:inst2242)
  16: at (instance fetch-sensor-robot :init)
  17: at (setq *fetch* (instance fetch-sensor-robot :init))
  18: at (fetch)
  19: at (progn (fetch) (send *fetch* :reset-pose))
  And more...
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 unbound variable gl::aglverticesgl::aglvertices in (setq glvertices gl::aglvertices)E:
```
と表示された

Naoki-Hiraoka, 2021-11-11:
`pr2.l`はeuscolladaの[このPR](https://github.com/jsk-ros-pkg/jsk_model_tools/pull/131)より前に変換・生成されたモデルで、`fetch.l`は後に変換・生成されたファイルなので発生しているエラーです。

knorth55, 2021-11-11:
pr2.l is regenerated in this PR #426
can you try this?

Naoki-Hiraoka, 2021-11-11:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/426 無しだと以下のコードで`error: unbound variable gl::aglvertices in (setq glvertices gl::aglvertices)`エラーになり、有りだとエラーになりませんでした。
```
(load "package://fetcheus/fetch-interface.l")
(load "package://pr2eus/pr2.l")
(fetch-init)
```

mqcmd196, 2021-11-11:
ありがとうございます。昨日の晩に@708yamaguchi さんと @tkmtnt7000 さんで確認したところ同じ結論に至りました。後で自分でも動作を確認してみます

k-okada, 2021-11-11:
https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/426 で冷蔵庫のデモができているか @mqcmd196 , カレーのプログラムが動くか @Kanazawanaoaki などなど確認してみてください．そうしたらマージします

mqcmd196, 2021-11-15:
@HiroIshida @Kanazawanaoaki @shmpwk テストしたら結果を教えてください，ゼミ生の環境に必要です

shmpwk, 2021-11-17:
自分のプログラムは動きました．
(後に金沢くんが指摘する`pr2-sensor-robot`メソッドを使っていなかったので)

Kanazawanaoaki, 2021-11-17:
遅くなりましたが，僕は
https://github.com/Kanazawanaoaki/jsk_demos/blob/8ee12e81f7fe0d46b6ac720ec0fb999f01342a16/jsk_2020_04_pr2_curry/euslisp/cook-with-pos-map/move-to-kitchen-with-map.l#L15 のように  
```
(setq *pr2* (instance pr2-sensor-robot :init))
```
を使っていたので，
https://github.com/jsk-ros-pkg/jsk_pr2eus/commit/81034eaf3ceb241a7d6681fff2a91b248e5bacf0#diff-412c1b20166961f159f5526b89e9b3b051a2f5eec4a2caee90fc4ddcc7e4da7eR12435 の変更によってnを指定しないとエラーがでるようになりました．https://github.com/jsk-ros-pkg/jsk_demos にはこれを使っているプログラムは無いようなので，僕のプログラムと僕が参考にしたhttps://github.com/jsk-ros-pkg/jsk_demos/pull/1282/files#diff-c38d27ce8591173e8296087714b11daac08bb0a4332d43b0905a89f72659fd4eR26 を修正すれば問題無いかもしれません．

k-okada, 2021-11-17:
なるほど，こういうのは後方互換性が崩れるので良くないですね．#466マージしちゃダメだったか．
一般的には，パブリックなメソッドかプライベートメソッドか，ということで，パライベートメソッドなら変更は気軽にしたら良いんだけど，eusはこの区別がないし，そもそもユーザがこのクラスのインスタンスを直接生成するとは思っていなかったら良いんだけど，そうでもなかったのかな #466 出直したのと，こういうことが二度とおこらないようにテストコードを追加ししました．