jsk-ros-pkg/jsk_robot #1442: PR2の簡単なシミュレーションをしたいときはどのlaunchファイルを立ち上げたら良いでしょうか。

nakane11, 2022-03-31:
```
roslaunch pr2eus_tutorials pr2_gazebo_empty.launch
```
で
```
(load "package://pr2eus/pr2-interface.l")

(pr2-init)
(objects (list *pr2*))

(defun test-1 ()
  (dotimes (i 10)
    (setq theta 0.15708) ;;[m/sec] [m/sec] [rad/sec] &optional (msec 1000) ;; msec is total animation time [msec] &key (stop t) (wait))

    (send *ri* :go-velocity 0 0 theta 1000 :wait t)
    (print (list 'send 0 0 theta))
    ))

(defun test-2 ()
  (send *pr2* :head :look-at (send *pr2* :larm :end-coords :worldpos))
  (send *irtviewer* :draw-objects)
  (send *ri* :angle-vector (send *pr2* :angle-vector) 2000) ;; no-blocking
  (send *ri* :wait-interpolation) ;; block
  (send *pr2* :head :look-at (send *pr2* :rarm :end-coords :worldpos))
  (send *irtviewer* :draw-objects)
  (send *ri* :angle-vector (send *pr2* :angle-vector) 2000)
  (send *ri* :wait-interpolation)
  )

```
のプログラムが動きました
`roslaunch pr2_gazebo pr2_empty_world.launch` だと `(send *ri* :go-velocity)` が`base_trajectory_action`がないというエラーになりました

Cc: @k-okada

shmpwk, 2022-03-31:
`roslaunch pr2eus_tutorials pr2_tabletop.launch remote:=true `を使ってください

knorth55, 2022-03-31:
`:go-velocity`はGazeboのLaunchでBaseTrajectoryActionを起動する必要があります
`:go-pos`ならeuslispのKinematics Simulatorで動くはずです