jsk-ros-pkg/jsk_robot #1447: daily workspace update script について

sktometometo, 2022-04-20:
fetch@fetch1075 の `$HOME/ros/melodic/src` で `wstool update` を実行すると、以下の様なエラーが出ます。

```bash
fetch@fetch1075:~/ros/melodic/src$ wstool update
/usr/lib/python2.7/dist-packages/wstool/config_yaml.py:74: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  yamldata = yaml.load(stream)
/usr/lib/python2.7/dist-packages/vcstools/tar.py:80: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  metadata = yaml.load(metadata_file.read())
[PR2/app_manager] Updating /home/fetch/ros/melodic/src/PR2/app_manager
[PR2/app_manager] Done.
[RobotWebTools/rosbridge_suite] Updating /home/fetch/ros/melodic/src/RobotWebTools/rosbridge_suite
[RobotWebTools/rosbridge_suite] Done.
[RoboticMaterials/FA-I-sensor] Updating /home/fetch/ros/melodic/src/RoboticMaterials/FA-I-sensor
[RoboticMaterials/FA-I-sensor] Done.
[fetchrobotics/fetch_ros] Updating /home/fetch/ros/melodic/src/fetchrobotics/fetch_ros
[fetchrobotics/fetch_ros] Done.
[fetchrobotics/robot_controllers] Updating /home/fetch/ros/melodic/src/fetchrobotics/robot_controllers
[fetchrobotics/robot_controllers] Done.
[jsk-ros-pkg/jsk_3rdparty] Updating /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_3rdparty
[jsk-ros-pkg/jsk_3rdparty] Done.
[jsk-ros-pkg/jsk_common] Updating /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_common
WARNING [vcstools] Command failed: 'git checkout master'
 run at: '/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_common'
 errcode: 1:
error: pathspec 'master' did not match any file(s) known to git.
[/vcstools]
[jsk-ros-pkg/jsk_demos] Updating /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_demos
WARNING [vcstools] Command failed: 'git checkout fetch15'
 run at: '/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_demos'
 errcode: 1:
error: pathspec 'fetch15' did not match any file(s) known to git.
[/vcstools]
[jsk-ros-pkg/jsk_pr2eus] Updating /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_pr2eus
[jsk-ros-pkg/jsk_pr2eus] Done.
[jsk-ros-pkg/jsk_recognition] Updating /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_recognition
[jsk-ros-pkg/jsk_recognition] Done.
[jsk-ros-pkg/jsk_robot] Updating /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot
WARNING [vcstools] Command failed: 'git checkout fetch15'
 run at: '/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot'
 errcode: 1:
error: pathspec 'fetch15' did not match any file(s) known to git.
[/vcstools]
[knorth55/app_manager_utils] Updating /home/fetch/ros/melodic/src/knorth55/app_manager_utils
WARNING [vcstools] Command failed: 'git checkout master'
 run at: '/home/fetch/ros/melodic/src/knorth55/app_manager_utils'
 errcode: 1:
error: pathspec 'master' did not match any file(s) known to git.
[/vcstools]
[mikeferguson/robot_calibration] Updating /home/fetch/ros/melodic/src/mikeferguson/robot_calibration
[mikeferguson/robot_calibration] Done.
[ros-drivers/audio_common] Updating /home/fetch/ros/melodic/src/ros-drivers/audio_common
WARNING [vcstools] Command failed: 'git checkout master'
 run at: '/home/fetch/ros/melodic/src/ros-drivers/audio_common'
 errcode: 1:
error: pathspec 'master' did not match any file(s) known to git.
[/vcstools]
[ros-drivers/rosserial] Updating /home/fetch/ros/melodic/src/ros-drivers/rosserial
[ros-drivers/rosserial] Done.
[ros-perception/slam_gmapping] Updating /home/fetch/ros/melodic/src/ros-perception/slam_gmapping
[ros-perception/slam_gmapping] Done.
[ros/common_msgs/visualization_msgs] Updating /home/fetch/ros/melodic/src/ros/common_msgs/visualization_msgs
/usr/lib/python2.7/dist-packages/vcstools/tar.py:168: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  metadata = yaml.load(metadata_file.read())
[ros/common_msgs/visualization_msgs] Done.
[strands-project/mongodb_store] Updating /home/fetch/ros/melodic/src/strands-project/mongodb_store
[strands-project/mongodb_store] Done.
[tork-a/roswww] Updating /home/fetch/ros/melodic/src/tork-a/roswww
[tork-a/roswww] Done.
[tork-a/visualization_rwt] Updating /home/fetch/ros/melodic/src/tork-a/visualization_rwt
[tork-a/visualization_rwt] Done.
[GT-RAIL/robot_pose_publisher] Updating /home/fetch/ros/melodic/src/GT-RAIL/robot_pose_publisher
[GT-RAIL/robot_pose_publisher] Done.
[IntelRealSense/realsense-ros] Updating /home/fetch/ros/melodic/src/IntelRealSense/realsense-ros
[IntelRealSense/realsense-ros] Done.
[fetchrobotics/fetch_open_auto_dock] Updating /home/fetch/ros/melodic/src/fetchrobotics/fetch_open_auto_dock
[fetchrobotics/fetch_open_auto_dock] Done.
[ros-planning/navigation] Updating /home/fetch/ros/melodic/src/ros-planning/navigation
[ros-planning/navigation] Done.
Exception caught during install: Error processing 'jsk-ros-pkg/jsk_common' : [jsk-ros-pkg/jsk_common] Update Failed of /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_common
Error processing 'jsk-ros-pkg/jsk_demos' : [jsk-ros-pkg/jsk_demos] Update Failed of /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_demos
Error processing 'jsk-ros-pkg/jsk_robot' : [jsk-ros-pkg/jsk_robot] Update Failed of /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot
Error processing 'knorth55/app_manager_utils' : [knorth55/app_manager_utils] Update Failed of /home/fetch/ros/melodic/src/knorth55/app_manager_utils
Error processing 'ros-drivers/audio_common' : [ros-drivers/audio_common] Update Failed of /home/fetch/ros/melodic/src/ros-drivers/audio_common

ERROR in config: Error processing 'jsk-ros-pkg/jsk_common' : [jsk-ros-pkg/jsk_common] Update Failed of /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_common
Error processing 'jsk-ros-pkg/jsk_demos' : [jsk-ros-pkg/jsk_demos] Update Failed of /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_demos
Error processing 'jsk-ros-pkg/jsk_robot' : [jsk-ros-pkg/jsk_robot] Update Failed of /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot
Error processing 'knorth55/app_manager_utils' : [knorth55/app_manager_utils] Update Failed of /home/fetch/ros/melodic/src/knorth55/app_manager_utils
Error processing 'ros-drivers/audio_common' : [ros-drivers/audio_common] Update Failed of /home/fetch/ros/melodic/src/ros-drivers/audio_common
```

これはエラーにもある通り、例えば `jsk-ros-pkg/jsk_robot` では以下のように `git checkout fetch15` がエラーになるためです。

```bash
fetch@fetch1075:~/ros/melodic/src/jsk-ros-pkg/jsk_robot$ git checkout fetch15
error: pathspec 'fetch15' did not match any file(s) known to git.
```

さらにこれは、複数のリモートでcheckout対象のブランチ名がある際に起きるためです。

```bash
fetch@fetch1075:~/ros/melodic/src/jsk-ros-pkg/jsk_robot$ git branch -a | grep fetch15
* (HEAD detached at origin/fetch15)
  battery-warning-email-fetch15
  fetch15-light-job
  fetch15-rebase
  fetch15-trash-can
  remotes/708yamaguchi/fetch-sound-no-replace-fetch15-
  remotes/affonso/fetch15
  remotes/mqcmd196/fetch15
  remotes/origin/20210920-fetch15
  remotes/origin/fetch15
  remotes/origin/fetch15-org
  remotes/origin/fetch15-rebase
  remotes/sktometometo/archived/fetch15-202106
  remotes/sktometometo/archived/fetch15-202109
  remotes/tkmtnt7000/fetch15-trash-can-rebased
```

これを対処するには、`wstool update` 時に実行されるコマンドが `git checkout <target branch>` ではなく, `git checkout -b <target branch> --track origin/<target branch>` になる必要があります。(実際には恐らく、すでに<target branch> がローカルにあるかどうかで場合分けが必要)

sktometometo, 2022-04-20:
今まで起きていなかったのは、local branch を初めに作ってからpullし続ける仕様だったためで、最近 https://github.com/knorth55/jsk_robot/blob/20669b842f2b046659abe8ce4f0179961c5e05ec/jsk_fetch_robot/jsk_fetch_startup/scripts/update_workspace.sh#L58-L66 のように target branch が remote と local で diverged した際に local のbranchを作り直してcheckout するようにした結果、これに伴い毎回 local のbranch を削除していて、branchの作り直しに失敗する状態の際に特定のリポジトリが上のような状態になっています。

sktometometo, 2022-04-20:
各リポジトリのローカルにtarget branchが存在していれば update-workspace.sh 実行時にpullされ、remoteのブランチにcheckoutされ直すので、もし上記のようなエラーが起きていた場合、エラーが起きているリポジトリに移動して以下のコマンドを実行してください。

```bash
git checkout -b <target branch> --track origin/<target branch>
```

708yamaguchi, 2022-04-20:
このissueは以下のプルリクエストと関連していると思うので、後から調べた人がわかりやすいようにリンクを貼っておきます。
https://github.com/knorth55/jsk_robot/pull/202

あまり良くわかっていないんですけど、そもそも上記のプルリクエストを作った理由は何でしょうか。
コミット名をみてもよく分からないので、どこかに追記してもらえると助かります。
cc @tkmtnt7000

sktometometo, 2022-04-20:
もともとは https://github.com/knorth55/jsk_robot/pull/197 のPRで出した https://github.com/knorth55/jsk_robot/commit/2c60a81039f0627702f7dcca585b631137154de1 のcommitが最初ですが、fetch@fetch15/fetch1075にあるlocalのリポジトリのブランチとremoteのブランチがdivergeしてしまった際に, wstool updateでは解消できないので自動で解決するように書いています。(jsk_recgonition等で発生した)

closed 2022-04-25 by sktometometo