jsk-ros-pkg/jsk_robot #1471: fetchの無線通信インターフェースが複数生成されると、動かないプログラムがある

708yamaguchi, 2022-04-25:
fetch15にUSBアクセスポイントを追加した関係で、wlan0とwlan1が生成されて、これまでのwlan0に依存していたプログラムが意図した動作をしていない問題がありました

https://github.com/knorth55/jsk_robot/blob/fetch15/jsk_fetch_robot/jsk_fetch_startup/scripts/network_monitor.py
https://github.com/knorth55/jsk_robot/blob/fetch15/jsk_fetch_robot/jsk_fetch_startup/scripts/log-wifi-link.sh

@sktometometo が提案した解決方法として、以下があります。
- スクリプト側のネットワークインターフェースの名前を変数にする
- 設定でネットワークインターフェースの名前をいじる

結局、以下の観点から前者の方法で解決しようと考えています。
```
NetworkManagerのプロファイル名のことを行っているのであれば、プロファイル名をWirelessWANとかにすればいいと思う。
後者はメンテナンスする人がudevを使った設定について知っている必要がある
前者は今のsupervisor+ROS+config.bashの環境変数設定のみの構成で実現できる。
ので、後者の場合はメンテナンスする人にudevに関する知識を要求するけど、個人的にはそれくらい調べたらと思うのでどちらでも良い。
```

708yamaguchi, 2022-04-25:
今は，自分たちで追加で取り付けていたUSBアクセスポイントを取り外しているので，この問題は発生していません．

sktometometo, 2022-05-17:
https://github.com/knorth55/jsk_robot/pull/227 によって修正されるはず。

closed 2022-05-17 by sktometometo