jsk-ros-pkg/jsk_robot #1474: naoqi_driver.launchが立ち上がらない

a-ichikura, 2022-04-27:
PepperとPCをリンクローカル接続し、naoqi_driver.launchを立ち上げたところエラーが出た。
```
leus@ubuntu:~/catkin_ws$ roslaunch naoqi_driver naoqi_driver.launch network_interface:=enp0s31f6
... logging to /home/leus/.ros/log/e39c984e-c602-11ec-8489-18cc1878776f/roslaunch-ubuntu-29940.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://169.254.52.171:44249/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

process[naoqi_driver-1]: started with pid [29962]
[W] 1651049468.264162 29975 qitype.signal: disconnect: No subscription found for SignalLink 18446744073709551615.
[W] 1651049468.265138 29962 qi.FutureSync: Error in future on destruction: 'Connect error: 113: No route to host' - continuing stack unwinding...
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  Connect error: 113: No route to host
================================================================================REQUIRED process [naoqi_driver-1] has died!
process has died [pid 29962, exit code -6, cmd /home/leus/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://169.254.99.161:9559 --roscore_ip=127.0.0.1 --network_interface=enp0s31f6 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/leus/.ros/log/e39c984e-c602-11ec-8489-18cc1878776f/naoqi_driver-1.log].
log file: /home/leus/.ros/log/e39c984e-c602-11ec-8489-18cc1878776f/naoqi_driver-1*.log
Initiating shutdown!
================================================================================
[naoqi_driver-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

a-ichikura, 2022-04-27:
次に、PepperとPCを同じネットワークに有線接続して、naoqi_driver.launchを立ち上げたところエラーがでた。
```
leus@ubuntu:~/catkin_ws$ roslaunch naoqi_driver naoqi_driver.launch network_interface:=enp0s31f6
... logging to /home/leus/.ros/log/e39c984e-c602-11ec-8489-18cc1878776f/roslaunch-ubuntu-451.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.97.82:37641/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    naoqi_driver (naoqi_driver/naoqi_driver_node)

ROS_MASTER_URI=http://localhost:11311

process[naoqi_driver-1]: started with pid [478]
Receiving information about robot model
Robot detected: Pepper B2C
set prefix successfully to naoqi_driver
using ip address: 127.0.0.1 @ enp0s31f6
found a catkin prefix /home/leus/catkin_ws/src/naoqi_driver/share/boot_config.json
load boot config from /home/leus/catkin_ws/src/naoqi_driver/share/boot_config.json
found a catkin URDF /home/leus/catkin_ws/src/naoqi_driver/share/urdf/pepper.urdf
terminate called after throwing an instance of 'qi::FutureUserException'
  what():  	ALProxy::ALProxy
	Can't find service: ROS-Driver-Audio
================================================================================REQUIRED process [naoqi_driver-1] has died!
process has died [pid 478, exit code -6, cmd /home/leus/catkin_ws/devel/lib/naoqi_driver/naoqi_driver_node --qi-url=tcp://192.168.97.189:9559 --roscore_ip=127.0.0.1 --network_interface=enp0s31f6 --namespace=naoqi_driver __name:=naoqi_driver __log:=/home/leus/.ros/log/e39c984e-c602-11ec-8489-18cc1878776f/naoqi_driver-1.log].
log file: /home/leus/.ros/log/e39c984e-c602-11ec-8489-18cc1878776f/naoqi_driver-1*.log
Initiating shutdown!
================================================================================
[naoqi_driver-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

a-ichikura, 2022-04-27:
＜解決策＞ from https://github.com/ros-naoqi/pepper_robot/issues/51

naoqi_driver/share/boot_config.json のaudio enabledをtrueからfalseにする。

kochigami, 2022-04-27:
なんで今まで私はこのエラーを経験したことなかったのかというと，Pepperと向き合ってプログラミングをあまりしていなかったから。
@a-ichikura はPepperと向き合って作業していたため，Pepperが色々自分に内蔵されているプログラムを立ち上げて，その中にaudioを使う音声認識があった．