jsk-ros-pkg/jsk_robot #1475: PR2のtwitterが更新されていません

nakane11, 2022-04-28:
https://twitter.com/pr2jsk

Kanazawanaoaki, 2022-04-28:
今確認して見たところ，https://github.com/knorth55/jsk_robot/blob/3dad74b99e5737c6e666927495792b79b29a8d3b/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_lifelog/pr2_tweet.launch#L6 で指定されている`/edgetpu_human_pose_estimator/output/image` のtopicがpublishされていなかったようです．
```
$ rostopic hz /edgetpu_human_pose_estimator/output/image
subscribed to [/edgetpu_human_pose_estimator/output/image]
no new messages
no new messages
```  
`/edgetpu_human_pose_estimator/output/image`が復活してtweetも復活したみたいです．
```
$ rostopic hz /edgetpu_human_pose_estimator/output/image
subscribed to [/edgetpu_human_pose_estimator/output/image]
average rate: 10.237
	min: 0.082s max: 0.110s std dev: 0.00910s window: 8
average rate: 10.063
	min: 0.059s max: 0.128s std dev: 0.01760s window: 18
average rate: 10.080
	min: 0.059s max: 0.128s std dev: 0.01734s window: 24
```

knorth55, 2022-04-28:
sudo initctl restart jsk-human-pose-estimatorしました

nakane11, 2022-04-28:
ありがとうございます

knorth55, 2022-04-28:
なんで止まっていたんだろう・・・

nakane11, 2022-04-29:
また止まっていました．edgetpu_human_pose_estimatorも使えたので気が付きませんでしたが，時間的に自分が原因だと思います．
[このlaunchファイル](https://github.com/nakane11/jsk_demos/blob/0619e4e4e05ddf3602c095355c1491dffa968ff4/hand_navigation_pr2/launch/hand_navigation.launch#L80)でnodeletを使ったのがよくなかったでしょうか

iory, 2022-05-24:
頻繁に```/edgetpu_human_pose_estimator/output/image```がpublishされなくなりますね。
本日以下の変更を加えて作業していたら、https://github.com/knorth55/jsk_robot/pull/239
audible warningの機能で
```Other sanity diagnostics:  edgetpu human pose estimator output image compressed Topic is not published```
と発話していました。
`edgetpu_human_pose_estimator`は`/kinect_head/rgb/image_rect_color`をsubscribeしていますが、こちらはpublishされていることを確認しました。
また`edgetpu_human_pose_estimator`も生きているようです。

```
applications@pr1040:~/ros/coral_ws/src/coral_usb_ros [pr2 *]$ rostopic hz /kinect_head/rgb/image_rect_color
subscribed to [/kinect_head/rgb/image_rect_color]
average rate: 30.317
        min: 0.010s max: 0.084s std dev: 0.02564s window: 27
^Caverage rate: 30.203
        min: 0.009s max: 0.084s std dev: 0.02788s window: 45
applications@pr1040:~/ros/coral_ws/src/coral_usb_ros [pr2 *]$ sudo tail -f -n10000    /var/log/upstart/jsk-human-pose-estimator.log
roscore/master is not yet running, will wait for it to start
roscore/master is not yet running, will wait for it to start
roscore/master is not yet running, will wait for it to start
master has started, initiating launch
... logging to /home/applications/.ros/log/43f1cc0e-db2b-11ec-b153-d05099c29feb/roslaunch-pr1040-17586.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/applications/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://pr1040:39755/
remote[c2-0] starting roslaunch
remote[c2-0]: creating ssh connection to c2:22
launching remote roslaunch child with command: [env ROS_MASTER_URI=http://c1:11311 /home/applications/ros/coral_ws/src/coral_usb_ros/launch/env.sh roslaunch -c c2-0 -u http://pr1040:39755/ --run_id 43f1cc0e-db2b-11ec-b153-d05099c29feb]
remote[c2-0]: ssh connection created

SUMMARY
========

PARAMETERS
 * /edgetpu_human_pose_estimator/device_id: 1
 * /edgetpu_human_pose_estimator/image_transport: raw
 * /edgetpu_human_pose_estimator/model_file: /home/application...
 * /rosdistro: indigo
 * /rosversion: 1.11.21

MACHINES
 * c1
 * c2

NODES
  /
    edgetpu_human_pose_estimator (coral_usb/edgetpu_human_pose_estimator.py)

ROS_MASTER_URI=http://c1:11311

core service [/rosout] found
[c2-0]: launching nodes...
[c2-0]: ROS_MASTER_URI=http://c1:11311
[c2-0]: process[edgetpu_human_pose_estimator-1]: started with pid [3142]
[c2-0]: ... done launching nodes
roscore/master is not yet running, will wait for it to start
roscore/master is not yet running, will wait for it to start
roscore/master is not yet running, will wait for it to start
master has started, initiating launch
... logging to /home/applications/.ros/log/eb37f02e-db30-11ec-b653-d05099c29feb/roslaunch-pr1040-18433.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/applications/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://pr1040:41399/
remote[c2-0] starting roslaunch
remote[c2-0]: creating ssh connection to c2:22
[... 91 lines cut ...]
 * /edgetpu_human_pose_estimator/model_file: /home/application...
 * /rosdistro: indigo
 * /rosversion: 1.11.21

MACHINES
 * c1
 * c2

NODES
  /
    edgetpu_human_pose_estimator (coral_usb/edgetpu_human_pose_estimator.py)

ROS_MASTER_URI=http://c1:11311

core service [/rosout] found
[c2-0]: launching nodes...
[c2-0]: ROS_MASTER_URI=http://c1:11311
[c2-0]: process[edgetpu_human_pose_estimator-1]: started with pid [26772]
[c2-0]: ... done launching nodes
applications@pr1040:~/ros/coral_ws/src/coral_usb_ros [pr2 *]$ rosnode ping /edgetpu_human_pose_estimator
rosnode: node is [/edgetpu_human_pose_estimator]
pinging /edgetpu_human_pose_estimator with a timeout of 3.0s
xmlrpc reply from http://pr1040s:55374/ time=1.163006ms

```