jsk-ros-pkg/jsk_robot #1476: [Fetch] Fetchのsanshiroへの接続方法について

sktometometo, 2022-04-28:
現在、fetch1075のsanshiroへの接続は、`hogehoge`というBSSIDをもつアクセスポイントへ紐付けていますが、このためにエレベータ付近など73B2から離れた場所へ移動すると無線の接続が安定しなくなります。
かといって、特定のBSSIDに紐付けないようにWiFiのプロファイルを設定してしまうと、普段ドックに居る際の接続が安定しません。

sktometometo, 2022-04-28:
解決策
- ネットワークプロファイルを複数用意して手動で切り替える。

k-okada, 2022-04-28:
>特定のBSSIDに紐付けないようにWiFiのプロファイルを設定してしまうと、普段ドックに居る際の接続が安定しません。

これは現在でもそうなっている？数年前よりアクセスポイントの強度を上げてもらっている等対応がすすでいるかもしれません。

解決策
・　エレベータ付近はもうエレベータ（通信が届かない）と思ってまずは一回実験してみる。何が問題になるかクリアにする。

sktometometo, 2022-04-28:
> 特定のBSSIDに紐付けないようにWiFiのプロファイルを設定してしまうと、普段ドックに居る際の接続が安定しません。

現在実験準備中にエレベータ付近で作業していた際に、BSSIDを固定しないネットワークプロファイルにしていて、その後73B2に戻った際に接続が切れることを確認しました。

k-okada, 2022-04-28:
B2で独自に設置sanshiro はまだあるのかな？
建物についているのはローミングをサポートしているという話をきいたことがあったけど，ちゃんと機能していないのかな？それともエレベータのWIFIの強度が強すぎるのかな？
--
◉ Kei Okada

sktometometo, 2022-04-28:
B2の野良sanshiroは2022/4/28現在なくなっています。
手元のスマートフォン等を接続した際は自動で近くのアクセスポイントにつながるように切り替わるのでローミングをすることそのものは可能だと思うのですが、FetchのPCではうまくいってませんね...

k-okada, 2022-04-28:
ノートPCはどうなのかな？
--
◉ Kei Okada

708yamaguchi, 2022-04-28:
Network Managerの`autoconnect-retries`を増やすとマシになったりしないでしょうか。

https://developer-old.gnome.org/NetworkManager/stable/settings-connection.html

sktometometo, 2022-04-28:
https://user-images.githubusercontent.com/9410362/165681599-cc24fa3e-1eb8-429e-ac7b-87007b53b700.mp4

1秒ごとにiwconfigを回しながら7階を一周してみました。ところどころ繋がっているアクセスポイントが切り替わっていることが確認できますが、一定以下まで接続が悪くならないと切り替わらない挙動な気がします。
以下の方法で調整できるか試してみてます。
https://askubuntu.com/questions/137460/wireless-roaming-aggressiveness

sktometometo, 2022-04-28:
Fetch1075で `sudo iwconfig wlan0 sens -70` と実行すると

> Error for wireless request "Set Sensitivity" (8B08) :
>    SET failed on device wlp0s20f3 ; Operation not supported.

と表示されているのですが、driverがサポートしていない可能性があります。

sktometometo, 2022-05-17:
この件ですが、現在Fetch1075では73B2での運用用と2号館での運用用の2つのNetworkManager プロファイルを作っておいて、73B2の外では手動で切り替えることで解決しています。

- `sanshiro-73B2`: 73B2前のアクセスポイントに固定
- `sanshiro-outside`: SSIDのみ指定、繋がっているアクセスポイントの電波が一定以上に弱くなると他のアクセスポイントに切り替わる。

PCのブート時は `sanshiro-73B2` に繋がっています。以下のコマンドで切り替えができます。

```bash
$ sudo nmcli c up sanshiro-outside
```

sktometometo, 2022-05-17:
上記の変更にともない、ネットワークプロファイル名を`sanshiro`で決め打ちにしていたプログラム(`network_monitor.py`)が動かなくなっていたはずでしたが、 https://github.com/knorth55/jsk_robot/pull/227 により修正されました。

closed 2022-05-23 by sktometometo