jsk-ros-pkg/jsk_robot #1482: [Fetch] L515のメッセージのタイムスタンプがとても古い

sktometometo, 2022-05-17:
以下の問題が `/l515_head/depth/color/points` について起こっている。

- `/l515_head/depth/color/points` を attention clipper で切り取ろうとしたら、Lookup Transform がFailして切り取れない現象が起きた。
- `/l515_head/depth/color/points`を表示しようとするとエラーがでて表示できない。

タイムスタンプがおかしそうなので以下のスクリプトで確認してみた。

```python
~ $ cat compare_timestamp.py 
#!/usr/bin/env python

import rospy
from sensor_msgs.msg import PointCloud2, Image

def callback(msg):

    rospy.loginfo('rospy.Time.now() - msg.header.stamp = {}'.format((rospy.Time.now() - msg.header.stamp).to_sec()))

if __name__ == '__main__':

    rospy.init_node('comapre_timestamp')
    sub = rospy.Subscriber('~input', PointCloud2, callback)
    #sub = rospy.Subscriber('~input', Image, callback)
    rospy.spin()
```

`/l515_head/depth/color/points` と `/head_camera/depth/points` に対して実行してみた。

```bash
~ $ ./compare_timestamp.py ~input:=/l515_head/depth/color/points
[INFO] [WallTime: 1652756457.982523] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25885.68982
[INFO] [WallTime: 1652756458.030443] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25885.7046223
[INFO] [WallTime: 1652756458.097531] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25885.738476
[INFO] [WallTime: 1652756458.169425] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25885.7772663
[INFO] [WallTime: 1652756458.243575] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25885.8182342
```

```bash
~ $ ./compare_timestamp.py ~input:=/head_camera/depth/points                                                                     
[INFO] [WallTime: 1652756266.785021] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.045728691  
[INFO] [WallTime: 1652756266.818866] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.046195514 
[INFO] [WallTime: 1652756266.845641] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.039635904  
[INFO] [WallTime: 1652756266.879731] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.04033447   
[INFO] [WallTime: 1652756266.911721] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.039020069  
[INFO] [WallTime: 1652756266.945281] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.039201982  
```

25885秒 = 7時間11分25秒 遅れてる。

一応、カラー画像のトピックでも確認

```bash
~ $ ./compare_timestamp.py ~input:=/l515_head/color/image_raw                                                                      
[INFO] [WallTime: 1652756741.667528] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25886.7781131
[INFO] [WallTime: 1652756741.697311] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25886.7747502
[INFO] [WallTime: 1652756741.723974] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25886.7682221
[INFO] [WallTime: 1652756741.759524] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25886.7706161
[INFO] [WallTime: 1652756741.791509] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25886.7693741
[INFO] [WallTime: 1652756741.824739] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25886.769443
[INFO] [WallTime: 1652756741.858045] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 25886.7695084
```

```bash
~ $ ./compare_timestamp.py ~input:=/head_camera/rgb/image_rect_color
[INFO] [WallTime: 1652756795.687069] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.03791023
[INFO] [WallTime: 1652756795.724110] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.041357286
[INFO] [WallTime: 1652756795.753328] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.037142522
[INFO] [WallTime: 1652756795.786732] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.03699804
[INFO] [WallTime: 1652756795.819938] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.036771066
[INFO] [WallTime: 1652756795.853146] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.036468513
[INFO] [WallTime: 1652756795.887848] [node:/comapre_timestamp] [func:callback]: rospy.Time.now() - msg.header.stamp = 0.037619121
```

sktometometo, 2022-05-17:
@knorth55 @98shimpei 同様の問題起こっていますか?

Naoki-Hiraoka, 2022-05-17:
自分の場合は、タイムスタンプがだんだん遅れていく問題は
```
<group ns="rs_l515">
    <param name="rgb_camera/global_time_enabled" value="true"/>
    <param name="l500_depth_sensor/global_time_enabled" value="true"/>
  </group>
```
とすると解消しました。

knorth55, 2022-05-17:
確かにtimestampが少しずれることはありました．
こんな方法があるんですね，ありがとうございます．

closed 2022-05-23 by sktometometo