jsk-ros-pkg/jsk_robot #1489: fetchのsupervisorのジョブが止まっている

708yamaguchi, 2022-05-25:
以下のようなログを最後にfetchのjsk-fetch-startupのジョブが止まっていました
どのプログラムが原因でしょうか。

```
$ sudo tail -n 1000 /var/log/ros/jsk-fetch-startup.log
（略）
waiting for roscore to come up
ERROR: Unable to communicate with master!
ERROR: Unable to communicate with master!
set ROS_IP and ROS_HOSTNAME to 133.11.216.218
Failure: No such entity
Failed to get sink information: No such entity
master has started, initiating launch
... logging to /home/fetch/.ros/log/107b73a6-dc5d-11ec-8091-6175a79f8585/roslaunch-fetch1075-1298.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/fetch/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
2;/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/launch/fetch_bringup.launch^G
RLException: Unable to contact my own server at [http://133.11.216.218:41817/].
This usually means that the network is not configured properly.

A common cause is that the machine cannot connect to itself.  Please check
for errors by running:

        ping 133.11.216.218

For more tips, please see

        http://wiki.ros.org/ROS/NetworkSetup

The traceback for the exception was written to the log file
```
[image]

sktometometo, 2022-05-26:
IPが振られる前に立ち上がっていそう.

pazeshun, 2022-05-26:
関係あるかわからないんですが、僕のグリッパでも、外部と通信できるようになる前にsupervisorのjobが立ち上がってしまい、外部のroscoreを見つけられないことが多く、結局外部にpingできるようになるまで待つ、というのをやってました。
https://github.com/start-jsk/jsk_apc/blob/8861d260f8ebcbf8246fb8ffb10e6affa04794c1/demos/sphand_ros/sphand_driver/supervisor/launch_left_gripper.conf#L3
https://github.com/start-jsk/jsk_apc/blob/8861d260f8ebcbf8246fb8ffb10e6affa04794c1/demos/sphand_ros/sphand_driver/scripts/launch_left_gripper.sh#L3-L7
https://github.com/pazeshun/sphand_ros/blob/7ae4e88e586b98847d2c39a6532ddf8c1362534a/sphand_driver/supervisor/launch_left_gripper.conf#L3