jsk-ros-pkg/jsk_robot #1496: [fetch1075] 頭部USBデバイスに頻繁にアクセスできなくなる

708yamaguchi, 2022-06-02:
## 症状

今日何度もfetch1075でキッチンデモを行っていたのですが、デモ終了時（ドックに戻った時）に`respeaker_node`が頻繁に落ちました。

以下のようなエラーが出た後に、
```
Expression 'alsa_snd_pcm_start( stream->capture.pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2934
Expression 'AlsaStart( stream, 0 )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3246
Expression 'AlsaRestart( self )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3313
Expression 'PaAlsaStream_HandleXrun( self )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3939
Expression 'PaAlsaStream_WaitForFrames( stream, &framesAvail, &xrun )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 4274
Expression 'alsa_snd_pcm_drop( stream->capture.pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3046
```
以下のログと共に`respeaker_node`が落ちました。
この時、`$ lsusb -v | grep -i respeaker`すると、respeakerが見えるときもあったし見えないときもありました。
```
[ERROR] [1654163461.020162] [/respeaker_node:rosout]: [Errno 19] No such device (it may have been disconnected)
[respeaker_node-92] process has finished cleanly
```

このあとRespeakerのUSBを抜き差しした瞬間や、抜き差しせずにずっと放置している時に、head_cameraのrostopicも見えなくなってしまいました。

## 対策

不具合が発生しているUSBデバイス（Respeakerおよびhead_camera）が頭部に集中していることから、fetch頭部のUSBハブを交換しました。
fetch移動時にUSBハブが振動して衝撃を与えないように、結束バンドを使って固定しました。

交換してから数時間くらいは不具合が見当たりませんが、当分様子を見ます。

## 原因

はっきりと分かったわけではないですが、USBハブを交換して症状が収まっているということは、キッチンデモ終了時にCPU負荷が上がったなどのソフトウェア的な理由ではなく、経年劣化やドック時の衝撃でUSB接続が甘くなったなどのハードウェア的な理由なのかな、と考えています。

knorth55, 2022-06-02:
頭部のUSBといっているのはどの部品なんでしょうか？
写真とかがあるとみたいです．

knorth55, 2022-06-02:
Fetch頭部のデフォルトのUSBの口ではなくて増設したやつの話だったか

708yamaguchi, 2022-06-02:
すいません、
今は頭部USBから伸ばしたUSBハブにrespeakerとinsta360、PS3コントローラ用給電ケーブルを取り付けていました。

nakane11, 2022-08-18:
今朝8:48頃にfetchを起動したところ、以下のエラーが出てrespeakerが認識されていませんでした。
```
No handlers could be found for logger "rosout"
Traceback (most recent call last):
  File "/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_3rdparty/respeaker_ros/scripts/respeaker_node.py", line 33, in <module>
    raise RuntimeError("Check the device is connected and recognized")
RuntimeError: Check the device is connected and recognized
[respeaker_node-96] process has died [pid 26885, exit code 1, cmd /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_3rdparty/respeaker_ros/scripts/respeaker_node.py __name:=respeaker_node __log:=/home/fetch/.ros/log/0bac85ce-1e87-11ed-b2f2-b12a439c0bfe/respeaker_node-96.log].
log file: /home/fetch/.ros/log/0bac85ce-1e87-11ed-b2f2-b12a439c0bfe/respeaker_node-96*.log
```
`$ lsusb -v | grep -i respeaker`と`$ dmesg --follow | grep -i usb`で抜き差しに反応していないことを確認し、次の作業で直りました。
1. 自分のpcにrespeakerとrespeakerに直接繋がっている白いケーブルを接続し、認識されることを確認
2. usbハブで余っている他のポート2箇所にそれぞれ接続しても認識されない
3. 同じハブに接続されているコントローラは認識されることを確認
4. コントローラが接続されていたポートにrespeakerを接続して認識
5. 元のrespeakerのポートに接続して認識される

708yamaguchi, 2022-08-18:
報告ありがとう。
ログを見るに、2022/08/18 2:56の次のログが2022/08/18 08:48くらいになっていました。
たぶんシャットダウンされているfetch1075を中根さんが起動してくれたということですよね。

昨日のfetch1075の状態を覚えていないので申し訳ないのですが、中根さんがfetchを起動してくれたときは以下のどっちの状態だったか覚えていますか？
- 完全にシャットダウンされた状態だった（fetch側面のボタンによるシャットダウン）
- PCの電源は切れているけどUSB給電はされ続けていた。（sudo shutdownによるシャットダウン。fetch1075の青い電源ボタンやrespeakerのLEDが光っていたらこの状態）

似たような問題が頭部スピーカにも起きている可能性があります。
スピーカやrespeakerが見えなくなることがある原因として考えられるのは、fetchの頭部にUSBデバイスを接続しすぎているせいで電圧降下？やバスの通信帯域の圧迫？が発生している、というところでしょうか。
https://github.com/jsk-ros-pkg/jsk_robot/issues/1551#issuecomment-1202259245

問題が頻発するようであれば、頭部のUSBデバイスを必要ない順に外して様子を見るのが良いのかなと思っています。
fetch1075であれば、スピーカは正直必須ではないです。その次にhead_cameraと役割がかぶるL515を抜けばいいのかな。

nakane11, 2022-08-18:
完全にシャットダウンされた状態で電源ボタンの光が消えていました。

708yamaguchi, 2022-08-18:
だとすると、以下のコメントのパターン1ですかね。
前回はスピーカとセルフパワーUSBハブの間に別のUSBハブを介さず直接接続することで解決したのですが、今回のrespeakerもそうしてみます。
https://github.com/jsk-ros-pkg/jsk_robot/issues/1551#issuecomment-1202259245

ただセルフパワーUSBハブの口も4つしかなく、いま接続したいデバイスの数が5つ？あるはずなので、限界はあります。
まずはPS3コントローラをfetch頭部につけるのをやめて、残り4つのデバイスをセルフパワーUSBハブに直接つけるのが良いのかな、と思ってきたのですが、どうでしょうか。
- L515
- insta360
- respeaker
- スピーカ
- PS3 controller

708yamaguchi, 2022-08-18:
> まずはPS3コントローラをfetch頭部につけるのをやめて、残り4つのデバイスをセルフパワーUSBハブに直接つけるのが良いのかな

いまこれをやっています。
PS3コントローラはPR2デモ用冷蔵庫の上で充電しています。
またデバイスに接続されていない問題が発生したら報告してもらえると助かります。