jsk-ros-pkg/jsk_robot #1505: jsk-ros-pkgへPRを出すタイミング・内容について

sktometometo, 2022-06-12:
議論の場があったほうがいいかなと思い、Issue立てました。

Naoki-Hiraoka, 2022-06-12:
コードを研究室でできるだけ統合していくという方針に対して、反対する人はいないのではないのでしょうか。ただ、そのための手段について、議論が必要と考えます。

現在の手段でうまくまわっているタイプのソフトウェアと、まわっていないタイプのソフトウェアがあります。まわっていないタイプのソフトウェアについては、何年も昔にはまわっていたようですが、現状まわっていないのを誰かがなんとかしてくれることに期待して今後の自分の開発に現在の手段を使うことは、リスクが大きいです。自分の力で現状をなんとかできればよいですが、それも難しいです。そのため、無いものには期待せず手持ちのカードのなかで勝負するために、現在の手段を採用しない、という選択をとることが現実味を帯びています.

さらに、現在の手段を採用しないことは自分のためだけではなく、後輩のためにもなってしまっています。僕の実経験として、OBが残していったmasterにmergeされていないコードを保守する手間は、ブランチ(PR)の状態で残したものよりも、リポジトリを分けた状態で残したものの方が楽です。

現状をなんとかするために、現状の手段の問題点を分析し、今後どのような手段をとるべきかを検討すべきと考えます。

iory, 2022-06-12:
> さらに、現在の手段に従わないことは自分のためだけではなく、後輩のためにもなってしまっています。僕の実経験として、OBが残していったmasterにmergeされていないコードを保守する手間は、ブランチ(PR)の状態で残したものよりも、リポジトリを分けた状態で残したものの方が楽です。

はい、動く状態でbranchを残しておくのは良いと思います。
理想はjsk_〇〇に個人がPRを送ることですが、理想はあくまで理想なので、この手段はなかなか難しいと思います。
なので、最近は卒業する前にgitlabにコードを残しておくということになっています。これは岡田先生が修論・卒論などで最後にチェックするので機能しています。
なので、平岡くんが無理してすべてのコードを保守するというのはリソース的にも無理しないほうが良いと思います。

すべてのコードをPR送るのは難しいと思うので、この機能はgeneralにみんなに使ってもらえるというものは積極的にPRを送ったり、
こういう問題がある・こういう機能がないですかという質問をissueで立ててもらえる良いと思います。
少なくとも、平岡くんが以下のPRで追加で作っていたパッケージは、JSKの他のロボットを用いている人たちにとっても有益なはずです。
ここでも理想はPRを送ることですが、PRを送ることはissueがあって初めて必要となるので、1段手順を飛ばしています。
なので、issueを立てるということから小さく始めましょう。

少なくとも、平岡くんが以下のPRで追加で作っていたパッケージは、JSKの他のロボットを用いている人たちにとっても有益なはずです。

https://github.com/jsk-ros-pkg/jsk_robot/pull/1504
https://github.com/Naoki-Hiraoka/tf_publishers/tree/master/odom_tf_publisher
https://github.com/Naoki-Hiraoka/jsk_robot/commit/4602a8a7c932ca55f5e09b25ae0cf7fe836c9c32

この問題ももともとはそんなに難しいことを要求しているのではなくて、ひとまず全員で情報共有できるのがベターですね。
使っているプログラムに問題があったら共有するという点については、反対する人もいないのではないかと思います。（もしこの点に関してなにかあれば言ってください）

> #1312 は相対位置をTFから取得する際のwaitfortransformに時間がかかることがありcallbackの周期が遅くなる問題があった

ここでまずissueが1つ立てられて、

> #1312 は歴史的経緯や後方互換性のためにロボットの姿勢が水平でないときに悪さをする処理が入っていたので

これでまた1つissueが立てられます。

で、さきほどofflineで @sktometometo に聞いたら、`CameraToBaseOffset`は毎回lookupTransformをしているけど、
base_linkに取り付いているとこの必要がないから`CameraToBaseOffset`とは異なるということを聞いたので、これでまた1つissueが立てられます。

で、`CameraToBaseOffset`はdocumentがないので、使い方がわかりませんというissueが1つ立てられます。

このPRだけで4つissueが立てられるので、これらを解決するPRを4つ送れるということになります。
それぞれのPRは粒度が小さくなるので問題もわかりやすくreviewする側も見やすくなって嬉しいですね（必ずしもこの粒度でやったほうが良いというわけではないですが）

issueをたてるのに心理的な障害があるとも思うので、他の方法としてはgoogle chatで聞いてみることや、研究会で聞いてみる、とりあえず助教の先生や先輩にオフラインで聞いてみるというのが良いです。おすすめはテキストの文面で聞いたあとに（これはgithubなどのissueやPRコメントも含みます）、直接オフラインで聞いてみたりすることです。

Naoki-Hiraoka, 2022-06-12:
> ここでも理想はPRを送ることですが、PRを送ることはissueがあって初めて必要となるので、1段手順を飛ばしています。
> なので、issueを立てるということから小さく始めましょう。

そうですね。何にせよそこから小さく始める必要がありますね。

題材によっては各人のキャパシティや研究領域の重なりの観点から返事をあまり期待できなかったり、もとのソフトウェア製作者やパッケージ全体が想定している研究領域と違う研究領域でプログラム使って問題になった場合にプログラムに問題があるわけではないのでissueを立てづらかったり、jsk_〇〇のプログラムが実際に使用しているプログラムと大きく乖離していてここにissueを立ててもなあと感じたり、というのがありました。#1312 がそうでした。それでも、情報共有目的にissueを立てておくことは有益なので、ここは自分も改めます。

knorth55, 2022-06-12:
話をややこしくするかなと思いますが、私個人の意見ではPRは適当でもいいから立てることはできる、そこまで難しいことではない、と考えています
それは私はPRは運が良ければマージされるもので基本的にはまーじされないもの、だと言う認識のもと出しているからです
この認識の方が特異な気がしますが、これをやっておくとIssueとコードの二つがそこにあって便利だなと思っています
ただマージされないPRを出すことに徒労感を感じるという見解も理解できます
なので、Issueなど納得のいくところから始めて行ってもらえるとありがたいです