jsk-ros-pkg/jsk_robot #1507: Visual odometry の使用用途についての調査

sktometometo, 2022-06-13:
T265を皆さんいろいろな使い方しているっぽいので、Visual odometryどう使っているか教えてください。
とくに以下が知りたい

- オドメトリの原点の座標系がどうなっているか
  + ロボットのベースリンクに合わさっているか、カメラのままかとか
  + z軸鉛直であることを利用しているかどうかとか
- `nav_msgs/Odometry` の各フィールドのうち何を使っているか
  + pose
  + pose の covariance
  + twist
  + twist の covariance

@98shimpei @Naoki-Hiraoka

ドローン班もつかってましたっけ? @tongtybj

sktometometo, 2022-06-13:
FetchはFetch15で使っていて、odometry原点の座標系はbase_linkの高さに合わせてあって、z軸鉛直であることは使っていない(設定上だいたいあってはいる).
基本的には amcl での自己位置推定用なので, pose があればいいはず (covariance は使われているかどうか不明)
@haraduka が論文書いたときは twist も使った。

Naoki-Hiraoka, 2022-06-13:
自分はT265をJAXONの全身行動で使っていて、local mapのための自己位置としてT265の出力をそのまま使っています。また、rtabmapなどのglobal mapのための自己位置推定用の入力としても使っています。そのため、使っているのは`pose`だけです。

XYZとyawは絶対値を取得しようとするとloop closureや誘拐ロボット問題でジャンプすることが避けられないので、連続性が要求されるlocal mapではオフセットが蓄積することを許容しています。一方で重力方向については、IMUから絶対値を常に取得可能であることと、全身バランスに直接的に影響を与えるため、local mapでも絶対値が正しいことを要求しています。

そのため、オドメトリの原点の座標系として、t265が出すトピックの原点の座標系をそのまま使っています. また、t265内部でloop closureや誘拐ロボット問題の処理を行わないようにオプションを設定しています。
```xaml
<param name="tracking_module/enable_relocalization" value="false"/>
<param name="tracking_module/enable_pose_jumping" value="false"/>
```

Naoki-Hiraoka, 2022-06-13:
この前 @98shimpei と確認したのですが、T265のcovarianceはその絶対値に数学的意味はあまりなくて、`confidence`が`High` `Medium` `Low` `Fail`のいずれかによって、それぞれ`0.01` `0.1` `1` `10`になるそうです。

https://github.com/IntelRealSense/realsense-ros/issues/770#issuecomment-510340471

https://github.com/IntelRealSense/realsense-ros/blob/e4938bbbfd7b285fe71f81fe69cebcc76dbf72fc/realsense2_camera/src/base_realsense_node.cpp#L1581-L1582

98shimpei, 2022-06-13:
JAXONの環境蓄積のために使っています。
オドメトリの原点はロボットのベースリンクに合わさっています。また、heightmapを使うためにオドメトリのZ軸が鉛直であることを利用しています。
poseを使っており、twistは使っていません。（最近poseのcovarianceを使うことを検討しています。）

98shimpei, 2022-06-13:
質問の意味を勘違いしていました。
オドメトリの原点は地面（初期状態の足の位置）の高さに合わせています。

sktometometo, 2022-06-13:
covariance は robot_pose_ekf みたいにKalmanFilterにodomを入力するとかしない限りあまり使われなさそうなので、とりあえず対応しなくても良さそう？CameraToBaseOffsetのrotationみたいに適当な変換するくらいなら0にするか一定値入れといたほうがいいかもしれないけど...