jsk-ros-pkg/jsk_robot #1512: [pr1012] PR1012sのrecify処理が遅い

knorth55, 2022-06-24:
PR1012sでの`/kinect_head/rgb/image_rect_color`のHzが遅い（10-15Hz）という話がありました．
PR1012での`/kinect_head/rgb/image_raw`，PR1012sでの`/kinect_head/rgb/image_raw`はどちらも30Hzでていて，
PR1012s内部でのRectify処理がそもそも遅いとみています．
PR1012sのCPU温度を確認したところ，100度に張り付いていて，これが原因ではないかと考えています．

### PR1012sでjsk-pr2-startupが走っているとき

critialな100度まで達している

```
applications@pr1012s:~ $ sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0: +100.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 0:         +94.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 1:         +95.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 2:         +92.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 3:         +91.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
```

### PR1012sでjsk-pr2-startupがとまっているとき

少し下がっているがそれでも高い

```
applications@pr1012s:~ $ sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +98.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 0:         +92.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 1:         +92.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 2:         +89.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
Core 3:         +88.0°C  (high = +80.0°C, crit = +100.0°C)  ALARM (CRIT)
```

### (参考) PR1040sでjsk-pr2-startupが走っているとき

すこし高いきもするが，90度には１つも達していない．

```
applications@pr1040s:~ $ sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +87.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:         +79.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +78.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +79.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +87.0°C  (high = +80.0°C, crit = +100.0°C)
```

ファンがまわっているか，グリスがはがれていないか，などを確認します．

cc. @Kanazawanaoaki @iory

Kanazawanaoaki, 2022-06-24:
@knorth55 とpr1012のc2を確認したところ，やはりファンが回っていなかったのでこれが原因のようです．  
新しいファンを発注しようと思います．

Kanazawanaoaki, 2022-06-27:
CPUファンを新品に交換してグリスも塗り直したところ，C2(pr1012s)の温度が低くなってrecify処理も遅くなることが無くなったように思います．  

#### PR1012sでjsk-pr2-startupが走っているとき(CPUファン交換後)
```
applications@pr1012s:~ $ sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +49.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:         +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +45.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +41.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +41.0°C  (high = +80.0°C, crit = +100.0°C)

power_meter-acpi-0
Adapter: ACPI interface
power1:        4.29 MW (interval =   1.00 s)
```
これを見ているとpr1040sはやはり高めな気がするので，不調になったらCPUファンとグリスの交換が必要そうです．  

recify処理も正常に戻ったようです．
```
applications@pr1012s:~ $ rostopic hz /kinect_head/rgb/image_rect_color
subscribed to [/kinect_head/rgb/image_rect_color]
average rate: 30.035
        min: 0.030s max: 0.037s std dev: 0.00223s window: 28
average rate: 28.437
        min: 0.007s max: 0.085s std dev: 0.01208s window: 55
average rate: 28.623
        min: 0.007s max: 0.085s std dev: 0.01143s window: 84
average rate: 28.973
        min: 0.007s max: 0.085s std dev: 0.00989s window: 114
average rate: 29.185
        min: 0.007s max: 0.085s std dev: 0.00886s window: 144
average rate: 29.329
        min: 0.007s max: 0.085s std dev: 0.00810s window: 174
average rate: 29.425
        min: 0.007s max: 0.085s std dev: 0.00756s window: 204
average rate: 29.505
        min: 0.007s max: 0.085s std dev: 0.00709s window: 235
average rate: 29.560
        min: 0.007s max: 0.085s std dev: 0.00673s window: 265
```
cc.@knorth55

knorth55, 2022-06-28:
今日朝ためしたところ，同様に50度以下でとまっています．
これが正しい挙動っぽいですね．
CPUファンがとまっていましたが，そもそもグリスも悪かったのではないか，と勝手に推測してます．
CPUファンをとりだしたときにはもうガチガチに乾いていたので，PR1040のほうも近々取り替えることになるかもしれないです．

```
applications@pr1012s:~ $ sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +49.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:         +47.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +46.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +43.0°C  (high = +80.0°C, crit = +100.0°C)

power_meter-acpi-0
Adapter: ACPI interface
power1:        4.29 MW (interval =   1.00 s)
```

knorth55, 2022-06-28:
pr1040sでstressテストしたところ100度にすぐに達しました．
こっちもかなりダメそうですね．

```
applications@pr1040s:~ $ sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0: +100.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +100.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +98.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +98.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +97.0°C  (high = +80.0°C, crit = +100.0°C)

power_meter-acpi-0
Adapter: ACPI interface
power1:        4.29 MW (interval =   1.00 s)
```

closed 2022-10-19 by knorth55