jsk-ros-pkg/jsk_robot #1514: [Fetch15] send *ri* :angle-vector failed

nakane11, 2022-06-28:
I tried to send angle-vector to fetch15 from my pc, but failed with the error as follows.
```
(send *ri* :angle-vector (send *robot* :reset-pose) 2000)
;; motion plan error at (-1 . PLANNING_FAILED)
[ WARN] [1656409551.432603088]: Collision is found in the current state.
Contact object pairs:
(head_box_link COLOBJ7612)
(head_pan_link COLOBJ7612)
(head_tilt_link COLOBJ7612)
(base_link base_camera_mount)
(keepout base_camera_mount)
nil
```
any angle-vector value including reset-pose did not work.

We noticed something wrong with [headbox-collision-object](https://github.com/jsk-ros-pkg/jsk_robot/blob/42def1f440806cf1a1c0e6332da4b77320eb82cd/jsk_fetch_robot/fetcheus/fetch-interface.l#L246-L255) which recently merged. https://github.com/jsk-ros-pkg/jsk_robot/pull/1186
[image]
Is it related to  the comment `;; fetch must be :reset-pose when we run this method`?
cc: @sktometometo

knorth55, 2022-06-28:
so the solution is adding the links into `touch-links`.

tkmtnt7000, 2022-06-28:
I have a feeling that the headbox is different from the current Fetch hardware configuration, and perhaps it might be a good idea to disable it or reflect current Fetch's head hardware by default.

sktometometo, 2022-06-28:
https://github.com/knorth55/jsk_robot/pull/276 で (send *ri* :angle-vector (send *fetch* :angle-vector) 5000) したときに動くことを確認.

sktometometo, 2022-06-28:
再発したらOpenしてください。クライアントの fetcheus が jsk-ros-pkg/master だと問題が起きるかも。

closed 2022-06-28 by sktometometo