jsk-ros-pkg/jsk_robot #1518: [fetch1075] height for label "motorcycle" (id:3) is not specifiedの出力がずっとでている

708yamaguchi, 2022-06-30:
fetchのcoral usbで、

```
rosservice call /edgetpu_object_node_manager/start "name: 'edgetpu_panorama_object_detector'"
```
したところ、以下のようなWARNが1秒おきくらいに出続けます。

```
[WARN] [1656606926.673252] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606926.861368] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606927.704377] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606928.609025] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606929.509044] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606929.709298] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606929.941138] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606929.942045] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606930.169404] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606930.170387] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606931.871651] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
[WARN] [1656606933.211335] [/edgetpu_panorama_object_detection_rect_array_in_panorama_to_bounding_box_array:rosout]: height for label "motorcycle" (id:3) is not specified
```

出力頻度を落とすか一度しか出力しないようにするのが良いと思うのですが、どうでしょうか。
また、どのプログラムを修正すればよいでしょうか。

sktometometo, 2022-07-01:
jsk_percetion の rect_array_in_panoram_to_bounding_box_array です。

sktometometo, 2022-07-01:
ROSCONSOLE_FORMAT っていじってないのだっけ。log に node name が出るようにできる。

sktometometo, 2022-07-01:
出てるね。

sktometometo, 2022-07-01:
throttle してしまうと欲しいエラーも落ちるので、つかうならthrottle_identicalにしてくれると嬉しい。
http://wiki.ros.org/rospy/Overview/Logging

sktometometo, 2022-07-01:
ソースコードの方を弄ってログレベルをinfoかdebugに落としてもいいかも。最近はこのスクリプトfixしてきたのでwarnで出さなくても困らない気がする。

708yamaguchi, 2022-07-01:
ありがとう

logdebugを使うようにプルリクエストを出しました
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2709

closed 2022-07-01 by 708yamaguchi