jsk-ros-pkg/jsk_robot #1520: [fetch15][fetch1075] fetchのネットワークが繋がらない

sktometometo, 2022-07-04:
2022/07/03の日曜日から、fetchがネットワークに接続できない。

- network_monitorが喋り続ける
- ping が通り続けない

デフォルトのネットワーク設定 `sanshiro-73B2` が落ちているので起動しようとすると以下のエラーを得た

```bash
fetch@fetch1075:~$ sudo nmcli c up sanshiro-73B2
[sudo] password for fetch: 
Error: Connection activation failed: The Wi-Fi network could not be found
```

fetch1075から見えているsanshiroのアクセスポイントは以下

```bash
fetch@fetch1075:~/Desktop$ sudo iwlist scan | grep -e sanshiro -e Address | grep sanshiro -B 1
eth0      Interface doesn't support scanning.

eth1      Interface doesn't support scanning.

lo        Interface doesn't support scanning.

          Cell 03 - Address: C8:84:A1:F0:D8:E2
                    ESSID:"sanshiro"
--
          Cell 15 - Address: 70:EA:1A:EF:9E:92
                    ESSID:"sanshiro"
--
          Cell 25 - Address: C8:84:A1:F0:9A:42
                    ESSID:"sanshiro"
--
          Cell 39 - Address: C8:84:A1:F0:D8:ED
                    ESSID:"sanshiro"
--
          Cell 01 - Address: C8:84:A1:F0:D8:ED
                    ESSID:"sanshiro"
--
          Cell 23 - Address: C8:84:A1:F0:9A:4D
                    ESSID:"sanshiro"
```

この中には`sanshiro-73B2`で設定されている`BC:54:2F:AC:7F:6A`は確かに無い。

どうやら73B2の目の前のアクセスポイントが死んでいるらしい。

アクセスポイントを特定の場所に固定しない `sanshiro-outside` であればネットワークがつながる。

```bash
fetch@fetch1075:~/Desktop$ sudo nmcli c up sanshiro-outside
Connection successfully activated (D-Bus active path: /org/freedesktop/NetworkManager/ActiveConnection/6)
```

sktometometo, 2022-07-04:
https://github.com/knorth55/jsk_robot/pull/279 でデフォルトで使用するべきネットワークを`sanshiro-outside`に設定.

注意点としてはネットワークマネージャー側でデフォルトで使用するネットワーク設定は `sanshiro-73B2` のままなので、このプルリクエストをマージした後の挙動としては

- ロボットのPCが起動
- `sanshiro-73B2` への接続を試行して失敗
- `network_monitor.py` がネットワーク接続の不調を検知してネットワーク設定を `sanshiro-outside` で起動
- ネットワークへ接続

となること。ネットワークマネージャー側の設定はgit管理されておらず復旧時に間違える可能性があるので今回は変更しない。

708yamaguchi, 2022-07-15:
最近fetch2台ともの無線の調子が悪かったですが、おそらく原因を見つけたので関連しているこのissueに書きます。

最近はこのissueをうけて、fetch15に加えてfetch1075でもアクセスポイントを指定せずにsanshiroを使う設定(sanshiro-outside)を使っていたと思います。
さきほどsanshiro-outsideを見ていたところ、`band=a`が抜けていることを発見しました。

この設定は、3ヶ月ほど前にfetch15の無線遅い問題が発生していたときに、岡田先生にアドバイスしてもらいsanshiroの5GHz帯にしか繋がらないようにしていたものです。
https://github.com/jsk-ros-pkg/jsk_robot/issues/1473#issue-1214068387

今はこの設定を追記したことで再び5GHz帯のみに接続するようになり、通信速度も多少は改善していると思います。
見落とさないように、コメントも書いておきました。
```
[wifi]
# 2022/07/15 yamaguchi
# Use only 5GHz
band=a
```

708yamaguchi, 2022-08-17:
これ最近も発生しています。
5GHz帯に接続されているのは確認しています。

調べていないですが、fetch1075で多発しているような気がします。

sktometometo, 2022-08-18:
原因はわからないけど、`sanshiro-73B2` プロファイルを、73B2内のアクセスポイントで設定して復活してもいいかも。
fetch15でのアクセスポイントの状態

```bash
~ $ sudo iwlist scan | grep sanshiro -B 5 -A 5
lo        Interface doesn't support scanning.

eth0      Interface doesn't support scanning.

eth1      Interface doesn't support scanning.

          Cell 01 - Address: 78:72:5D:B6:CC:A9
                    Channel:116
                    Frequency:5.58 GHz (Channel 116)
                    Quality=61/70  Signal level=-49 dBm  
                    Encryption key:on
                    ESSID:"sanshiro"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000000a8c336522b
                    Extra: Last beacon: 2252ms ago
--
          Cell 18 - Address: C8:84:A1:F0:9A:49
                    Channel:64
                    Frequency:5.32 GHz (Channel 64)
                    Quality=28/70  Signal level=-82 dBm  
                    Encryption key:on
                    ESSID:"sanshiro"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000001224e9a9c88
                    Extra: Last beacon: 3016ms ago
--
          Cell 22 - Address: 78:72:5D:96:5D:79
                    Channel:100
                    Frequency:5.5 GHz (Channel 100)
                    Quality=26/70  Signal level=-84 dBm  
                    Encryption key:on
                    ESSID:"sanshiro"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000000a8c378757c
                    Extra: Last beacon: 3012ms ago
--
          Cell 37 - Address: 78:72:5D:B6:CC:A6
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=-34 dBm  
                    Encryption key:on
                    ESSID:"sanshiro"
                    Bit Rates:5.5 Mb/s; 6 Mb/s; 9 Mb/s; 11 Mb/s; 12 Mb/s
                              18 Mb/s; 24 Mb/s; 36 Mb/s
                    Bit Rates:48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000000a8c35c2022
--
          Cell 58 - Address: 70:EA:1A:EF:9E:99
                    Channel:116
                    Frequency:5.58 GHz (Channel 116)
                    Quality=25/70  Signal level=-85 dBm  
                    Encryption key:on
                    ESSID:"sanshiro"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000001226e0c5766
                    Extra: Last beacon: 2248ms ago
```

sktometometo, 2022-08-18:
Fetch15 と Fetch1075 の `sanshiro-73B2` 設定のBSSID を `78:72:5D:B6:CC:A9` に設定し、起動時に`sanshiro-outside` ではなく　`sanshiro-73B2` を使用するようにして、かつ https://github.com/knorth55/jsk_robot/pull/321 で config.bash を更新して `sanshiro-73B2` を使用するように変更しました。

708yamaguchi, 2022-08-18:
ありがとう、ただ今はそのBSSIDに接続しているにも関わらず速度が遅いですね
```
fetch@fetch1075:~$ iwconfig
wlan1     IEEE 802.11  ESSID:"sanshiro"  
          Mode:Managed  Frequency:5.58 GHz  Access Point: 78:72:5D:B6:CC:A9   
          Bit Rate=150 Mb/s   Tx-Power=17 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=54/70  Signal level=-56 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:6  Invalid misc:52   Missed beacon:0

eth0      no wireless extensions.

eth1      no wireless extensions.

wlan0     IEEE 802.11bgn  ESSID:"fetch1075"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:2.412 GHz  Access Point: C0:06:C3:1B:19:81   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

fetch@fetch1075:~$ ping 8.8.8.8fetch@fetch1075:~$ ping 8.8.8.8
```
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=115 time=286 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=115 time=301 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=115 time=229 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=115 time=419 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=115 time=324 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=115 time=254 ms
```

708yamaguchi, 2022-08-18:
デバッグのために、普段はアクセスポイント用WiFiドングルを抜いておくのはどうでしょうか。

sktometometo, 2022-08-18:
デバッグのために行う分には全然構わないし、USBドングルを抜いた結果通信速度が早くなるという現象が確認できるのであれば常時抜いておいてもらって構わない。

sktometometo, 2022-08-18:
ただ、わかりやすい場所においておいてもらえると助かる。

708yamaguchi, 2022-08-18:
了解です、ありがとう

保管場所ですが、金沢くんの席の上のfetch棚っぽいところに入れておいて「fetchアクセスポイント用USBドングル」みたいなシールを貼っておきます。

sktometometo, 2022-08-18:
まずそのデバッグのためには定期的に通信速度を計測して記録するプログラムが必要か。

708yamaguchi, 2022-08-18:
個人的にはそこまで厳密にしなくても、pingが100msを超える状態が続いたらみんな気づくはずで、その時にまた報告するくらいでいいかなと思ってる。

ちなみにログ自体はとってる
ただjsk-log-wifiは動いてないかも

```
fetch@fetch1075:~$ tail -n 5 /var/log/ros/jsk-network-monitor.log
PING google.com (172.217.26.238) 56(84) bytes of data.

--- google.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 106.522/106.522/106.522/0.000 ms
```

```
fetch@fetch1075:~$ tail -n 5 /var/log/ros/jsk-log-wifi.log.1
2022-05-07T20:59:57+09:00, sanshiro, 5500,49.,-61.,-256, 200.0 MBit/s VHT-MCS 9 40MHz short GI VHT-NSS 1
2022-05-07T21:00:07+09:00, sanshiro, 5500,49.,-61.,-256, 180.0 MBit/s VHT-MCS 8 40MHz short GI VHT-NSS 1
2022-05-07T21:00:17+09:00, sanshiro, 5500,49.,-61.,-256, 200.0 MBit/s VHT-MCS 9 40MHz short GI VHT-NSS 1
2022-05-07T21:00:27+09:00, sanshiro, 5500,49.,-61.,-256, 200.0 MBit/s VHT-MCS 9 40MHz short GI VHT-NSS 1
2022-05-07T21:00:37+09:00, sanshiro, 5500,49.,-61.,-256, 180.0 MBit/s VHT-MCS 8 40MHz short GI VHT-NSS 1
```

708yamaguchi, 2022-08-18:
jsk-log-wifiは以下のプルリクエストで修正しました。
https://github.com/knorth55/jsk_robot/pull/323

なのでモニタリング自体は常時出来ている状態になりました。

708yamaguchi, 2022-08-18:
2022/08/18 18:00くらいからUSBドングルを外しています。
通信速度が遅い状態が多発したら、また教えてください。

sktometometo, 2022-08-18:
比較のためには両方の状態が必要では

708yamaguchi, 2022-08-18:
wifi速度を測定するスクリプト書いたので、比較を手伝ってもらえると助かります。

USBドングルはデフォルトではついているものではないので、つける場合は副作用がないことまで確認してもらえると助かります。

708yamaguchi, 2022-08-18:
> 比較のためには両方の状態が必要では、それ無しでいきなり外されると、流石に外したかったから外したんじゃってつっこまなくちゃならなくなる。

そもそも最初にUSBドングルをつけるときに比較が必要だったのでは、とも思います。

sktometometo, 2022-08-18:
それ言い出したらPCから何までもとに戻さないと話通らないでしよ。

knorth55, 2022-08-19:
よくわかってないんだけど、問題と現状なにをチェックしたのかのスライドはあるのかな？
どれくらいの頻度で起きる問題で、どれくらいの重要性があるのかな？
SANSHIROが部屋の中に置かれたけど、それでもだめだったのかな？
USBのWifiドングルが原因かもしれないという話はどこからでてきたのかな？
頭にあるUSBのハブを給電していたものだけの１つにしたけど、それでもだめだったのかな？

708yamaguchi, 2022-08-19:
すいません、僕の勘違いだと思うので、wifiドングルを戻しておきます
問題の重要度は低いので、大丈夫です

knorth55, 2022-08-19:
なるほど了解です。
情報共有のためにスライドを簡単にでもつくって共有してくれると嬉しいです。
重要度が低いというのはそこまで頻発しなくなったって理解であってますか？

708yamaguchi, 2022-08-19:
無線問題は再現性を取るのが難しく、上に書いたpingが遅くなることがある程度しか書けないです。

重要度が低いというのは、問題が発生しない日はロボットを使えるので問題ないという理解です。

knorth55, 2022-08-19:
これは日単位でおきることなのかな？再起動したら解決したりするのかな？

> 無線問題は再現性を取るのが難しく、上に書いたpingが遅くなることがある程度しか書けないです。

CPU使用率はどんな感じだろう？
mongodbのreplicationが裏で走っていて通信帯域を食っていたりしないかな？

708yamaguchi, 2022-08-19:
問題の頻度はわからないですが、おそらく一時的にpingが遅かっただけで僕の勘違いでした。

今度問題が発生したら、CPU使用率なども含め報告します

sktometometo, 2022-08-22:
https://github.com/knorth55/jsk_robot/pull/325 で可視化してデバック中。
以下の点について確認したい

- 一日のうち特定のタイミングで遅くなっていないか
- 特定の曜日のみ遅くなっていないか
- 特定のデバイスを使用しているときのみ遅くなっていないか

sktometometo, 2022-08-23:
Wi-Fiドングルを外す際は `NETWORK_DEFAULT_WIFI_INTERFACE` の設定の変更が必要になります。
参考: https://github.com/knorth55/jsk_robot/blob/eb8dad91c2158f774ec9cc6adbd50a8f9e8d6156/jsk_fetch_robot/jsk_fetch_startup/config/config.bash#L43

現在はデバッグで再びWi-Fiドングル取り付け中のため、変更は不要です。

sktometometo, 2022-08-23:
今後の実験予定

- 8/23: PCI WiFi をsanshiroへ接続、Wi-Fiドングルをアクセスポイントに設定
- 8/24: PCI WiFi をsanshiroへ接続、Wi-Fiドングルを取り外す
- 8/25: PCI WiFi をアクセスポイントに設定、Wi-Fiドングルをsanshiroに接続
- 8/26: PCI WiFi を無効化、Wi-Fiドングルをsanshiroへ接続

knorth55, 2022-09-11:
停電の影響か，sanshiroの73B2内のアクセスポイントが点滅（エラー？）をしていて，sanshiroにつながりません．
とりあえず今日は様子見ということで一旦放置しています．

cc. @mqcmd196 @tkmtnt7000 @softyanija

knorth55, 2022-09-11:
>8/23: PCI WiFi をsanshiroへ接続、Wi-Fiドングルをアクセスポイントに設定
8/24: PCI WiFi をsanshiroへ接続、Wi-Fiドングルを取り外す
8/25: PCI WiFi をアクセスポイントに設定、Wi-Fiドングルをsanshiroに接続
8/26: PCI WiFi を無効化、Wi-Fiドングルをsanshiroへ接続

@sktometometo そういえばこの比較って結果はどうなったんですか？

sktometometo, 2022-09-11:
これデータだけ取ってまだ解析してないです。PSJ一段落したのでやります

mqcmd196, 2022-09-12:
> 停電の影響か，sanshiroの73B2内のアクセスポイントが点滅（エラー？）をしていて，sanshiroにつながりません．
> とりあえず今日は様子見ということで一旦放置しています．

こちらは担当者の方に対応していただき解決しました．

knorth55, 2022-09-13:
Fetch1075で`sanshiro-73B2`をつかっていて、BSSIDが`78:72:5D:B6:CC:A9`に固定されているのはなんでだろう？
電波強度的には`78:72:5D:B6:CC:A6`のほうが強そうだけど、typo？

```
fetch@fetch1075:~$ nmcli -f in-use,ssid,bssid,signal,bars dev wifi
IN-USE  SSID                  BSSID              SIGNAL  BARS 
        fetch1075             C0:06:C3:1B:19:81  100     ▂▄▆█ 
        JSK-JAXON-sub         E4:F4:C6:19:A3:5E  99      ▂▄▆█ 
        pr1040LAN             00:90:FE:FD:F2:5D  95      ▂▄▆█ 
        UTokyo-Guest          78:72:5D:B6:CC:AC  94      ▂▄▆█ 
        pr1040WAN             E4:8D:8C:6B:E4:60  94      ▂▄▆█ 
        eng.ut                78:72:5D:B6:CC:A0  92      ▂▄▆█ 
        sanshiro              78:72:5D:B6:CC:A6  92      ▂▄▆█ 
        mercury               26:DF:B6:3F:30:E0  92      ▂▄▆█ 
        Buffalo-G-2650        C4:3C:EA:3B:26:52  90      ▂▄▆█ 
        ist_members           78:72:5D:B6:CC:A7  90      ▂▄▆█ 
        eng.ut-1x             78:72:5D:B6:CC:A5  90      ▂▄▆█ 
        UTokyo-WiFi           78:72:5D:B6:CC:AD  90      ▂▄▆█ 
        bld2-guest            78:72:5D:B6:CC:A4  89      ▂▄▆█ 
        eduroam               78:72:5D:B6:CC:AB  89      ▂▄▆█ 
        eduroam               78:72:5D:B6:CC:A4  87      ▂▄▆█ 
        ist_members           78:72:5D:B6:CC:A8  85      ▂▄▆█ 
        bld2-guest            78:72:5D:B6:CC:AB  85      ▂▄▆█ 
        UTokyo-WiFi           78:72:5D:B6:CC:A2  85      ▂▄▆█ 
        eng.ut-1x             78:72:5D:B6:CC:AA  85      ▂▄▆█ 
        UTokyo-Guest          78:72:5D:B6:CC:A3  85      ▂▄▆█ 
        0001softbank          78:72:5D:B6:CC:A6  85      ▂▄▆█ 
        eng.ut                78:72:5D:B6:CC:AF  85      ▂▄▆█ 
        DN1002O3S2O5M2QM74MM  00:0E:8E:95:29:60  79      ▂▄▆_ 
*       sanshiro              78:72:5D:B6:CC:A9  67      ▂▄▆_ 
        KengoroLAN-5G         74:03:BD:16:54:85  62      ▂▄▆_ 
        COMS-W-5G             74:03:BD:16:79:55  59      ▂▄▆_ 
        AraiLab_73C1_5G       B0:6E:BF:9E:BE:AC  29      ▂___ 
        --                    24:4B:FE:C5:78:D8  29      ▂___ 

```

knorth55, 2022-09-13:
あ、けど強度がつよいだけで、73B2にあるのは`78:72:5D:B6:CC:A9`のほうですね。

sktometometo, 2022-09-14:
5GHzと2.4GHzじゃなかったかな

sktometometo, 2022-09-14:
5GHzのアクセスポイントに固定してます