jsk-ros-pkg/jsk_robot #1521: `BC:54:2F:AC:7F:6A` のアクセスポイントが死んでいる

sktometometo, 2022-07-04:
Fetchでデフォルトで接続するようにしている `BC:54:2F:AC:7F:6A` のsanshiroアクセスポイントが死んでいる.

Fetchでの症状はhttps://github.com/jsk-ros-pkg/jsk_robot/issues/1520を参照

sktometometo, 2022-07-04:
73B2のテーブル前からのwifiスキャン結果は以下

```bash
~/Desktop $ sudo iwlist scan | grep "Cell" -A 11
          Cell 01 - Address: C0:06:C3:1B:19:81
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=60/70  Signal level=-50 dBm  
                    Encryption key:on
                    ESSID:"fetch1075"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000053cdc899
                    Extra: Last beacon: 4136ms ago
--
          Cell 02 - Address: A2:64:8F:78:A0:58
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=53/70  Signal level=-57 dBm  
                    Encryption key:on
                    ESSID:"DIRECT-SR5900PA21F1C"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000041209a97034
                    Extra: Last beacon: 4132ms ago
                    IE: Unknown: 00144449524543542D53523539303050413231463143
--
          Cell 03 - Address: C0:06:C3:1B:19:7C
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=52/70  Signal level=-58 dBm  
                    Encryption key:on
                    ESSID:"KI00070"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000000e9c65ee713
                    Extra: Last beacon: 4132ms ago
--
          Cell 04 - Address: C8:84:A1:F0:D8:E0
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=62/70  Signal level=-48 dBm  
                    Encryption key:on
                    ESSID:"eng.ut"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=00000016bcbf7c93
                    Extra: Last beacon: 4136ms ago
--
          Cell 05 - Address: C8:84:A1:F0:D8:E1
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=61/70  Signal level=-49 dBm  
                    Encryption key:on
                    ESSID:"eduroam"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=00000016bcbf921d
                    Extra: Last beacon: 4136ms ago
--
          Cell 06 - Address: C8:84:A1:F0:D8:E2
[... 494 lines cut ...]
                    Channel:132
                    Frequency:5.66 GHz (Channel 132)
                    Quality=27/70  Signal level=-83 dBm  
                    Encryption key:on
                    ESSID:"bld2-guest"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000001643cb6c56
                    Extra: Last beacon: 1456ms ago
                    IE: Unknown: 000A626C64322D6775657374
--
          Cell 45 - Address: E4:8D:8C:6B:E4:60
                    Channel:149
                    Frequency:5.745 GHz
                    Quality=70/70  Signal level=-29 dBm  
                    Encryption key:off
                    ESSID:"pr1040WAN"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=00000ad470c489e0
                    Extra: Last beacon: 1052ms ago
                    IE: Unknown: 000970723130343057414E
```

sktometometo, 2022-07-04:
同じCiscoのハードウェアなら共有していそうな `BC:54:2F:??:??:??` のアクセスポイントは無い

708yamaguchi, 2022-07-15:
最近fetch2台ともの無線の調子が悪かったですが、おそらく原因を見つけたので関連しているこのissueに書きます。

最近はこのissueをうけて、fetch15に加えてfetch1075でもアクセスポイントを指定せずにsanshiroを使う設定(sanshiro-outside)を使っていたと思います。
さきほどsanshiro-outsideを見ていたところ、`band=a`が抜けていることを発見しました。

この設定は、3ヶ月ほど前にfetch15の無線遅い問題が発生していたときに、岡田先生にアドバイスしてもらいsanshiroの5GHz帯にしか繋がらないようにしていたものです。
https://github.com/jsk-ros-pkg/jsk_robot/issues/1473#issue-1214068387

今はこの設定を追記したことで再び5GHz帯のみに接続するようになり、通信速度も多少は改善していると思います。
見落とさないように、コメントも書いておきました。
```
[wifi]
# 2022/07/15 yamaguchi
# Use only 5GHz
band=a
```

708yamaguchi, 2022-07-15:
見落としていたのですが、https://github.com/jsk-ros-pkg/jsk_robot/issues/1520 のissueの方が関連度が高そうだったので、先ほどのコメントを消してhttps://github.com/jsk-ros-pkg/jsk_robot/issues/1520#issuecomment-1185865094 に書き直しました。

sktometometo, 2022-09-15:
この問題、おそらく解決したのでCloseします。

closed 2022-09-15 by sktometometo