jsk-ros-pkg/jsk_robot #1530: rosinstall is not found.

soonhyo, 2022-07-15:
I am using myCobot 280 M5 and tried to use jsk_mycobot_robot in this Pull Request https://github.com/jsk-ros-pkg/jsk_robot/pull/1421 .

But https://raw.githubusercontent.com/jsk-ros-pkg/jsk_robot/master/jsk_mycobot_robot/mycobot.rosinstall in READEME.md does not exist.

cc @sktometometo

pazeshun, 2022-07-15:
Use https://raw.githubusercontent.com/tongtybj/jsk_robot/PR/mycobot/eus/jsk_mycobot_robot/mycobot.rosinstall
URL in README becomes available after #1421 is merged into master.

soonhyo, 2023-01-24:
It worked with the rosinstall file. But It needed to change like this,
```
# THIS IS AN AUTOGENERATED FILE, LAST GENERATED USING wstool ON 2023-01-20
- git:
    local-name: jsk_robot
    uri: https://github.com/tongtybj/jsk_robot.git
    version: PR/mycobot/eus
# the repository of mycobot driver
# this repositoy is under construction now
# so we need to use the pathed branch instead of the master branch
# PR: https://github.com/elephantrobotics/mycobot_ros/pull/48
- git:
    local-name: mycobot_ros
    uri: https://github.com/tongtybj/mycobot_ros.git
    version: interface
```

closed 2023-01-24 by soonhyo