jsk-ros-pkg/jsk_robot #1546: [fetch15] 頭部ASUSカメラが見えない

708yamaguchi, 2022-07-23:
fetch15体内で
```
rostopic hz /head_camera/rgb/image_raw -w5
```
しても画像がこず、
```
sudo supervisorctl stop jsk-fetch-startup
sudo supervisorctl stop robot
python usb_reset.py
sudo supervisorctl start robot
```
をしたのですが結局カメラ画像が見えることはありませんでした。

usb_reset.pyのソースコード
https://gist.github.com/708yamaguchi/ea46ca863c288659e94bbecbaf9eadf4

usb_reset.pyのログ
この作業をする最中、常にASUSカメラはlsusbで認識されており、usb_reset.pyも正しくリセットを出来ていたように思います。
```
fetch@fetch15:~$ python usb_reset.py 
Bus 005 Device 006: ID 18d1:9302 Google Inc. 
Bus 005 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 005 Device 013: ID 8087:0b37 Intel Corp. 
Bus 005 Device 005: ID 18d1:9302 Google Inc. 
Bus 005 Device 004: ID 8086:0b64 Intel Corp. 
Bus 005 Device 002: ID 2109:0817 VIA Labs, Inc. 
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 013: ID 2886:0018  
Bus 004 Device 011: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 004 Device 009: ID 1130:1620 Tenx Technology, Inc. 
Bus 004 Device 006: ID 2109:2817 VIA Labs, Inc. 
Bus 004 Device 005: ID 2e1a:1000  
Bus 004 Device 004: ID 222a:004d  
Bus 004 Device 012: ID 8087:0aa7 Intel Corp. 
Bus 004 Device 010: ID 26ce:01a2  
Bus 004 Device 008: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 004 Device 002: ID 0d8c:0126 C-Media Electronics, Inc. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1d27:0609 ASUS 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

^^^^^^^^^^^^^^^^^^^^
Bus 001 Device 002: ID 1d27:0609 ASUS 
001
/dev/bus/usb/001/002
rosrun openni2_camera usb_reset /dev/bus/usb/001/002
Resetting USB device /dev/bus/usb/001/002
Reset successful
```

openni2_cameraのエラー
```
[ INFO] [1658560551.724646020] [/head_camera/head_camera_nodelet_manager:ros.openni2_camera]: Device "1d27/0609@1/2" found.                                                                                 
Warning: USB events thread - failed to set priority. This might cause loss of data...                                                                                                                       
[ INFO] [1658560551.761208318] [/head_camera/head_camera_nodelet_manager:ros.openni2_camera]: No matching device found.... waiting for devices. Reason: openni2_wrapper::OpenNI2Device::OpenNI2Device(const 
string&) @ /tmp/binarydeb/ros-melodic-openni2-camera-1.5.1/src/openni2_device.cpp @ 75 : Device open failed                                                                                                 
        Could not open "1d27/0609@1/2": Device is in safe mode. Cannot start any stream! 
```

/var/log/ros/robot.logのログです。
https://gist.github.com/708yamaguchi/3c11a6292ab584242384ea33c6af2a49

iory, 2022-07-23:
fetchのカメラが消える問題を調べるため、山口くんと次のような手順を行いました。

```
$ lspci
00:00.0 Host bridge: Intel Corporation Device 9b53 (rev 03)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 03)
00:02.0 VGA compatible controller: Intel Corporation Device 9bc8 (rev 03)
00:12.0 Signal processing controller: Intel Corporation Device 06f9
00:14.0 USB controller: Intel Corporation Device 06ed
00:14.2 RAM memory: Intel Corporation Device 06ef
00:16.0 Communication controller: Intel Corporation Device 06e0
00:17.0 SATA controller: Intel Corporation Device 06d2
00:1c.0 PCI bridge: Intel Corporation Device 06ba (rev f0)
00:1c.3 PCI bridge: Intel Corporation Device 06bb (rev f0)
00:1d.0 PCI bridge: Intel Corporation Device 06b0 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device 0684
00:1f.3 Audio device: Intel Corporation Device 06c8
00:1f.4 SMBus: Intel Corporation Device 06a3
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device 06a4
00:1f.6 Ethernet controller: Intel Corporation Device 0d4d
01:00.0 PCI bridge: Pericom Semiconductor PI7C9X111SL PCIe-to-PCI Reversible Bridge (rev 02)
02:04.0 USB controller: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller (rev 62)
02:04.1 USB controller: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller (rev 62)
02:04.2 USB controller: VIA Technologies, Inc. USB 2.0 (rev 65)
03:00.0 Network controller: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak] (rev 10)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. Device 8125 (rev 05)
05:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981
```
lspciでxtionがつながっているpciを確認。`01:00.0 PCI bridge: Pericom Semiconductor PI7C9X111SL PCIe-to-PCI Reversible Bridge (rev 02)`
この状態で`lsusb`をして`ASUS`が認識されていることを先に確認。

```
Bus 002 Device 006: ID 18d1:9302 Google Inc.
Bus 002 Device 004: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 003: ID 18d1:9302 Google Inc.
Bus 002 Device 005: ID 8086:0b64 Intel Corp.
Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 03e7:2150
Bus 001 Device 012: ID 2886:0018
Bus 001 Device 010: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 008: ID 1130:1620 Tenx Technology, Inc.
Bus 001 Device 005: ID 2109:2817 VIA Labs, Inc.
Bus 001 Device 004: ID 2e1a:1000
Bus 001 Device 003: ID 222a:004d
Bus 001 Device 011: ID 8087:0aa7 Intel Corp.
Bus 001 Device 009: ID 26ce:01a2
Bus 001 Device 007: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 002: ID 0d8c:0126 C-Media Electronics, Inc.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 1d27:0609 ASUS
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

次に対象のPCIをremoveする。

```
sudo bash -c 'echo "1" > /sys/bus/pci/devices/0000\:01:00.0/remove'
```

この状態で`lsusb`を見て`ASUS`が消えることを確認。

```
Bus 002 Device 006: ID 18d1:9302 Google Inc.
Bus 002 Device 004: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 003: ID 18d1:9302 Google Inc.
Bus 002 Device 005: ID 8086:0b64 Intel Corp.
Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 03e7:2150
Bus 001 Device 012: ID 2886:0018
Bus 001 Device 010: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 008: ID 1130:1620 Tenx Technology, Inc.
Bus 001 Device 005: ID 2109:2817 VIA Labs, Inc.
Bus 001 Device 004: ID 2e1a:1000
Bus 001 Device 003: ID 222a:004d
Bus 001 Device 011: ID 8087:0aa7 Intel Corp.
Bus 001 Device 009: ID 26ce:01a2
Bus 001 Device 007: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 002: ID 0d8c:0126 C-Media Electronics, Inc.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

このあとに

```
sudo bash -c 'echo "1" > /sys/bus/pci/rescan'
```
を行うと再び`ASUS`が再接続される。

`fetch1075`だとrescanしたあとに`roslaunch openni2_launch openni2.launch`をすると
`/camera/depth_registered/points`が出力される。
また、新たに予備のXtionを電源ボタン上のUSB3.0につないでもxtionが起動することが確認できた。

fetch15においても予備のXtionを電源のUSB3.0にさすと起動を確認。これでおそらくsoftware的な問題で起動しない可能性は低くなったと思われる。
fetch15頭部のl515が刺さっていたUSBハブに指しても起動しpointcloudが出力されることを確認。

次にfetch15でカメラが起動しない問題が起きた時に、抜き差しして治るかどうか確認することで電源供給の理由で再度復活できるかの確認を行う。

708yamaguchi, 2022-09-29:
この問題が最近もfetch15で発生しました。

#### 症状

`/head_camera/rgb/image_rect_color`が見えていなくてtwitterが更新されていないことに@nakane11 が気づきました
毎日深夜4時に`$ sudo reboot`しているはずなのですが、ASUSカメラが見えないままでした。

#### 試したこと

ベース側面の電源ボタンを押してリブートしたところ、`/head_camera/rgb/image_rect_color`が見えるようになりました。

`$ sudo reboot`ではマザーボードに接続されているUSBデバイスへの電源が供給されたままになるのですが、電源ボタンによるリブートだとUSBデバイスへの電源供給を断つことができます。（マザーボードの設定が悪い？）
これによりカメラデバイスがリセットされて画像が出力されるようになったのかな、と思っています。

#### つぎに試したいこと

このissueの存在を忘れており、

> 次にfetch15でカメラが起動しない問題が起きた時に、抜き差しして治るかどうか確認することで電源供給の理由で再度復活できるかの確認を行う。

をしていなかったので、これを試します。

もうちょっとマザーボードの設定を調べてからにはなると思いますが、fetch15に入っているマザーボードと同じマザーボードがJSKに余っていると思うので、それを使って自作PCを作ってみて`$ sudo reboot`時にUSBデバイスへの電源供給が断てるかを調べる。

tkmtnt7000, 2022-11-22:
この問題が最近(2022/11/18)もFetch15で発生しました．

このissueの存在を忘れていたので`usb_reset.py` を試していません．

**やったこと**
1. 電源ボタンからrebootして電源供給を断つ ==> 頭部ASUSカメラは起動しない
2. 電源を落とした状態でFetchのお腹を開けてカメラのUSB端子を抜き差し ==> 頭部ASUSカメラが見えるようになった

**今回の事象から考えられる原因**
1. 接触不良
2. ASUSカメラ自体の経年劣化による故障
3. その他

Fetch15同様に`sudo reboot`のみで毎日再起動されているFetch1075で同じようにカメラが見えない問題が起こっていないとするとハードウェアの問題も考えられそう

**応急処置できそうなこと**
1. `usb_reset.py`みたいなものを入れる
2. https://www.google.com/search?q=usb+%E3%82%B9%E3%82%A4%E3%83%83%E3%83%81%E4%BB%98%E3%81%8D%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB&sxsrf=ALiCzsZpi-h1BmBq1LTnDLiuTfIWJ6d99g:1668850121415&source=lnms&tbm=shop&sa=X&ved=2ahUKEwimga6V97n7AhU_sVYBHXlZBd0Q_AUoAXoECAIQAw&biw=1232&bih=794&dpr=2  みたいなものでUSBオンオフできるようにしておく