jsk-ros-pkg/jsk_robot #1550: [fetch1075] L515からカメラ画像がでないことがある

708yamaguchi, 2022-07-29:
本日L515カメラの接続をUSB3にしたと思うのですが、以下のようにUSB接続が一時的に切れて再接続したときにUSB2として認識されるようです。
（人間がUSBケーブルを抜いたわけではなく、勝手に再接続されていた）

L515がUSB3として認識されているときとUSB2では利用できる解像度が異なります。
なのでrealsense nodeを立ち上げなおしてもL515がUSB2として認識され続ける限り、USB3用の解像度を設定したrealsense nodeからはカメラ画像がpublishされないです。

/var/log/ros/robot.logの抜粋
```
[ WARN] [1659104317.477222880] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: No RealSense devices were found!
[ WARN] [1659104323.773941379] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: No RealSense devices were found!
[ INFO] [1659104329.783194726] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]:  
[ WARN] [1659104329.967137124] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: Device f0232270 is connected using a 2.1 port. Reduced performance is expected.
[ INFO] [1659104329.966496378] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: Device with serial number f0232270 was found.
[ INFO] [1659104329.966534032] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.3/4-2.3:1.0/video4linux/video0 was found.
[ INFO] [1659104329.966546917] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: Device with name Intel RealSense L515 was found.
[ INFO] [1659104329.967053350] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: Device with port number 4-2.3 was found.
[ INFO] [1659104329.967103727] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: Device USB type: 2.1
[ INFO] [1659104329.967187464] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]: Resetting device...
[ INFO] [1659104335.980614464] [/l515_head/l515_head_realsense2_camera:ros.realsense2_camera]:  
```

cc @knorth55
baxterで同様の問題が起きたときはUSBを抜き差ししていると聞きました。
結局はUSB抜き差しが解決策ということですかね。

knorth55, 2022-07-29:
追記すると，7時ごろに作業を終えて再起動して，全て起動していたのですが，9時頃に突然頭周りのUSBデバイスが全て一時的に接続が落ちてしまっていて，上の現象はこれが原因ではないかと考えています．
この接続がおちたあとには，
- RealsenseはUSB2.0で接続される
- Respeakerはつながらない
- Insta360はlibuvcからつながらない
といった現象がおきていて，再起動やケーブルの挿し直しをすれば治るという状況でした．

いま原因を究明するために，頭のスピーカーとアクセスポイントがついていたUSBハブを一旦取り外して様子をみています．
このUSBハブが挙動が怪しいという話は @tkmtnt7000 からでていたので，それをまず検証しようと思っています．
この構成で同じような現象がでるかどうかを検証しています．

次に検証したらいいと思っているのが，Realsenseの画質を少し下げても同じ現象がおきるのかです．

708yamaguchi, 2022-07-29:
補足ありがとうございます。

なので今のfetch1075の頭部状況は以下のようになっています。
ベースから頭部に伸ばしてきたUSB3.0ケーブルにセルフパワーUSB3.0ハブがついていて、そのハブに
- L515
- insta360
- respeaker
- PS3 controller

の4つのデバイスが接続されています。