jsk-ros-pkg/jsk_robot #1551: [Fetch1075] スピーカーからの出力が出ない。

sktometometo, 2022-08-02:
喋ったり音出したりしなくなっているっぽい。

sktometometo, 2022-08-02:
jsk-fetch-startup.conf に直書きされている AUDIO_DEVICE のせいかな？

708yamaguchi, 2022-08-02:
なるほど、たしかに
今はL515の検証のために頭部スピーカを外していました。
ここ数日くらいはL515は安定して動いているように見えるので、スピーカをつけてもいいかもですね。
https://github.com/jsk-ros-pkg/jsk_robot/issues/1550

僕の机の上にスピーカがあるので、必要なら付けてもらっても大丈夫です。
必要ないなら後で僕が付けておきます。

knorth55, 2022-08-02:
Fetch1075は頭部のスピーカやアクセスポイントがついていたUSBハブを外したらL515は問題なく動いていそうですね．
USBハブをちゃんとしたものを買って試してみてもいいかもしれませんね．

708yamaguchi, 2022-08-02:
USBハブを別のものに取り換えて，頭部にスピーカとアクセスポイントを接続しました．（USBハブは薄くてUSB3.0が4ポートあるもの）

気づいたこととして，USBハブの接続パターンによってスピーカが使えるかが変わりそうです．

接続パターン1
- セルフパワーUSBハブ -> バスパワーUSBハブ -> スピーカと接続
- fetchを電源ボタンを押して完全にシャットダウンしてから起動したときにスピーカが起動しない（スピーカのオレンジ色のLEDが点灯せず，fetchも話さない）
- スピーカが起動している状態でsudo rebootすると，スピーカは起動したまま

接続パターン2
- セルフパワーUSBハブ -> スピーカと接続
- fetchを電源ボタンから完全にシャットダウンしてから起動しても，スピーカは起動する

今は接続パターン2の状態にしています．
またfetch1075が話さなくなったら，ここに書いてもらえると助かります．

iory, 2022-09-15:
@nakane11 と最近頭部のスピーカーの給電が頻繁に落ちているということで、別のスピーカーを取り付けてみました。
こちらはUSBによる給電とイヤホンジャックで接続できるもので、イヤホンジャックをrespeakerに挿しています。
これで様子見をしてみましょう。

[image]

708yamaguchi, 2022-09-15:
fetchは立ち上げ時に使用するスピーカを選択するようにしています。
https://github.com/jsk-ros-pkg/jsk_robot/blob/5da39b42196f42c69b7073d57b75e5b71bf8c091/jsk_fetch_robot/jsk_fetch_startup/supervisor_scripts/jsk-fetch-startup.conf#L10

呼んでいるのはここで、デバイス名の取得方法もコメントに書いているので試してもらえると助かります。
https://github.com/jsk-ros-pkg/jsk_robot/blob/5da39b42196f42c69b7073d57b75e5b71bf8c091/jsk_fetch_robot/jsk_fetch_startup/scripts/setup_audio.bash

knorth55, 2022-09-15:
お、いいね、耳ができた（スピーカーだけど）