jsk-ros-pkg/jsk_robot #1553: [fetch] L515カメラから得られる点群の向きがおかしい

708yamaguchi, 2022-08-04:
@sktometometo 

以下のように、fetch15が左を向いたときにHeadカメラから得られる点群は、正しい位置にfetch1075を映しているように見えます。
一方で、L515カメラは点群が変な位置に表示されています。

|Head|L515|
|---|---|
|[image]|[image]|

多分原因はL515カメラのTFがズレていることです。
URDFを見ていくと、`head_tilt_link`から`head_l515_mount_link`までのリンクの座標を変換するのが間違えているのかなと思いました。
https://github.com/knorth55/jsk_robot/blob/78960957a54989f245c4cc70cb50035b1cc4f3a1/jsk_fetch_robot/jsk_fetch_startup/robots/head_l515.urdf.xacro
[image]

L515カメラを正しく配置したいのですが、以下の2つを質問したいです。
- virtualなんとかみたいなjointとかlinkは何のために使われていますか？
- どのrpyの値を変更すると他のコードに影響を与えずにL515の姿勢だけを変更できますか？

sktometometo, 2022-08-05:
- `head_l515_virtual_mount_link` はおそらくmount部品取り付けの基準位置として使ってました。`head_l515_mount_link`に直接統合してもよさそうです。
- L515の姿勢だけをいじるには `head_l515_virtual_joint` の rpy をいじればいいです。

sktometometo, 2022-08-06:
@708yamaguchi これはclose?

708yamaguchi, 2022-08-06:
今日大学でfetchを動かしてみて大丈夫そうだったらcloseします

closed 2022-08-07 by 708yamaguchi