jsk-ros-pkg/jsk_robot #1555: [fetch1075] 深夜にフリーズしていた

708yamaguchi, 2022-08-07:
fetch1075が、2022/08/07（日）の00:37にフリーズしていました。
fetch背面モニタがこの時間で停止していてpingが通らないことから確認しました。

最近の変更で怪しいのはL515の点群を出力するようにしたことです。（何かしらのリソースが足りなくなった？）
https://github.com/knorth55/jsk_robot/pull/308

他に怪しいのは、以下のようなログが/var/log/ros/robot.logに出力されていたこと
```
"l_button" type="submit" value="Sign in"></center>
</form>
</div>
</body>
</html>
<html>
<head>
<title>Login</title>
<link rel="stylesheet" href="static/css/logpro.css">
</head>
<body>
<div class="login">

Login Failed.

<form action="/login" method="post" class="loginform">
<input type="hidden" name="_xsrf" value="2|3a239c18|9c318da284a5d2362684c1d68db45b82|1659786913"/>
<center>
<table>
<tr>
<td>Username:</td>
<td><input id="u_input" type="text" name="username"></td>
</tr>
<tr>
<td>Password:</td>
<td><input id="p_input" type="password" name="password"></td>
</tr>
</table>
<input id="l_button" type="submit" value="Sign in"></center>
</form>
</div>
</body>
</html>
<html>
<head>
<title>Login</title>
<link rel="stylesheet" href="static/css/logpro.css">
</head>
<body>
<div class="login">

Login Failed.

<form action="/login" method="post" class="loginform">
<input type="hidden" name="_xsrf" value="2|ef0f63c4|ea24b8fed8268611aad155a3a563c6ec|1659787512"/>
<center>
<table>
<tr>
<td>Username:</td>
<td><input id="u_input" type="text" name="username"></td>
</tr>
<tr>
<td>Password:</td>
<td><input id="p_input" type="password" name="password"></td>
</tr>
</table>
<input id="l_button" type="submit" value="Sign in"></center>
</form>
</div>
</body>
</html>
<html>
<head>
<title>Login</title>
<link rel="stylesheet" href="static/css/logpro.css">
</head>
<body>
<div class="login">

Login Failed.

<form action="/login" method="post" class="loginform">
<input type="hidden" name="_xsrf" value="2|b7ffc5d1|a8b352679d7080ec950d964aab8127e1|1659788112"/>
<center>
<table>
<tr>
<td>Username:</td>
<td><input id="u_input" type="text" name="username"></td>
</tr>
<tr>
<td>Password:</td>
<td><input id="p_input" type="password" name="password"></td>
</tr>
</table>
<input id="l_button" type="submit" value="Sign in"></center>
</form>
</div>
</body>
</html>
[ INFO] [1659788703.065296912] [/robot_driver:ros.robot_controllers_interface.ControllerManager]: Stopped head_controller/point_head
[ INFO] [1659788703.065329628] [/robot_driver:ros.robot_controllers_interface.ControllerManager]: Started head_controller/follow_joint_trajectory
<html>
<head>
<title>Login</title>
<link rel="stylesheet" href="static/css/logpro.css">
</head>
<body>
<div class="login">

Login Failed.

<form action="/login" method="post" class="loginform">
<input type="hidden" name="_xsrf" value="2|178156d1|e8ab42368081b30b7e6ef1ac2062a5d7|1659788712"/>
<center>
<table>
<tr>
<td>Username:</td>
<td><input id="u_input" type="text" name="username"></td>
</tr>
<tr>
<td>Password:</td>
<td><input id="p_input" type="password" name="password"></td>
</tr>
</table>
<input id="l_button" type="submit" value="Sign in"></center>
</form>
</div>
</body>
</html>
<html>
<head>
<title>Login</title>
<link rel="stylesheet" href="static/css/logpro.css">
</head>
<body>
<div class="login">

Login Failed.

<form action="/login" method="post" class="loginform">
<input type="hidden" name="_xsrf" value="2|f88fa12e|4386e487690f306b8f1d7c093c9a1d7f|1659789312"/>
<center>
<table>
<tr>
<td>Username:</td>
<td><input id="u_input" type="text" name="username"></td>
</tr>
<tr>
<td>Password:</td>
<td><input id="p_input" type="password" name="password"></td>
</tr>
</table>
<input id="l_button" type="submit" value="Sign in"></center>
</form>
</div>
</body>
</html>
[ INFO] [1659789853.951919584] [/robot_driver:ros.robot_controllers_interface.ControllerManager]: Stopped head_controller/follow_joint_trajectory
[ INFO] [1659789853.951949315] [/robot_driver:ros.robot_controllers_interface.ControllerManager]: Started head_controller/point_head
<html>
<head>
<title>Login</title>
<link rel="stylesheet" href="static/css/logpro.css">
</head>
<body>
<div class="login">

Login Failed.

<form action="/login" method="post" class="loginform">
<input type="hidden" name="_xsrf" value="2|c0ff93a8|134fd4787779a46a5255849a1126c308|1659789912"/>
<center>
<table>
<tr>
<td>Username:</td>
<td><input id="u_input" type="text" name="username"></td>
</tr>
<tr>
<td>Password:</td>
<td><input id="p_input" type="password" name="p
```

ずっとrespeaker_nodeが再起動し続けていたこと
```
[respeaker_node-96] restarting process
process[respeaker_node-96]: started with pid [11341]
No handlers could be found for logger "rosout"
Traceback (most recent call last):
  File "/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_3rdparty/respeaker_ros/scripts/respeaker_node.py", line 33, in <module>
    raise RuntimeError("Check the device is connected and recognized")
RuntimeError: Check the device is connected and recognized
[respeaker_node-96] process has died [pid 11341, exit code 1, cmd /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_3rdparty/respeaker_ros/scripts/respeaker_node.py __name:=respeaker_node __log:=/home/fetch/.ros/log/5c3003c8-1530-11ed-8a50-b30ca110c64c/respeaker_node-96.log].
log file: /home/fetch/.ros/log/5c3003c8-1530-11ed-8a50-b30ca110c64c/respeaker_node-96*.log
```

syslog
https://gist.github.com/708yamaguchi/66515482f827181ef63b57c808805163
robot.log
https://gist.github.com/708yamaguchi/6adda287fa70aae2e7d10a74ea0c27af
jsk-fetch-startup.log
https://gist.github.com/708yamaguchi/f6756e84e9e5d2e3d91115fe0602fd8a

その他に何か心当たりのある人がいれば教えてください。

関連？
https://github.com/jsk-ros-pkg/jsk_robot/issues/1472

sktometometo, 2022-08-10:
2022/08/10 AM09:00頃来たタイミングでも確認。同様にフリーズしていた.

[image]

2022/08/10 00:00:00 からのシステムログ

[hoge.log](https://github.com/jsk-ros-pkg/jsk_robot/files/9295511/hoge.log)

mqcmd196, 2022-08-12:
[image]
[image]

8/10 15:25(デスクトップに表示されている時間)~19:32(確認時)
8/12 15:40(デスクトップに表示されている時間)~17:24(確認時)
にフリーズしていたことを確認しました

708yamaguchi, 2022-08-12:
タイミングから考えて，個人的には頭部USB3.0化+L515の常時使用が怪しいと考えています．

実験で必要な時以外はL515のUSBを抜いておくのが当面のデバッグかなと思うのですが，どうでしょうか．

mqcmd196, 2022-08-12:
毎日運用しているキッチンデモとかでも多分

nakane11, 2022-08-19:
2022/8/19 9:18にフリーズしたようで、9:30に確認しました
[image]

iory, 2022-08-19:
フリーズした時間の/var/log/syslogが下記です。

```
Aug 19 09:18:21 fetch1075 kernel: [19036.648195] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:22 fetch1075 kernel: [19037.298970] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:23 fetch1075 kernel: [19038.549677] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:24 fetch1075 kernel: [19038.698995] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:25 fetch1075 kernel: [19039.800548] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:26 fetch1075 kernel: [19041.051234] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:27 fetch1075 kernel: [19042.306436] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:28 fetch1075 kernel: [19043.557040] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:30 fetch1075 kernel: [19044.807758] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:31 fetch1075 kernel: [19045.752137] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Aug 19 09:18:31 fetch1075 kernel: [19046.058533] uvcvideo: Failed to query (GET_CUR) UVC controAug 19 09:44:39 fetch1075 systemd-modules-load[362]: Inserted module 'lp'
Aug 19 09:44:39 fetch1075 systemd-modules-load[362]: Inserted module 'ppdev'
Aug 19 09:44:39 fetch1075 systemd-modules-load[362]: Inserted module 'parport_pc'
Aug 19 09:44:39 fetch1075 systemd-udevd[392]: Network interface NamePolicy= disabled on kernel command line, ignoring.
Aug 19 09:44:39 fetch1075 keyboard-setup.sh[349]: cannot open file /run/tmpkbd.XFZeFz
Aug 19 09:44:39 fetch1075 systemd[1]: Starting Flush Journal to Persistent Storage...
Aug 19 09:44:39 fetch1075 systemd[1]: Started udev Kernel Device Manager.
Aug 19 09:44:39 fetch1075 kernel: [    0.000000] microcode: microcode updated early to revision 0xea, date = 2021-03-08
Aug 19 09:44:39 fetch1075 systemd[1]: Started udev Coldplug all Devices.
Aug 19 09:44:39 fetch1075 kernel: [    0.000000] Linux version 5.4.0-80-generic (buildd@lcy01-amd64-028) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #90~18.04.1-Ubuntu SMP Tue Jul 13 19:40:02 UTC 2021 (Ubuntu 5.4.0-80.90~18.04.1-generic
5.4.124)
Aug 19 09:44:39 fetch1075 kernel: [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-80-generic root=UUID=40c5ef36-2ab8-4adc-aa10-bbfd89f9d478 ro net.ifnames=0 biosdevname=0 quiet
Aug 19 09:44:39 fetch1075 kernel: [    0.000000] KERNEL supported cpus:
Aug 19 09:44:39 fetch1075 mtp-probe: checking bus 4, device 5: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.1"
Aug 19 09:44:39 fetch1075 kernel: [    0.000000]   Intel GenuineIntel
Aug 19 09:44:39 fetch1075 kernel: [    0.000000]   AMD AuthenticAMD
Aug 19 09:44:39 fetch1075 kernel: [    0.000000]   Hygon HygonGenuine
Aug 19 09:44:39 fetch1075 kernel: [    0.000000]   Centaur CentaurHauls
```

特段怪しいものはないですが、強いて言うなら、
```
uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```
がたくさん出ていますね。

```
fetch@fetch1075:~$ sudo less /var/log/syslog | grep "uvcvideo: Failed to query (GET_CUR) UVC control" | wc
  43187  886287 5484749
fetch@fetch1075:~$ sudo less /var/log/syslog | wc
 206485 3099102 21181313
```

nakane11, 2022-09-13:
2022/9/12の19:13:53から
```
uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```
が出始めて19:30にフリーズしました
```
Sep 12 19:30:19 fetch1075 kernel: [55686.604535] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:20 fetch1075 kernel: [55687.599683] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:20 fetch1075 kernel: [55687.855445] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:21 fetch1075 kernel: [55688.650303] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:21 fetch1075 kernel: [55689.106289] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:23 fetch1075 kernel: [55690.356946] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:24 fetch1075 kernel: [55691.607862] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:24 fetch1075 kernel: [55691.701400] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:25 fetch1075 kernel: [55692.751993] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:25 fetch1075 kernel: [55692.858677] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:26 fetch1075 kernel: [55694.109407] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:28 fetch1075 kernel: [55695.360362] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:28 fetch1075 kernel: [55695.360435] xhci_hcd 0000:00:14.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
Sep 12 19:30:29 fetch1075 kernel: [55696.611109] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 12 19:30:29 fetch1075 kernel: [55696.803927] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
Sep 13 09:36:11 fetch1075 systemd-modules-load[347]: Inserted module 'lp'
Sep 13 09:36:11 fetch1075 systemd-modules-load[347]: Inserted module 'ppdev'
Sep 13 09:36:11 fetch1075 systemd-modules-load[347]: Inserted module 'parport_pc'
Sep 13 09:36:11 fetch1075 systemd-udevd[389]: Network interface NamePolicy= disabled on kernel command line, ignoring.
Sep 13 09:36:11 fetch1075 systemd[1]: Starting Flush Journal to Persistent Storage...
Sep 13 09:36:11 fetch1075 keyboard-setup.sh[355]: cannot open file /run/tmpkbd.XFZeFz
Sep 13 09:36:11 fetch1075 systemd[1]: Started udev Kernel Device Manager.
Sep 13 09:36:11 fetch1075 systemd[1]: Started udev Coldplug all Devices.
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 4, device 4: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-3"
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 5, device 6: "/sys/devices/pci0000:00/0000:00:14.0/usb5/5-2"
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 4, device 2: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-1"
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 5, device 7: "/sys/devices/pci0000:00/0000:00:14.0/usb5/5-4/5-4.3"
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 5, device: 6 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 4, device: 4 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 4, device: 2 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 4, device 8: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.2"
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 4, device 6: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.1"
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 5, device: 7 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 4, device: 8 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 4, device 11: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-13"
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 4, device: 6 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 4, device: 11 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 5, device 5: "/sys/devices/pci0000:00/0000:00:14.0/usb5/5-6/5-6.4"
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 5, device: 5 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 4, device 13: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.4"
Sep 13 09:36:11 fetch1075 mtp-probe: checking bus 1, device 2: "/sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:04.2/usb1/1-4"
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 4, device: 13 was not an MTP device
Sep 13 09:36:11 fetch1075 mtp-probe: bus: 1, device: 2 was not an MTP device
```

mqcmd196, 2022-09-13:
FYI, このエラーでググるとrealsense周りで似たようなdmesgが大量に出ているという話が結構ヒットしますね