jsk-ros-pkg/jsk_robot #1576: [jsk_robot_startup] active_user.lでunix:unlinkができていない

nakane11, 2022-08-26:
fetch15のjsk-fetch-startup.logで下記のエラーがずっと出ています
```
Get PID 1840Call Stack (max depth 20):
  0: at (unix:unlink "/tmp/fullname_~d.txt" (unix:getpid))
  1: at (let (fullname) (unix:system (format nil "getent passwd `whoami` | cut -d ':' -f 5 | cut -d ',' -f 1 > /tmp/fullname_~d.txt" (unix:getpid))) (warn "Get PID ~A" (unix:getpid)) (with-open-file (f (format nil "/tmp/fullname_~d.txt" (unix:getpid))) (setq fullname (read-line f))) (unix:unlink "/tmp/fullname_~d.txt" (unix:getpid)) fullname)
  2: at (get-fullname)
  3: at (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*))
  4: at (progn (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*)))
  5: at (if (not (and (boundp '*user-name*) *user-name*)) (progn (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*))))
  6: at (when (not (and (boundp '*user-name*) *user-name*)) (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*)))
  7: at (unless (and (boundp '*user-name*) *user-name*) (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*)))
Call Stack (max depth 20):
  0: at (ros::ros-error ros::errstr)
  1: at (let ((ros::errstr (with-output-to-string (ros::str) (lisp::print-error-message ros::err ros::str)))) (ros::ros-error ros::errstr))
  2: at #<compiled-code #X55f7c91743e0 signals>
  3: at #<compiled-code #X55f7c91743e0 signals>
  4: at (unix:unlink "/tmp/fullname_~d.txt" (unix:getpid))
  5: at (let (fullname) (unix:system (format nil "getent passwd `whoami` | cut -d ':' -f 5 | cut -d ',' -f 1 > /tmp/fullname_~d.txt" (unix:getpid))) (warn "Get PID ~A" (unix:getpid)) (with-open-file (f (format nil "/tmp/fullname_~d.txt" (unix:getpid))) (setq fullname (read-line f))) (unix:unlink "/tmp/fullname_~d.txt" (unix:getpid)) fullname)
  6: at (get-fullname)
  7: at (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*))
  8: at (progn (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*)))
  9: at (if (not (and (boundp '*user-name*) *user-name*)) (progn (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*))))
 10: at (when (not (and (boundp '*user-name*) *user-name*)) (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*)))
 11: at (unless (and (boundp '*user-name*) *user-name*) (let ((fullname (get-fullname)) (username (get-username))) (if (string= fullname "") (setq *user-name* username) (setq *user-name* fullname)) (ros::set-param "/active_user/launch_user_name" *user-name*)))
[ERROR] [1661498302.971673075] [/active_user:ros.roseus]: ARGUMENT-ERROR: mismatch argument in (ros::ros-error ros::errstr)

[active_user-34] process has died [pid 1840, exit code 1, cmd /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_roseus/roseus/bin/roseus /home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot/jsk_robot_common/jsk_robot_startup/lifelog/active_user.l motor_state:=/robot_state odom_state:=/odom_combined joint_states:=/joint_states diagnostics:=/diagnostics __name:=active_user __log:=/home/fetch/.ros/log/4a72b8ce-24a8-11ed-88a1-812d97cff02d/active_user-34.log].
log file: /home/fetch/.ros/log/4a72b8ce-24a8-11ed-88a1-812d97cff02d/active_user-34*.log
```

表示されているのはARGUMENT-ERRORですが昨日mergeされたhttps://github.com/jsk-ros-pkg/jsk_robot/pull/1552 の変更で追加された[(get-fullname)](https://github.com/jsk-ros-pkg/jsk_robot/blob/3e2952f3bafd4f7200b70267b3075865e9042710/jsk_robot_common/jsk_robot_startup/lifelog/active_user.l#L89-L102)で`unix:unlink`が上手くいっていない気がします
```
fetch@fetch15:/tmp$ ls | grep fullname | wc
  18213   18213  339851
fetch@fetch15:/tmp$ ls | grep fullname | wc
  18439   18439  344067
fetch@fetch15:/tmp$ ls | grep 1840
fullname_11840.txt
fullname_18401.txt
fullname_18402.txt
fullname_18406.txt
fullname_18408.txt
fullname_1840.txt
fullname_31840.txt
```

knorth55, 2022-08-26:
なおしました． #1577

sktometometo, 2022-08-26:
じつはこれだと不十分っぽいのでPR出しておきます

nakane11, 2022-08-26:
ありがとうございます

closed 2022-08-26 by nakane11