jsk-ros-pkg/jsk_robot #1579: Pepperの体内でのjsk_pepper_startupパッケージの更新方法がよく分かりません

kochigami, 2022-08-31:
Oliveの体内で研究会のアプリを立ち上げてみたいのですが，
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot/jsk_pepper_startup/apps

Pepper体内にあるjsk_pepper_startupパッケージをどのように更新すれば良いのでしょうか．
今は，jsk_pepper_startup.launchは体内で立ち上がるのですが，
https://github.com/jsk-ros-pkg/jsk_robot/pull/1528 がマージされる前に用意していただいたものなので，appsは含まれていないものになっています．
（最終的には，naoqi_bridgeや自分が作ったデモ用のパッケージも体内に入れたいと考えております．）

毎回パッケージの内容が更新される度に
https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot/cross#readme
の手順を行う必要がありますでしょうか．
（このReadmeの中身も分からないことが多く，またissueを立てると思います・・・すみません．）

k-okada, 2022-08-31:
１）手元のPCのソース（
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_naoqi_robot/cross/repos/pepper.repos)
とか，をロボット体内にコピーできる環境を作る。(./install.sh ... まで）
２）手元のPCのソース（jsk_pepper_startup, pepper_eus)をロボットの体内にコピーする（make user,
./install の部分）
の2段階あります。
通常は１）はやらなくてよくて2）だけやる必要がありますが，
2をやるのに最初の一回だけ１）（の./install.sh 以外）をやっておく必要があります

https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_naoqi_robot/cross#readme
はunitreeのコピーそのままでダメですね。
--
◉ Kei Okada

kochigami, 2022-09-02:
教わった => #1583

closed 2022-09-02 by kochigami