jsk-ros-pkg/jsk_robot #1580: [fetch1075] FetchのTweetがおかしい

sktometometo, 2022-09-01:
Fetch1075のTweetがBase64 encodingされた何かっぽいんですが、これは何かのデバック中ですか?

[image]

sktometometo, 2022-09-01:
09:30のキッチンデモのDockのタイミングくらいでtweetされてる。

tkmtnt7000, 2022-09-01:
昨日あたりまでは正常にツイートできていそうだったのでおかしいですね。。。午後に確認してみます。

sktometometo, 2022-09-01:
8/30 にも起きてる。ここ数週間たまにこんな感じで荒ぶっている気がします。最近何かTweet周りいじられてる?

[image]

sktometometo, 2022-09-01:
8/29 にも。 8/31がたまたま起きなかった?
[image]

tkmtnt7000, 2022-09-01:
僕が最近やっているsmachから通知を行うためにrostwitterを変更していて，ツイートにBase64をを直接送れるようにしています．本来は，画像に変換されて出力されるようにしています．

sktometometo, 2022-09-01:
了解です、なるほど。

tkmtnt7000, 2022-09-01:
該当となるPRをリンクさせておきます．https://github.com/knorth55/jsk_3rdparty/pull/31，https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/375

sktometometo, 2022-09-01:
RSJあるので来週以降とかにやりましょう

tkmtnt7000, 2022-09-01:
はい．ちょっと待ち時間がある作業をしたときに確認したのですが，
`[ERROR] [1661819674.539695] [/tweet:rosout]: Incorrect padding`
おそらくこのエラーが原因と考えています．
base64から画像に変換する際に正しくdecodeできていないことから発生しているようです．
ログ：全体　https://drive.google.com/file/d/10pi7y2w1XHbQrtCz_76kXcFwHnA9k0-E/view?usp=sharing

iory, 2022-09-01:
@tkmtnt7000 
base64のパースで空白で区切ってもらうという前提を設けました。
こちらのPRで修正しました。
https://github.com/knorth55/jsk_3rdparty/pull/32
https://github.com/tkmtnt7000/jsk_robot/pull/8

sktometometo, 2022-09-02:
Seems to be fixed.

closed 2022-09-02 by sktometometo