jsk-ros-pkg/jsk_robot #1590: [fetch1075] go-to-kitchen failes with error

mqcmd196, 2022-09-10:
Kitchen demo failes with following error
```
[33mExecuting state :inspect-kitchen
[0m[INFO] [1662784079.009872] [/server_name:rosout]: Received SMACH status
[INFO] [1662784079.019202] [/server_name:rosout]: - status -> INSPECT-KITCHEN
[INFO] [1662784079.022798] [/server_name:rosout]: - info_str -> 
[Text2Wave] Invalid voice_name (四国めたん-ノーマル). Use default voice.
[Text2Wave] Speak using voice_name (四国めたん-ノーマル)..
[Text2Wave] Using cached sound file (/home/fetch/.ros/voicevox_texttospeech/cache/fceb778912b21e4fdeebe44cb7d1e1f5--2.wav) for 到着しました
[INFO] [1662784079.024687] [/server_name:rosout]: - description_str -> キッチンに移動したよ
[INFO] [1662784079.027225] [/server_name:rosout]: - image_str -> GAsAADkQHGPPv/k5HQAAAGhlYWRfY2FtZXJhX3JnYl9vcHRpY2FsX2ZyYW1lGgAA
[WARN] [1662784079.029275] [/server_name:rosout]: smach does not have INFO, see https://github.com/jsk-ros-pkg/jsk_robot/tree/master/jsk_robot_common/jsk_robot_startup#smach_to_mailpy for more info
Call Stack (max depth 20):
  0: at (if label-names (format nil "~Aがあったよ" label-names) "")
  1: at (let ((label-names (notify-recognition :location "kitchen")) (notify-text (if label-names (format nil "~Aがあったよ" label-names) ""))) (inspect-kitchen :tweet (cdr (assoc 'tweet userdata))) (set-alist 'description (format nil "キッチンの様子を見たよ。~A" notify-text) userdata))
  2: at (funcall action userdata)
  3: at (funcall action userdata)
  4: at (cond ((send self :submachine) (send action :execute userdata :step step)) ((functionp action) (funcall action userdata)) (t (send self :name)))
  5: at (setq res (cond ((send self :submachine) (send action :execute userdata :step step)) ((functionp action) (funcall action userdata)) (t (send self :name))))
  6: at (let (res) (send self :remap userdata :invert nil) (setq res (cond ((send self :submachine) (send action :execute userdata :step step)) ((functionp action) (funcall action userdata)) (t (send self :name)))) (send self :remap userdata :invert t) res)
  7: at (send last-state :execute userdata)
  8: at (let* ((last-state astate) (trans (send self :next-arc-list astate)) (exec-result (send last-state :execute userdata))) (ros::ros-debug "trans: ~A" trans) (setq trans (remove-if-not #'(lambda (tr) (send tr :check exec-result)) trans)) (setq trans-list (append trans-list (list trans))) (case (length trans) (0 (error "undefined transition ~A from ~A~%" exec-result last-state)) (1 t) (t (case (length active-state) (1 (warn "multiple transitions ~A from ~A~%" exec-result last-state)) (t t)))) (when (not (eq astate last-state)) (error "active state has not changed ~A -> ~A~%" last-state astate) (return-from :execute-impl (send astate :name))) (setq next-active-state (append next-active-state (send-all trans :to))) (if (send astate :submachine) (send (send astate :submachine) :reset-state)) (push exec-result ret))
  9: at (tagbody (warning-message 3 "Executing state ~A~%" (send astate :name)) (let* ((last-state astate) (trans (send self :next-arc-list astate)) (exec-result (send last-state :execute userdata))) (ros::ros-debug "trans: ~A" trans) (setq trans (remove-if-not #'(lambda (tr) (send tr :check exec-result)) trans)) (setq trans-list (append trans-list (list trans))) (case (length trans) (0 (error "undefined transition ~A from ~A~%" exec-result last-state)) (1 t) (t (case (length active-state) (1 (warn "multiple transitions ~A from ~A~%" exec-result last-state)) (t t)))) (when (not (eq astate last-state)) (error "active state has not changed ~A -> ~A~%" last-state astate) (return-from :execute-impl (send astate :name))) (setq next-active-state (append next-active-state (send-all trans :to))) (if (send astate :submachine) (send (send astate :submachine) :reset-state)) (push exec-result ret)) (go #:dolist7258522))
 10: at (progn (setq astate (pop #:dolist7258521)) (tagbody (warning-message 3 "Executing state ~A~%" (send astate :name)) (let* ((last-state astate) (trans (send self :next-arc-list astate)) (exec-result (send last-state :execute userdata))) (ros::ros-debug "trans: ~A" trans) (setq trans (remove-if-not #'(lambda (tr) (send tr :check exec-result)) trans)) (setq trans-list (append trans-list (list trans))) (case (length trans) (0 (error "undefined transition ~A from ~A~%" exec-result last-state)) (1 t) (t (case (length active-state) (1 (warn "multiple transitions ~A from ~A~%" exec-result last-state)) (t t)))) (when (not (eq astate last-state)) (error "active state has not changed ~A -> ~A~%" last-state astate) (return-from :execute-impl (send astate :name))) (setq next-active-state (append next-active-state (send-all trans :to))) (if (send astate :submachine) (send (send astate :submachine) :reset-state)) (push exec-result ret)) (go #:dolist7258522)))
 11: at (if (endp #:dolist7258521) (setq astate nil) (progn (setq astate (pop #:dolist7258521)) (tagbody (warning-message 3 "Executing state ~A~%" (send astate :name)) (let* ((last-state astate) (trans (send self :next-arc-list astate)) (exec-result (send last-state :execute userdata))) (ros::ros-debug "trans: ~A" trans) (setq trans (remove-if-not #'(lambda (tr) (send tr :check exec-result)) trans)) (setq trans-list (append trans-list (list trans))) (case (length trans) (0 (error "undefined transition ~A from ~A~%" exec-result last-state)) (1 t) (t (case (length active-state) (1 (warn "multiple transitions ~A from ~A~%" exec-result last-state)) (t t)))) (when (not (eq astate last-state)) (error "active state has not changed ~A -> ~A~%" last-state astate) (return-from :execute-impl (send astate :name))) (setq next-active-state (append next-active-state (send-all trans :to))) (if (send astate :submachine) (send (send astate :submachine) :reset-state)) (push exec-result ret)) (go #:dolist7258522))))
 12: at (tagbody #:dolist7258522 (if (endp #:dolist7258521) (setq astate nil) (progn (setq astate (pop #:dolist7258521)) (tagbody (warning-message 3 "Executing state ~A~%" (send astate :name)) (let* ((last-state astate) (trans (send self :next-arc-list astate)) (exec-result (send last-state :execute userdata))) (ros::ros-debug "trans: ~A" trans) (setq trans (remove-if-not #'(lambda (tr) (send tr :check exec-result)) trans)) (setq trans-list (append trans-list (list trans))) (case (length trans) (0 (error "undefined transition ~A from ~A~%" exec-result last-state)) (1 t) (t (case (length active-state) (1 (warn "multiple transitions ~A from ~A~%" exec-result last-state)) (t t)))) (when (not (eq astate last-state)) (error "active state has not changed ~A -> ~A~%" last-state astate) (return-from :execute-impl (send astate :name))) (setq next-active-state (append next-active-state (send-all trans :to))) (if (send astate :submachine) (send (send astate :submachine) :reset-state)) (push exec-result ret)) (go #:dolist7258522)))))
 13: at (let ((astate nil) (#:dolist7258521 active-state)) nil (tagbody #:dolist7258522 (if (endp #:dolist7258521) (setq astate nil) (progn (setq astate (pop #:dolist7258521)) (tagbody (warning-message 3 "Executing state ~A~%" (send astate :name)) (let* ((last-state astate) (trans (send self :next-arc-list astate)) (exec-result (send last-state :execute userdata))) (ros::ros-debug "trans: ~A" trans) (setq trans (remove-if-not #'(lambda (tr) (send tr :check exec-result)) trans)) (setq trans-list (append trans-list (list trans))) (case (length trans) (0 (error "undefined transition ~A from ~A~%" exec-result last-state)) (1 t) (t (case (length active-state) (1 (warn "multiple transitions ~A from ~A~%" exec-result last-state)) (t t)))) (when (not (eq astate last-state)) (error "active state has not changed ~A -> ~A~%" last-state astate) (return-from :execute-impl (send astate :name))) (setq next-active-state (append next-active-state (send-all trans :to))) (if (send astate :submachine) (send (send astate :submachine) :reset-state)) (push exec-result ret)) (go #:dolist7258522))))) nil)
 14: at (block nil (let ((astate nil) (#:dolist7258521 active-state)) nil (tagbody #:dolist7258522 (if (endp #:dolist7258521) (setq astate nil) (progn (setq astate (pop #:dolist7258521)) (tagbody (warning-message 3 "Executing state ~A~%" (send astate :name)) (let* ((last-state astate) (trans (send self :next-arc-list astate)) (exec-result (send last-state :execute userdata))) (ros::ros-debug "trans: ~A" trans) (setq trans (remove-if-not #'(lambda (tr) (send tr :check exec-result)) trans)) (setq trans-list (append trans-list (list trans))) (case (length trans) (0 (error "undefined transition ~A from ~A~%" exec-result last [... cut ...]
 RoboticMaterials/FA-I-sensor         git master  (-)                                                     ca3e80b00429                                                    github.com/RoboticMaterials/FA-I-sensor.git
 RobotWebTools/rosbridge_suite        git <detached>  (0.11.9)                                            8d94b0be7ca0                                                    github.com/RobotWebTools/rosbridge_suite.git
 PR2/app_manager                      git fetch15                                                         cfa5a0704b38                                                    github.com/knorth55/app_manager.git
```

mqcmd196, 2022-09-10:
https://github.com/knorth55/jsk_robot/pull/343 ?

tkmtnt7000, 2022-10-05:
It seems to be fixed.

closed 2022-10-19 by knorth55