jsk-ros-pkg/jsk_robot #1598: [Fetch] Wi-Fi roaming aggressiveness configuration

sktometometo, 2022-09-14:
Related to https://github.com/jsk-ros-pkg/jsk_robot/pull/1595.

I am trying to update Wi-Fi roaming aggressiveness configuration with Fetch15 and Fetch1075.

Result of `lspci` is

```bash
~ $ lspci
00:00.0 Host bridge: Intel Corporation Device 9b53 (rev 03)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 03)
00:02.0 VGA compatible controller: Intel Corporation Device 9bc8 (rev 03)
00:12.0 Signal processing controller: Intel Corporation Device 06f9
00:14.0 USB controller: Intel Corporation Device 06ed
00:14.2 RAM memory: Intel Corporation Device 06ef
00:16.0 Communication controller: Intel Corporation Device 06e0
00:17.0 SATA controller: Intel Corporation Device 06d2
00:1c.0 PCI bridge: Intel Corporation Device 06ba (rev f0)
00:1c.3 PCI bridge: Intel Corporation Device 06bb (rev f0)
00:1d.0 PCI bridge: Intel Corporation Device 06b0 (rev f0)
00:1f.0 ISA bridge: Intel Corporation Device 0684
00:1f.3 Audio device: Intel Corporation Device 06c8
00:1f.4 SMBus: Intel Corporation Device 06a3
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device 06a4
00:1f.6 Ethernet controller: Intel Corporation Device 0d4d
01:00.0 PCI bridge: Pericom Semiconductor PI7C9X111SL PCIe-to-PCI Reversible Bridge (rev 02)
02:04.0 USB controller: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller (rev 62)
02:04.1 USB controller: VIA Technologies, Inc. VT82xx/62xx UHCI USB 1.1 Controller (rev 62)
02:04.2 USB controller: VIA Technologies, Inc. USB 2.0 (rev 65)
03:00.0 Network controller: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak] (rev 10)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. Device 8125 (rev 05)
05:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981
```

Wi-Fi device is `03:00` so

```bash
~ $ lspci -s 03:00 -vv
03:00.0 Network controller: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak] (rev 10)
        Subsystem: Intel Corporation Device 2110
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 179
        Region 0: Memory at b1300000 (64-bit, non-prefetchable) [size=8K]
        Capabilities: <access denied>
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi
```

And `ifconfig` result is 

```bash
~ $ ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether a8:a1:59:43:af:c4  txqueuelen 1000  (イーサネット)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  memory 0xb1500000-b1520000  

eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.42.42.1  netmask 255.255.255.0  broadcast 10.42.42.255
        inet6 fe80::aaa1:59ff:fe43:afc2  prefixlen 64  scopeid 0x20<link>
        ether a8:a1:59:43:af:c2  txqueuelen 1000  (イーサネット)
        RX packets 187741041  bytes 18685962853 (18.6 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20106694  bytes 2141546954 (2.1 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 129  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (ローカルループバック)
        RX packets 940199220  bytes 647190067053 (647.1 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 940199220  bytes 647190067053 (647.1 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 133.11.216.217  netmask 255.255.255.0  broadcast 133.11.216.255
        inet6 fe80::7cd4:81f8:7171:c630  prefixlen 64  scopeid 0x20<link>
        ether bc:54:2f:2e:f3:42  txqueuelen 1000  (イーサネット)
        RX packets 58050344  bytes 4063036913 (4.0 GB)
        RX errors 0  dropped 46  overruns 0  frame 0
        TX packets 97482203  bytes 106992565114 (106.9 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Then, result of aggresiveness configuration with `iwconfig` is

```bash
~ $ sudo iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"sanshiro"  
          Mode:Managed  Frequency:5.68 GHz  Access Point: 78:72:5D:B6:CC:A9   
          Bit Rate=180 Mb/s   Tx-Power=17 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=53/70  Signal level=-57 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:308  Invalid misc:71   Missed beacon:0

~ $ sudo iwconfig wlan0 sens -60
Error for wireless request "Set Sensitivity" (8B08) :
    SET failed on device wlan0 ; Operation not supported.

```

sktometometo, 2022-09-14:
mac80211 module seems to be a driver  for this card.
https://www.kernel.org/doc/html/v4.9/80211/mac80211.html

sktometometo, 2022-09-14:
`iwlwifi` is a driver for Intel wireless card and it seems to be based on mac80211.
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi

sktometometo, 2022-09-15:
`bgscan` configuration of `wpa_supplicant` works well.
https://wiki.archlinux.org/title/wpa_supplicant#Roaming

But `wpa_supplicant` is controlled by `NetworkManager` so we can't configure `wpa_supplicant` manually.

sktometometo, 2022-09-15:
When NetworkManager runs and Wi-Fi connection is enabled, bgscan configuration can be seen.

```bash
~ $ sudo wpa_cli get_network 0 bgscan
Selected interface 'wlp0s20f3'
"simple:1:-60:3600
```

And `bgscan` can be configured

```bash
~ $ sudo wpa_cli set_network 0 bgscan "\"simple:1:-60:3600\""
Selected interface 'wlp0s20f3'
OK
```

For more information about `wpa_cli` and `bgscan`, please see

https://wiki.archlinux.org/title/Wpa_supplicant

sktometometo, 2022-09-15:
@k-okada

sktometometo, 2022-09-15:
`bgscan` configuration is reset if network is reset.

```bash
~ $ sudo wpa_cli get_network 0 bgscan
[sudo] sktometometo のパスワード: 
Selected interface 'wlp0s20f3'
"simple:10:-60:3600"sktometometo@Persing
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:]
[12:40 午後 木  9月 15]
~ $ sudo nmcli c down sanshiro
接続 'sanshiro' が正常に非アクティブ化されました (D-Bus アクティブパス: /org/freedesktop/NetworkManager/ActiveConnection/8)
sktometometo@Persing
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:]
[12:40 午後 木  9月 15]
(failed reverse-i-search)`c dup': sudo nmcli ^Cdown sanshiro
sktometometo@Persing
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:]
[12:40 午後 木  9月 15]
~ $ sudo nmcli c up sanshiro
接続が正常にアクティベートされました (D-Bus アクティブパス: /org/freedesktop/NetworkManager/ActiveConnection/9)
sktometometo@Persing
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:]
[12:40 午後 木  9月 15]
~ $ sudo wpa_cli get_network 0 bgscan
Selected interface 'wlp0s20f3'
"simple:30:-80:86400"
```

sktometometo, 2022-09-15:
# Default configuration.

https://user-images.githubusercontent.com/9410362/190309023-a28a3837-6847-4450-8345-a22ebaf9418d.mp4

# Aggressive configuration

`bgscan` is set to `"simple:30:-60:3600"`
Wi-Fi AP is changed when outside 73B2 and inside 73B2

https://user-images.githubusercontent.com/9410362/190309103-2c5ca81e-3bb9-4555-aaee-ac47de3d53ba.mp4

knorth55, 2022-09-15:
つまり，これをすればoutside設定がいらなそう？
Fetchでもできるんだろうか

sktometometo, 2022-09-15:
@knorth55 起動してNetworkManagerがWi-Fiのプロファイルを有効化するたびにスクリプトがwpa_supplicantを設定し直すようにすればFetchでも動くと思いますが、thresholdは固定値なのでよきチューニングが必要になりますね。

knorth55, 2022-09-16:
@sktometometo じゃあどうしようか．こっちのaggressive configurationを試してみますか．

sktometometo, 2022-09-18:
技術的に可能であることはこれで示しましたが、いろいろ変更する必要があるのと、73B2内のみで使う人にとっては切り替えがたまに起こって遅くなって使用感が悪くなるので、関係者の合意が取れないとFetchでは反映しないほうがいいのではと思います。