jsk-ros-pkg/jsk_robot #1599: [fetch15] tweetの画像を添付できない

nakane11, 2022-09-15:
fetch15が9/15 6:30のツイートを最後に画像を貼れなくなっていてこのエラーが原因かと思います
/tweet_imageは正常です
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 892, in emit
    self.flush()
  File "/usr/lib/python2.7/logging/__init__.py", line 852, in flush
    self.stream.flush()
IOError: [Errno 28] No space left on device
Logged from file core.py, line 140
```
[20220915-fetch15-jsk-fetch-startup.log](https://gist.github.com/nakane11/3d427f3481c9e4994952d85105db2253#file-20220915-fetch15-jsk-fetch-startup-log)

nakane11, 2022-09-15:
```
fetch@fetch15:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev             16G     0   16G   0% /dev
tmpfs           3.2G  7.0M  3.1G   1% /run
/dev/nvme0n1p2  916G  869G     0 100% /
tmpfs            16G     0   16G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs            16G     0   16G   0% /sys/fs/cgroup
/dev/nvme0n1p1  511M  6.7M  505M   2% /boot/efi
/dev/loop1       64M   64M     0 100% /snap/core20/1623
/dev/loop2       48M   48M     0 100% /snap/snapd/16778
/dev/loop0      401M  401M     0 100% /snap/gnome-3-38-2004/112
/dev/loop3      219M  219M     0 100% /snap/gnome-3-34-1804/77
/dev/loop5       45M   45M     0 100% /snap/certbot/2344
/dev/loop6      219M  219M     0 100% /snap/gnome-3-34-1804/72
/dev/loop4       56M   56M     0 100% /snap/core18/2566
/dev/loop7      2.7M  2.7M     0 100% /snap/gnome-system-monitor/174
/dev/loop9       62M   62M     0 100% /snap/core20/1611
/dev/loop8      347M  347M     0 100% /snap/gnome-3-38-2004/115
/dev/loop10      56M   56M     0 100% /snap/core18/2560
tmpfs           3.2G   32K  3.2G   1% /run/user/1000
/dev/loop11     2.7M  2.7M     0 100% /snap/gnome-calculator/920
/dev/loop12     768K  768K     0 100% /snap/gnome-characters/741
/dev/loop13      47M   47M     0 100% /snap/snapd/16292
/dev/loop15     640K  640K     0 100% /snap/gnome-logs/112
/dev/loop16     114M  114M     0 100% /snap/core/13425
/dev/loop14      82M   82M     0 100% /snap/gtk-common-themes/1534
/dev/loop17     768K  768K     0 100% /snap/gnome-characters/761
/dev/loop18     640K  640K     0 100% /snap/gnome-logs/106
/dev/loop19      44M   44M     0 100% /snap/certbot/2192
/dev/loop20     2.5M  2.5M     0 100% /snap/gnome-calculator/884
/dev/loop21      92M   92M     0 100% /snap/gtk-common-themes/1535
/dev/loop22     2.7M  2.7M     0 100% /snap/gnome-system-monitor/178
/dev/loop23     128K  128K     0 100% /snap/bare/5
/dev/loop24     115M  115M     0 100% /snap/core/13741

```

iory, 2022-09-15:
```
fetch@fetch15:/var/log$     du -h -d 1 | sort -h
du: cannot read directory './gdm3': Permission denied
du: cannot read directory './speech-dispatcher': Permission denied
du: cannot read directory './letsencrypt': Permission denied
4.0K    ./chrony
4.0K    ./dist-upgrade
4.0K    ./gdm3
4.0K    ./speech-dispatcher
4.0K    ./sysstat
8.0K    ./hp
40K     ./letsencrypt
48K     ./unattended-upgrades
64K     ./cups
256K    ./apt
884K    ./installer
1.3M    ./mongodb
7.5M    ./supervisor
4.1G    ./journal
371G    ./ros
670G    .
```
rosのログが溜まってますね。

@knorth55 @708yamaguchi  
定期的にこれらのログってどこかに保存しているもしくは削除していますか？

708yamaguchi, 2022-09-15:
> 定期的にこれらのログってどこかに保存しているもしくは削除していますか？

ログのGoogle Driveへのアップロードなどはしていないと思います。
ただ、jsk-fetch-startup.log.3みたいなファイルがあるので、自動的にlogrotate的なことはしている気がしています。

supervisorの設定でログファイルの上限サイズを低めに設定するのがよいでしょうか。
```
logfile_maxbytes

The maximum number of bytes that may be consumed by the activity log file before it is rotated (suffix multipliers like “KB”, “MB”, and “GB” can be used in the value). Set this value to 0 to indicate an unlimited log size.
```
http://supervisord.org/configuration.html

iory, 2022-09-15:
> supervisorの設定でログファイルの上限サイズを低めに設定するのがよいでしょうか。

それが良いと思います。

あと

```
root@fetch15:/var/log# ls -lh | sort -k 5 -h
-rw-------     1 root              root               0 Dec  4  2020 boot.log
-rw-r-----     1 root              adm                0 Dec  4  2020 apport.log
-rw-r--r--     1 root              root               0 Jul  5 11:17 alternatives.log
-rw-r--r--     1 root              root               0 Sep  1 00:06 dpkg.log
total 296G
-rw-r--r--     1 root              root             125 Apr  4  2021 alternatives.log.3.gz
-rw-r--r--     1 root              root             128 Jan  7  2021 alternatives.log.6.gz
-rw-r--r--     1 root              root             143 Mar 29  2021 alternatives.log.4.gz
-rw-rw----     1 root              utmp             186 Aug 26 10:58 btmp.1.gz
-rw-r--r--     1 root              root             237 Feb 16  2021 alternatives.log.5.gz
-rw-r-----     1 syslog            adm              252 Sep 14 00:59 haproxy.log.2.gz
-rw-r--r--     1 root              root             274 Nov  1  2021 dpkg.log.9.gz
-rw-r--r--     1 root              root             435 Jan 21  2022 dpkg.log.7.gz
-rw-r--r--     1 root              root             475 Jul 30  2021 dpkg.log.11.gz
-rw-r-----     1 root              adm              496 Dec  4  2020 apport.log.1
-rw-r--r--     1 root              root             501 Jul  5 10:46 alternatives.log.1
-rw-r--r--     1 root              root             564 Jul 29  2021 alternatives.log.2.gz
-rw-r-----     1 syslog            adm              616 Sep 12 04:01 haproxy.log.5.gz
-rw-r--r--     1 root              root             908 Jan 13  2022 dpkg.log.8.gz
-rw-r--r--     1 root              root             970 Jun 24 03:06 dpkg.log.3.gz
-rw-r--r--     1 root              root             986 Jul  7 12:07 dpkg.log.2.gz
-rw-r--r--     1 root              root            1.1K Apr 11 10:41 dpkg.log.6.gz
-rw-r--r--     1 root              root            1.1K Apr 21 10:00 certbot_renewal.log
-rw-r--r--     1 root              root            1.1K Oct 22  2021 dpkg.log.10.gz
-rw-r--r--     1 root              root            1.4K Sep 15 04:01 gpu-manager.log
-rw-r--r--     1 root              root            1.6K Aug 26 03:06 dpkg.log.1
-rw-r--r--     1 root              root            1.7K May 28 03:01 dpkg.log.4.gz
-rw-------     1 root              root            1.9K Mar 31 15:57 ubuntu-advantage.log.6.gz
-rw-r--r--     1 root              root            2.0K Apr 29 03:01 dpkg.log.5.gz
-rw-rw----     1 root              utmp            2.7K Sep 15 10:54 btmp
-rw-------     1 root              root            3.2K Jun 30 15:52 ubuntu-advantage.log.3.gz
-rw-------     1 root              root            3.3K Jul 31 15:53 ubuntu-advantage.log.2.gz
-rw-------     1 root              root            3.4K Apr 30 15:40 ubuntu-advantage.log.5.gz
-rw-------     1 root              root            3.5K May 31 16:58 ubuntu-advantage.log.4.gz
-rw-r-----     1 syslog            adm             3.5K Aug 28 04:01 mail.log.3.gz
-rw-r-----     1 syslog            adm             3.6K Aug 21 04:01 mail.log.4.gz
-rw-r-----     1 syslog            adm             3.7K Sep 12 23:58 haproxy.log.4.gz
-rw-r-----     1 syslog            adm             3.9K Sep  4 04:01 mail.log.2.gz
drwx------     2 speech-dispatcher root            4.0K Apr 23  2018 speech-dispatcher
drwxr-sr-x+    3 root              systemd-journal 4.0K Dec  4  2020 journal
drwxrwxr-x     2 root              root            4.0K Dec  4  2020 installer
drwxr-x---     2 root              adm             4.0K Sep  1 00:08 unattended-upgrades
drwxr-xr-x     2 _chrony           _chrony         4.0K Aug 26  2020 chrony
drwxr-xr-x     2 mongodb           mongodb         4.0K Sep 11 19:56 mongodb
drwxr-xr-x     2 root              root            4.0K Dec 24  2020 supervisor
drwxr-xr-x     2 root              root            4.0K Jan 17  2020 sysstat
drwxr-xr-x     2 root              root            4.0K Jul 10  2020 dist-upgrade
drwxr-xr-x     2 root              root            4.0K Sep  1 00:06 apt
drwxr-xr-x     2 root              root            4.0K Sep 15 00:10 cups
drwxr-xr-x     3 root              root            4.0K Aug  7  2020 hp
drwx--x--x     2 root              gdm             4.0K Feb 19  2019 gdm3
-rw-r-----     1 syslog            adm             4.5K Sep 11 19:51 haproxy.log.6.gz
-rw-r-----     1 syslog            adm             4.8K Jul 31 23:51 haproxy.log.46.gz
-rw-r-----     1 syslog            adm             5.1K Jul 29 23:55 haproxy.log.48.gz
-rw-rw-r--     1 root              utmp            5.1K Sep  1 00:04 wtmp.1.gz
-rw-r-----     1 syslog            adm             5.4K Aug 24 00:01 haproxy.log.23.gz
-rw-r-----     1 syslog            adm             5.5K Aug 13 00:05 haproxy.log.34.gz
-rw-r-----     1 syslog            adm             5.5K Sep  7 00:06 haproxy.log.9.gz
-rw-r-----     1 syslog            adm             5.6K Aug 11 23:59 haproxy.log.35.gz
-rw-r-----     1 syslog            adm             5.6K Aug 27 23:58 haproxy.log.19.gz
-rw-r-----     1 syslog            adm             5.8K Aug 17 00:04 haproxy.log.30.gz
-rw-r-----     1 syslog            adm             5.8K Aug 31 00:06 haproxy.log.16.gz
[... 61 lines cut ...]
-rw-r-----     1 syslog            adm             1.6M Aug 28 04:01 kern.log.3.gz
-rw-r-----     1 syslog            adm             2.2M Sep  4 06:05 kern.log.2.gz
[... cut ...]
-rw-r-----     1 syslog            adm             2.4G Sep 14 00:22 syslog.1.gz-2022091400.backup
-rw-r-----     1 syslog            adm             2.4G Sep 14 00:31 syslog.4.gz
-rw-r-----     1 syslog            adm              17G Sep 15 11:28 syslog
-rw-r-----     1 syslog            adm              61G Sep 15 00:12 syslog.1
-rw-r-----     1 syslog            adm             214G Sep 15 11:28 cron.log
```
cronのログが異常に大きいのも問題そうですね。

```
Sep 15 11:21:44 fetch15 atd[27241]: File a0055601a45573 is in wrong format - aborting
```
これがずっとcron.logで出ているのですが誰かcronに設定している何かを変えたでしょうか？

708yamaguchi, 2022-09-15:
> これがずっとcron.logで出ているのですが誰かcronに設定している何かを変えたでしょうか？

このエラー、以前（1~2年前とか）にも見たことがあります。
その時にどうやって解決したかは忘れてしまったのですが、根本解決は出来ず適当にググったウェブページを見て対症療法していた気がします。

708yamaguchi, 2022-09-15:
以前の会話の記録を発掘できたので、貼ります

Naoto Tsukamoto, 1月19日, 14:21
Fetch15noJan 19 14:21:06 fetch15 atd[27367]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27368]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27369]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27370]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27371]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27372]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27373]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27374]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27375]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27376]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27377]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27378]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27379]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27380]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27381]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27382]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27383]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27384]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27385]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27386]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27388]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27387]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27389]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27390]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27391]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27392]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27393]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27394]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27395]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27396]: File a0037d01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27397]: File a0037e01a0d46f is in wrong format - aborting
Jan 19 14:21:06 fetch15 atd[27398]: File a0037d01a0d46f is in wrong format - aborting

Naoto Tsukamoto, 1月19日, 14:24
Fetch15の体内容量が100%になっていて，確認したところ/var/log/以下のcron.logが168GB, cron.log.1が200GB専有していたので中身を確認したところ以上の文でうまっていました．
タブ補完が聞かないくらいに容量がなかったので応急処置としてcron.log.1（最終更新日が1/16 13:17）を削除しました．
どなたかcronを書き換えた方はいらっしゃいますか？

Naoto Tsukamoto, 1月19日, 14:48, 編集済み
Fetch1075にはatというコマンド自体がありませんでしたのでFetch15のatをpurgeしようとしました．するとlibrealsense2-udev-rules*が依存しているようでfetch@fetch15:/var/log$ sudo apt purge at
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-5.4.0-66-generic linux-hwe-5.4-headers-5.4.0-66 linux-image-5.4.0-66-generic
  linux-modules-5.4.0-66-generic linux-modules-extra-5.4.0-66-generic
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  at* librealsense2-udev-rules*
0 upgraded, 0 newly installed, 2 to remove and 559 not upgraded.
After this operation, 183 kB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
fetch@fetch15:/var/log$ dpkg -l | grep librealsense
ii  librealsense2:amd64                             2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) Cross Platform API - runtime
ii  librealsense2-dbg:amd64                         2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) Camera Capture API - debug symbols
ii  librealsense2-dev:amd64                         2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) Camera Capture API - development files
ii  librealsense2-dkms                              1.3.16-0ubuntu1                                  all          Modified kernel modules for librealsense2
ii  librealsense2-gl:amd64                          2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) - GLSL-enabled extensions
ii  librealsense2-gl-dbg:amd64                      2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) GLSL extensions - debug symbols
ii  librealsense2-gl-dev:amd64                      2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) Camera Capture API - development files
ii  librealsense2-net:amd64                         2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) - Network Bridge extension
ii  librealsense2-net-dbg:amd64                     2.43.0-0~realsense0.4277                         amd64        Intel(R) RealSense(tm) Network Bridge extension - debug symbols
[... cut ...]
at（アット）は、Unix系オペレーティングシステムのコマンドであり、任意のコマンドを任意の指定した時間に1回実行するようスケジュールする。ということなので、誰かが手動でatしない限り、二度と実行されない（つまりこの問題も再発しない）のでは？という気もしています

EDIT
上のコメントは間違っていて、atqは2021/12/9から定期的に（1時間に1回くらい）走っているんですね。
また再発する可能性は十分あるような気がします。

Yoshiki Obinata, 1月19日, 15:07
hitsory | grep at などでは怪しいコマンドは確認されませんでした．(Thu Dec 9とかなので，古すぎてヒットしないだけだと思いますが)

Naoya Yamaguchi, 1月19日, 15:10
ログのチェックありがとう、手動で実行されたわけではなさそうですね
再発防止策として、以下のようなスクリプトを実行するのが良いのかもしれませんね

https://gaso.hatenablog.com/entry/2014/03/25/Ubuntuサーバでdisk残量をチェックして警告メールを投げ

iory, 2022-09-15:
情報ありがとうございます。
cron.logは削除してみて容量をあけたあとに上の対処法をやってみましょうか。

ところでatはrealsenseとかで実行されてしまっているんだろうか

knorth55, 2022-09-15:
だいたいこういうのmongodbのreplication失敗だと身構えたけど、今回は違いそう？
atq?のcron出力を止めれないかな
BaxterやJAXONとかでも出てるんだろうか？

sktometometo, 2022-09-15:
これ systemd の `atd.service` をstop and disableすればいい話?

sktometometo, 2022-09-15:
`sudo systemctl stop atd.service` と `sudo systemctl disable atd.service` しました。

`/var/spool/cron/atjobs` にはjobがある, `/var/spool/cron/atspool` には何のjobもない。

```bash
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ sudo ls /var/spool/cron/atjobs/
[sudo] password for fetch: 
'=003d701a3af0a'  '=0055301a45573'  '=0055801a45573'   a0055201a45573   a0055601a45573
'=003d801a3af0a'  '=0055501a45573'   a003d701a3af0a    a0055301a45573   a0055701a45573
'=0055101a45573'  '=0055601a45573'   a003d801a3af0a    a0055401a45573   a0055801a45573
'=0055201a45573'  '=0055701a45573'   a0055101a45573    a0055501a45573
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ sudo cat /var/spool/cron/atjobs/a003d701a3af0a
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ sudo ls /var/spool/cron/atjobs/=0055801a45573
'/var/spool/cron/atjobs/=0055801a45573'
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ sudo cat /var/spool/cron/atjobs/=0055801a45573
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ sudo ls /var/spool/cron/at
atjobs/  atspool/ 
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ sudo ls /var/spool/cron/atspool/
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ 
```

sktometometo, 2022-09-15:
`atd.service`はFetch1075 でも同様のエラーを吐いているので、`stop/disable` しました。

```bash
sktometometo@fetch1075
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:/home/sktometometo/ros/ws_main]
[01:05 午後 木  9月 15]
~ $ sudo systemctl status atd.service 
[sudo] password for sktometometo: 
● atd.service - Deferred execution scheduler
   Loaded: loaded (/lib/systemd/system/atd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2022-09-15 04:01:14 JST; 9h ago
     Docs: man:atd(8)
 Main PID: 982 (atd)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/atd.service
           └─982 /usr/sbin/atd -f

 9月 15 13:01:46 fetch1075 atd[2537]: File a0081501a5eec7 is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2539]: File a0080601a5eebc is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2540]: File a0080501a5eebc is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2541]: File a0081501a5eec7 is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2542]: File a0081601a5eec7 is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2543]: File a0080601a5eebc is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2544]: File a0080501a5eebc is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2546]: File a0081601a5eec7 is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2545]: File a0081501a5eec7 is in wrong format - aborting
 9月 15 13:01:46 fetch1075 atd[2548]: File a0080501a5eebc is in wrong format - aborting
```

sktometometo, 2022-09-15:
atjobs のファイルの中身が空だったけど、これのせいでwrong format ってなっているっぽい
atq でスケジュールされた日付を見てももう関係無い日みたいだったので、atjobsの中身は消しました。

```bash
~ $ sudo atq - list jobs
2054	Fri Jul 29 23:52:00 2022 a root
2069	Sat Jul 30 00:03:00 2022 = root
2054	Fri Jul 29 23:52:00 2022 = root
2053	Fri Jul 29 23:52:00 2022 = root
2053	Fri Jul 29 23:52:00 2022 a root
2069	Sat Jul 30 00:03:00 2022 a root
2070	Sat Jul 30 00:03:00 2022 = root
2070	Sat Jul 30 00:03:00 2022 a root
```

sktometometo, 2022-09-15:
定期的に実行されているわけではなく、おそらくlibrealsense-udev-rules をインストールしたタイミングっぽいので、インストール時にatが走るようになっているとかでしょうか?

708yamaguchi, 2022-09-15:
確かに`librealsense2-udev-rules`はatに依存していそうですね
```
fetch@fetch15:~$ apt-cache depends librealsense2-udev-rules
librealsense2-udev-rules
  Depends: at
    at:i386
  Depends: rsync
    rsync:i386
```

sktometometo, 2022-09-15:
https://askubuntu.com/questions/74523/how-can-i-install-a-package-without-installing-some-dependencies

dependency をインストールせずにdebパッケージをインストールする方法もあるようですが、パッケージ管理でHackするよりsystemdの設定をちょっといじるほうが影響が少ないかなと思いますが、systemd disableする方法で問題のある方いらっしゃいますか?

sktometometo, 2022-09-15:
`librealsense2-udev-rules` のインストールのタイミングでは特に何も追加されてい無さそうです。

```bash
sktometometo@fetch1075
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:/home/sktometometo/ros/ws_main]
[01:57 午後 木  9月 15]
~ $ sudo apt purge at librealsense2-udev-rules
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-5.4.0-42-generic linux-hwe-5.4-headers-5.4.0-42 linux-image-5.4.0-42-generic linux-modules-5.4.0-42-generic
  linux-modules-extra-5.4.0-42-generic mbrola mbrola-en1 ros-melodic-ddynamic-reconfigure
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  at* librealsense2-udev-rules*
0 upgraded, 0 newly installed, 2 to remove and 799 not upgraded.
After this operation, 185 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 361305 files and directories currently installed.)
Removing librealsense2-udev-rules:amd64 (2.51.1-0~realsense0.7525) ...
Removing at (3.1.20-3.1ubuntu2) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
(Reading database ... 361278 files and directories currently installed.)
Purging configuration files for at (3.1.20-3.1ubuntu2) ...
Processing triggers for systemd (237-3ubuntu10.50) ...
Processing triggers for ureadahead (0.100.0-21) ...
[master 1e5a93f] committing changes in /etc after apt run
 Author: Koki Shinjo <sktometometo@gmail.com>
 12 files changed, 94 deletions(-)
 delete mode 100644 at.deny
 delete mode 100755 init.d/atd
 delete mode 100644 pam.d/atd
 delete mode 120000 rc0.d/K01atd
 delete mode 120000 rc1.d/K01atd
 delete mode 120000 rc2.d/S01atd
 delete mode 120000 rc3.d/S01atd
 delete mode 120000 rc4.d/S01atd
 delete mode 120000 rc5.d/S01atd
 delete mode 120000 rc6.d/K01atd
 delete mode 120000 systemd/system/multi-user.target.wants/atd.service
sktometometo@fetch1075
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:/home/sktometometo/ros/ws_main]
[01:58 午後 木  9月 15]
~ $ sudo apt install librealsense2-udev-rules
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-5.4.0-42-generic linux-hwe-5.4-headers-5.4.0-42 linux-image-5.4.0-42-generic linux-modules-5.4.0-42-generic
  linux-modules-extra-5.4.0-42-generic mbrola mbrola-en1 ros-melodic-ddynamic-reconfigure
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  at
The following NEW packages will be installed:
  at librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 799 not upgraded.
Need to get 47.3 kB of archives.
After this operation, 185 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://jp.archive.ubuntu.com/ubuntu bionic/main amd64 at amd64 3.1.20-3.1ubuntu2 [39.5 kB]
Get:2 https://librealsense.intel.com/Debian/apt-repo bionic/main amd64 librealsense2-udev-rules amd64 2.51.1-0~realsense0.7525 [7,786 B]
Fetched 47.3 kB in 1s (41.2 kB/s)              
Selecting previously unselected package at.
(Reading database ... 361276 files and directories currently installed.)
Preparing to unpack .../at_3.1.20-3.1ubuntu2_amd64.deb ...
Unpacking at (3.1.20-3.1ubuntu2) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.51.1-0~realsense0.7525_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.51.1-0~realsense0.7525) ...
Setting up at (3.1.20-3.1ubuntu2) ...
Created symlink /etc/systemd/system/multi-user.target.wants/atd.service → /lib/systemd/system/atd.service.
Setting up librealsense2-udev-rules:amd64 (2.51.1-0~realsense0.7525) ...

Postinst script activated...
Presets deployment:
no src dir
tgt dir exists
Presets deployment is skipped
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for ureadahead (0.100.0-21) ...
Processing triggers for systemd (237-3ubuntu10.50) ...
[master 57506bf] committing changes in /etc after apt run
 Author: Koki Shinjo <sktometometo@gmail.com>
 12 files changed, 94 insertions(+)
 create mode 100644 at.deny
 create mode 100755 init.d/atd
 create mode 100644 pam.d/atd
 create mode 120000 rc0.d/K01atd
 create mode 120000 rc1.d/K01atd
 create mode 120000 rc2.d/S01atd
 create mode 120000 rc3.d/S01atd
 create mode 120000 rc4.d/S01atd
 create mode 120000 rc5.d/S01atd
 create mode 120000 rc6.d/K01atd
 create mode 120000 systemd/system/multi-user.target.wants/atd.service
sktometometo@fetch1075
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:/home/sktometometo/ros/ws_main]
[01:58 午後 木  9月 15]
~ $ sudo atq
sktometometo@fetch1075
[gitbranch:]
[ROS VERSION:1]
[ROS workspace:/home/sktometometo/ros/ws_main]
[01:58 午後 木  9月 15]
~ $ 

```

iory, 2022-09-15:
logrotateなどでcron.logのサイズ上限を決めておくのはどうでしょうか。
もしかしたらlibrealsenseの依存には入っていたので、atがrealsenseに影響を与える可能性はあるかと思います。
https://qiita.com/shotets/items/e13e1d1739eaea7b1ff9

sktometometo, 2022-09-15:
`librealsense2-dkms` とか `librealsense2` ではなく `librealsense2-udev-rules` の依存に入っているのが奇妙ですよね。
Realsense周りで問題が置きたらIssueたてて下さい。チェックします。

sktometometo, 2022-09-18:
Fetch15のcron.logはバックアップを取ったので削除しました。

sktometometo, 2022-09-18:
容量問題そのものは解消し、atd.service の無効化により再発しなくなったはずなのでCloseします。
再発したり関連する問題が起きた場合はReopenするか関連付けて新しいIssue立てるかしてください。

closed 2022-09-18 by sktometometo

sktometometo, 2022-09-18:
Log は 共有ドライブの Movies/Fetch/fetch15 log dumps/20220915 以下に上げました。