jsk-ros-pkg/jsk_robot #1623: [PR1040] Realtime loopが遅くなる

tkmtnt7000, 2022-09-29:
# 症状
10分に１回の頻度でaudible warningが
```
[rosout][INFO] 2022-09-06 17:47:05,625: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 701.6667 hz
```
のようなエラーを言っている。9/6に続いて、9/28-29にかけてもこの問題が起きた。9/6のチャットは下に添付。
原因を調査中。

# 9/29に確認したこと
## htopとsensorsからCPU使用率、温度を確認
- 症状が出ているときのhtopとsensorsを確認すると以下の写真のようになっている。C1、C2ともにCPU使用率や温度は高くないが、C1ではメモリをほとんど使い果たしていてスワップ領域をすべて使っていることがわかる。
[image]
[image]
- `sudo initctl stop robot`をした後のhtopとsensorsの写真。メモリやスワップ領域がプログラムの終了で解放されたことがわかる。
[image]
[image]

### 補足資料

9/6(火)のPR2チャットより
-------------------
Naoto Tsukamoto, 9月6日, 18:15,
ちょっとあまりわかってないんですが，audible_warningで10分に一回の頻度で
```
[rosout][INFO] 2022-09-06 17:47:05,625: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 701.6667 hz
```
と喋っているので報告しておきます．realtime loopがおかしくなっているんですかね

Naoto Tsukamoto, 9月6日, 18:18

Iori Yanokura, 9月6日, 18:22
関係なさそうだけどkinect関係も死んでいる
ので、一回initctl restart robotしてみてもrealtime loopのエラーが出るか確認してもらえるといいです

Naoto Tsukamoto, 9月6日, 18:23
わかりました．誰も使っていなさそうなのでやってみます．

Shingo Kitagawa, 9月6日, 18:52
なにか処理が重そう？
htopとかsensorsとかで負荷や温度をみてみるとよいかもね

Naoto Tsukamoto, 9月6日, 18:56

[image]
[image]

Naoto Tsukamoto, 9月6日, 18:57, 編集済み
遅くなりましたが，sudo initctl stop/start robotで一応直ってそうです．
一応直ってしまった後ですけど貼っておきました．

Shingo Kitagawa, 9月6日, 19:00
現状は問題なさそうですね

Shingo Kitagawa, 9月6日, 19:00, 編集済み
またあったときは見てみてください

Naoto Tsukamoto, 9月6日, 19:01
はい．次起きたら今回の分と比較できそうです．

tkmtnt7000, 2022-09-29:
cc: @708yamaguchi @nakane11

knorth55, 2022-09-29:
もしかしたら，audio_playのメモリーリークかも？
https://github.com/ros-drivers/audio_common/issues/205

audio_playを最新にしてみました．

nakane11, 2022-10-02:
このエラーが現在も10分おきに起きていて、logを見ると10/2 13:04頃からのようです。
17:11時点でのhtopとsensorsです。
[image]
[image]

iory, 2022-10-02:
毎回725Hzになっているのが気になりますね
```
root@pr1040:~# sudo cat /var/log/upstart/jsk-pr2-startup.log | grep audible
[INFO] [1664683475.002465] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664683475.002853] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664684076.007752] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664684076.008389] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop realtime loop used too much time in the last 30 seconds.; halting, realtime loop only ran at 725.0000 hz
[INFO] [1664684677.002995] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664684677.003662] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop realtime loop used too much time in the last 30 seconds.; halting, realtime loop only ran at 725.0000 hz
[INFO] [1664685278.008647] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664685278.009132] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664685879.012320] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664685879.012986] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664686480.025976] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664686480.027511] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664687081.003305] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664687081.003796] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664687682.005253] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664687682.005679] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664688283.010285] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664688283.010756] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664688884.006588] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664688884.007046] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop realtime loop used too much time in the last 30 seconds.; halting, realtime loop only ran at 725.0000 hz
[INFO] [1664689485.001044] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664689485.001408] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664690086.020079] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664690086.020651] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664690687.013863] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664690687.014340] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664691288.014715] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664691288.015344] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664691889.005943] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664691889.006386] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664692490.007370] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664692490.007854] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664693091.009920] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664693091.010481] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[... cut ...]
[INFO] [1664700904.003645] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664700904.004161] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
[INFO] [1664701505.012168] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1664701505.012541] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 725.0000 hz
```

knorth55, 2022-10-02:
HtopをMEMORYやVIRT、RES、SHRでソートしてみてください

nakane11, 2022-10-02:
## MEMORY
[image]
## VIRT
[image]
## RES 
[image]
## SHR
[image]

708yamaguchi, 2022-10-02:
@nakane11 
すいません
北川さんのコメントに気づかずに、僕と矢野倉さんでinitctl stop/start robotをしてしまいまいした。
なので、中根さんの撮ってくれたスクリーンショットは問題が発生している瞬間のものではないかもしれないです。

knorth55, 2022-10-02:
そうだね…

nakane11, 2022-10-03:
10/3 23:54のhtop 問題が発生して数十分後
#### CPU
[image]
#### MEM
[image]
#### VIRT
[image]
#### RES
[image]
#### SHR
[image]

708yamaguchi, 2022-10-04:
石田さんの調べてくださっていたrealtime loopの話について

> 昨日の23時頃からrealtime loop関連のwarningが出ていたようです.　https://gist.github.com/HiroIshida/1378c04e096246d45178af3680430d1c
> 因果関係は明らかではありませんが, サーボがつくかどうかチェックするためにサーボを入れたのが5時すぎだったと思うのですが, それ以降audible warningはでなくなりました. (distにあるコマンドの実行は5:30頃です). ちなみに, サーボは入ってロボットは問題なく動いています.

先ほどPR1012でも同じwarningが出たのでサーボON/OFFをしてみたところ、サーボOFFした瞬間にwarningが消えて、再びサーボONしてもwarningがでなくなりました
[image]

708yamaguchi, 2022-10-04:
サーボOFF状態でもON状態でもこの問題は発生していて、
- サーボOFF状態だと、このwarningが出てもHzが更新されず、同じHzのwarningが出続ける（多分、https://github.com/jsk-ros-pkg/jsk_robot/issues/1623#issuecomment-1264591444 と同じ）
- サーボON状態だと、Hzが更新されているようで、warningが出たり消えたりする

708yamaguchi, 2022-10-06:
昨日からずっとサーボオフで放置していたPR1012で、realtime loopが10Hzくらいしか出ていませんでした。（本来は1000Hz）
原因の推測などは出来ていませんが、異様にループが遅かったので報告します。

[image]