jsk-ros-pkg/jsk_robot #1626: [Fetch1075] jsk-fetch-startup の supervisorconf が起動しない

sktometometo, 2022-09-29:
`jsk_fetch_startup` の supervisorconf が立ち上がらない。
ログは

```
waiting for roscore to come up
set ROS_IP to "" and ROS_HOSTNAME to "fetch1075"
Failure: No such entity
Failed to get sink information: No such entity
master has started, initiating launch
... logging to /home/fetch/.ros/log/a6832ab2-3f98-11ed-8ab9-890df9f1bcd6/roslaunch-fetch1075-20244.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/fetch/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

RLException: error loading <rosparam> tag: 
        file does not exist [/home/fetch/ros/melodic/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup/config/fetch1075_dialogflow_hotword.yaml]
XML is <rosparam command="load" file="$(find jsk_fetch_startup)/config/$(arg hostname)_dialogflow_hotword.yaml" ns="/dialogflow_client/hotword"/>
The traceback for the exception was written to the log file
```

`fetch` ユーザーのワークスペースコンフィギュレーションは

```
fetch@fetch1075:~/ros/melodic/src/jsk-ros-pkg/jsk_robot/jsk_fetch_robot/jsk_fetch_startup$ wstool info
/usr/lib/python2.7/dist-packages/wstool/config_yaml.py:74: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  yamldata = yaml.load(stream)
/usr/lib/python2.7/dist-packages/vcstools/tar.py:168: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  metadata = yaml.load(metadata_file.read())
/usr/lib/python2.7/dist-packages/vcstools/tar.py:80: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  metadata = yaml.load(metadata_file.read())
workspace: /home/fetch/ros/melodic/src

 Localname                          S SCM Version (Spec)                                                  UID  (Spec)                                                     URI  (Spec) [http(s)://...]
 ---------                          - --- --------------                                                  -----------                                                     ---------------------------
 BehaviorTree.ROS                     git master                                                          59cd301ad48c                                                    github.com/BehaviorTree/BehaviorTree.ROS
 roseus_resume                        git eus10                                                           99e80a591537                                                    github.com/Affonso-Gui/roseus_resume.git
 jsk-ros-pkg/geneus                   git master                                                          ec388e279ce4                                                    github.com/jsk-ros-pkg/geneus.git
 euslisp/jskeus                       git master                                                          3040a1aa1cce                                                    github.com/euslisp/jskeus.git
 jsk-ros-pkg/jsk_roseus               git fetch15                                                         c13459fd04d3                                                    github.com/knorth55/jsk_roseus.git
 euslisp/Euslisp                      git eus10                                                           6851cbb6d607                                                    github.com/Affonso-Gui/EusLisp.git
 jsk-ros-pkg/coral_usb_ros            git master                                                          547651210749                                                    github.com/jsk-ros-pkg/coral_usb_ros.git
 ros-planning/navigation              git fetch15                                                         9babf61c6d64                                                    github.com/708yamaguchi/navigation.git
 fetchrobotics/fetch_open_auto_dock   git fetchdev                                                        eac866e6cea5                                                    github.com/sktometometo/fetch_open_auto_dock.git
 IntelRealSense/realsense-ros         git fetch15                                                         e155795440b2                                                    github.com/Affonso-Gui/realsense.git
 GT-RAIL/robot_pose_publisher         git <detached>  (0.2.4)                                             b51d2d070bfb                                                    github.com/GT-RAIL/robot_pose_publisher.git
 tork-a/visualization_rwt             git fetch15                                                         4a77d8404313                                                    github.com/knorth55/visualization_rwt.git
 tork-a/roswww                        git develop                                                         c4b4d43ee703                                                    github.com/tork-a/roswww.git
 strands-project/mongodb_store        git fetch15                                                         269d642df964                                                    github.com/708yamaguchi/mongodb_store.git
 ros/common_msgs/visualization_msgs   tar common_msgs-release-release-kinetic-visualization_msgs-1.12.7-0 common_msgs-release-release-kinetic-visualization_msgs-1.12.7-0 github.com/ros-gbp/common_msgs-release/archive/release/kinetic/visualization_msgs/1.12.7-0.tar.gz
 ros-perception/slam_gmapping         git <detached>  (1.3.10)                                            5a707e079364                                                    github.com/ros-perception/slam_gmapping.git
 ros-drivers/rosserial                git fetch15                                                         90f24a789552                                                    github.com/708yamaguchi/rosserial.git
 ros-drivers/audio_common             git master                                                          127912fda4d5                                                    github.com/ros-drivers/audio_common.git
 mikeferguson/robot_calibration       git <detached>  (0.5.5)                                             27b8193570d6                                                    github.com/mikeferguson/robot_calibration.git
 knorth55/app_manager_utils           git master                                                          f700a7ac9136                                                    github.com/knorth55/app_manager_utils
 jsk-ros-pkg/jsk_robot                git fetch15                                                         1ed18fdcb97f                                                    github.com/knorth55/jsk_robot.git
 jsk-ros-pkg/jsk_recognition          git develop/fetch                                                   85ada153bb6b                                                    github.com/sktometometo/jsk_recognition.git
 jsk-ros-pkg/jsk_pr2eus               git master                                                          4700cf84db91                                                    github.com/jsk-ros-pkg/jsk_pr2eus.git
 jsk-ros-pkg/jsk_demos                git fetch15                                                         815c5c3ad7d4                                                    github.com/knorth55/jsk_demos.git
 jsk-ros-pkg/jsk_common               git fetch15                                                         255d46c756c6                                                    github.com/knorth55/jsk_common.git
 jsk-ros-pkg/jsk_3rdparty             git fetch15                                                         662f0b55c6e6                                                    github.com/knorth55/jsk_3rdparty.git
 fetchrobotics/robot_controllers      git melodic-devel                                                   010855f0e6a8                                                    github.com/fetchrobotics/robot_controllers.git
 fetchrobotics/fetch_ros              git melodic-devel                                                   9c38de60834e                                                    github.com/fetchrobotics/fetch_ros.git
 RoboticMaterials/FA-I-sensor         git master  (-)                                                     ca3e80b00429                                                    github.com/RoboticMaterials/FA-I-sensor.git
[... cut ...]
 PR2/app_manager                      git fetch15                                                         cfa5a0704b38                                                    github.com/knorth55/app_manager.git

Also detected these repositories in the workspace, add using 'wstool scrape' or 'wstool set':

 Localname                       SCM   URI  (Spec) [http(s)://...]
 ---------                       ---   ---------------------------
 locusrobotics/catkin_virtualenv --git https://github.com/locusrobotics/catkin_virtualenv.git
```

`jsk_robot` のブランとが `jsk-ros-pkg` の `develop/fetch` でなくて `knorth55` の `fetch15` になっているせい?

knorth55, 2022-09-29:
`develop/fetch`になってなかった？

sktometometo, 2022-10-05:
Seems to be solved.

closed 2022-10-05 by sktometometo