jsk-ros-pkg/jsk_robot #1638: [PR1012] バッテリ充電制御ボードが焼けた

knorth55, 2022-10-04:
バッテリ充電制御ボードのグラウンドをはんだ付けして修理した際に，どこかがショートしていたため，ボードが焼けてしまった．
ボードから火が出たが電池の制御回路リレーが壊れたため，火はとまった．
燃えた回路ボードにつないでいたスマートバッテリは壊れてしまった．

cc. @708yamaguchi @iory @nakane11

knorth55, 2022-10-04:
現在LikeABird社の新しい充電制御ボードをためしている．

708yamaguchi, 2022-10-04:
PR1012ですが、1系統だけ新しいバッテリ制御基板に入れ替えた状態で、
・バッテリの充電
・AC充電時の動作
・AC非充電時の動作（バッテリ駆動で1分くらい動くことの確認。長時間動くかはまだ不明）
ができることを確認しました。

なので、電気回路として必要な機能は復活したように思います。

knorth55, 2022-10-05:
PR2に電池管理ボードを治具で固定せずに今の状態で長時間動かせるかテストしたいですね
移動はもちろんできませんが、サーボを入れることはできそう

少し怖い点は、PowerBoardの故障原因が特定できていないところです
新しいPowerBoardを壊したくないので、一旦元に戻して電池管理ボードが十分安全に動くか確認してから、新しいのに取り替えたほうがいいと思います、手間ですが
もし長時間動かしてると、電池管理ボードがたまに電流や電圧が不安定になるとかがあったら、それで新しいPowerBoardを壊す可能性があります
あれはマジで替えがない慎重に扱いたいです

708yamaguchi, 2022-10-05:
> PR2に電池管理ボードを治具で固定せずに今の状態で長時間動かせるかテストしたいですね
> 移動はもちろんできませんが、サーボを入れることはできそう

昨日は、ACを繋いだまま30分くらいサーボを入れっぱなしにしていましたが、PR1012の電源が落ちたりはしませんでした。
（EtherCAT Masterのパケットドロップが発生していたため、たまに首がガクガクしてはいました）
今日から、より長期間のテストをしていきたいです。

> 新しいPowerBoardを壊したくないので、一旦元に戻して電池管理ボードが十分安全に動くか確認してから、新しいのに取り替えたほうがいいと思います、手間ですが

元のPower Boardがホコリの掃除で直ったかの確認も兼ねて、一旦元のPower Boardに取り替えるのは良さそうですね。
1週間くらい様子をみて電池管理ボードに不具合がない、かつ、元のPower Boardが改善されていないという状態だったら、新しいPower Boardをしかたなく使う、という感じでしょうか。

knorth55, 2022-10-05:
> 1週間くらい様子をみて電池管理ボードに不具合がない、かつ、元のPower Boardが改善されていないという状態だったら、新しいPower Boardをしかたなく使う、という感じでしょうか。

そうしましょう

knorth55, 2022-10-19:
PR1012ではLikeAbirdのBMCボードで動くことを確認した．
バッテリ充電制御ボードは互換品で動くことを確認したのでIssueをCloseします．

追記：このBMCボードを追加発注する必要があるが，まだしていない．

closed 2022-10-19 by knorth55