jsk-ros-pkg/jsk_robot #1639: [PR1012] Right shoulderのMCBのrx_errorが多くてサーボががくがくする

knorth55, 2022-10-04:
right shoulderのMCBのrx_errorが多いためかサーボががくがくしてしまう．
right shoulder tiltのmcbを新しいMCBに変更した？（panだっけ？）

@708yamaguchi @iory @nakane11

708yamaguchi, 2022-10-04:
修理ログです
https://docs.google.com/presentation/d/1BfNpDABFXVAvmg5MGEGX4T-JvfMy-w1FoQza1Ui6v5M/edit#slide=id.p

knorth55, 2022-10-18:
`shoulder_lift` `port0`と`shoulder_pan` `port0`でもまだ少し出ている．
`shoulder_lift`のMCBを替えたがそれで正しかったのか？
けど，もとよりは`rx error`の数は減っている気がする．
色々な問題が併発していたので問題が切り分けられていなかった？
`shoulder_pan` -> `shoulder_lift`の経路に問題がある，MCBを替えて改善したということは`shoulder_lift` MCBのRJ４５コネクタの接触不良？
とすると`shoulder_pan` MCBのRJ45コネクタも接触不良なのではないか？
`shoulder_list` MCBの接触不良は直せないか？

```
/Motors/EtherCAT Device #10 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #10 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #10 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #10 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_gripper_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_wrist_l_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_wrist_l_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_wrist_r_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_wrist_r_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (laser_tilt_mount_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (laser_tilt_mount_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (led_projector) RX Error Port 0: 0
/Motors/EtherCAT Device (led_projector) RX Error Port 1: 0
/Motors/EtherCAT Device (r_elbow_flex_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_elbow_flex_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_forearm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_forearm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_gripper_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 0: 367
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 1: 1
/Motors/EtherCAT Device (r_upper_arm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_upper_arm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_wrist_l_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_wrist_l_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (torso_lift_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (torso_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Master RX Errors: 43
[Max Device]
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 0: 367
```

nakane11, 2023-03-11:
最近もサーボががくがくすることが多いです

robot.logで15分ほど`readMailboxInternal : ERROR asking for read repeat after dropping 1 packets`が出ている間、以下のように`r_shoulder_lift_motor`の`rx error`が増えていることを確認しました

- エラーが出はじめたとき
```
[Eroor Exist Dvice]
/Motors/EtherCAT Device #03 (WG014) RX Error Port 0: 1
/Motors/EtherCAT Device #03 (WG014) RX Error Port 2: 54
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 255
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 45751
/Motors/EtherCAT Device (r_gripper_motor) RX Error Port 0: 8
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 0: 42338
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 1: 4126
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 1: 2523
```
- エラーが止まりサーボががくがくしなくなった後
```
[Eroor Exist Dvice]
/Motors/EtherCAT Device #03 (WG014) RX Error Port 0: 1
/Motors/EtherCAT Device #03 (WG014) RX Error Port 2: 54
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 255
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 45751
/Motors/EtherCAT Device (r_gripper_motor) RX Error Port 0: 8
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 0: 42649
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 1: 4126
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 1: 2523
```

knorth55, 2023-03-11:
`r_shoulder_lift_motor`は2022年9月21日に交換しているんだけど，それがまた悪いというのは不思議ですね．
交換したときに少し改善したとは思うのですが，それでもまだがくがくしているとなると，`r_shoulder_lift_motor`以外のMCBにも問題がないかチェックしたくなりますね．

とりあえず `r_shoulder_pan_motor` -> `r_shoulder_lift_motor`の間のEthernetケーブルを前回は変えていなかったと思うので，そこを変えてみるのはありかもです．
`r_shoulder_pan_motor`の`port1`で値がかわっていないのに，`r_shoulder_lift_motor`の`port0`で値が増えているので，`r_shoulder_lift_motor`の`port0`のEthernetケーブルの口かコネクタの接触不良という可能性もあります．

nakane11, 2023-03-12:
`r_shoulder_lift_motor`の`port 0`に加えて`r_shoulder_pan_motor`の`port 1`と`r_gripper_motor`もだめです
サーボが落ちた後右手を動かしていると入ります
```
[Eroor Exist Dvice]
/Motors/EtherCAT Device #03 (WG014) RX Error Port 0: 2
/Motors/EtherCAT Device #03 (WG014) RX Error Port 2: 54
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 255
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 147371
/Motors/EtherCAT Device (r_gripper_motor) RX Error Port 0: 17884
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 0: 127082
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 1: 2
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 1: 9954
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 1: 2782
```
```
[Eroor Exist Dvice]
/Motors/EtherCAT Device #03 (WG014) RX Error Port 0: 2
/Motors/EtherCAT Device #03 (WG014) RX Error Port 2: 54
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 255
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 147371
/Motors/EtherCAT Device (r_gripper_motor) RX Error Port 0: 23071
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 0: 127392
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 1: 2
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 1: 10452
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 1: 2782
```

k-okada, 2023-03-15:
これって思ったんだけど，ハードウェア原因もあると思うんだけど，OSがリアルタイムでないから周期が守られずに，落としている，という事はないかな？

問題の切り分け方法がむつかしそうだけど，できる限り立ち上がっているノードを落として，CPUの負荷を最大限下げてみる，ということなのかな．
ただ，OSが問題だったら，特定のドライバだけで落ちるという事はないような気もするし，チェックしている`for j in joints:`　の jointsに入っている順番で特定のだけ落ちるという事もあるのかな．．．

nakane11, 2023-03-15:
突然サーボが落ちる以外に最近多い症状をメモしておきます.

- **キャリブレーション中にサーボが落ちて「runstopをオンにしてください」と言われる**
`/power_board/state`のrunstopは反応するが`/pr2_ethercat/motors_halted`がFalseにならない
腕をしばらく動かすと治る
https://github.com/PR2/pr2_ethercat_drivers/blob/b53698eaadbcb020c939be866eabba2d17280aa1/ethercat_hardware/src/ethercat_hardware.cpp#L317-L325

- **キャリブレーションが終わった直後にサーボが落ちる**
このときimuの値を受け取るときにチェックサムのエラーが発生している
[CPU負荷の問題？](https://github.com/ros-drivers/microstrain_3dmgx2_imu/issues/5#issuecomment-72483021)
腕を動かしてもサーボが付かないことが多い気がする
```
[INFO] [1678504460.448833] [/calibrate_pr2:rosout]: Bringing down calibration node
[INFO] [1678504460.864305] [/calibrate_pr2:rosout]: Calibration completed in 115.686224 sec
[WARN] [1678504461.128410] [/calibrate_pr2:rosout]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
Robot is successfully started.
[ INFO] [1678504461.630496848] [/realtime_loop:ros.pr2_mechanism_controllers]: Trajectory:: interpolation type linear
[ INFO] [1678504461.630588829] [/realtime_loop:ros.pr2_mechanism_controllers]: LaserScannerTrajController: Periodic Command set. Duration=1.0000 sec
[ INFO] [1678504461.641517406] [/realtime_loop:ros.pr2_mechanism_controllers]: Successfully spawned 
[INFO] [1678504462.497580] [/default_controllers_spawner:rosout]: Loaded controllers: base_controller, base_odometry, head_traj_controller, laser_tilt_controller, torso_controller, r_gripper_controller, l_gripper_controller, r_arm_controller, l_arm_controller
[INFO] [1678504462.500788] [/default_controllers_spawner:rosout]: Started controllers: base_controller, base_odometry, head_traj_controller, laser_tilt_controller, torso_controller, r_gripper_controller, l_gripper_controller, r_arm_controller, l_arm_controller
[ INFO] [1678504462.512538470] [/robot_pose_ekf:ros.robot_pose_ekf]: Initializing Odom sensor
[ INFO] [1678504462.533892087] [/robot_pose_ekf:ros.robot_pose_ekf]: Odom sensor activated
[ INFO] [1678504462.566248308] [/robot_pose_ekf:ros.robot_pose_ekf]: Kalman filter initialized with odom measurement
[l_arm_controller_loose/traj_lock-54] process has finished cleanly
log file: /var/log/ros/20230311-121205_80651e06-bfba-11ed-8592-d05099c20283/l_arm_controller_loose-traj_lock-54*.log
[head_traj_controller_loose/traj_lock-56] process has finished cleanly
log file: /var/log/ros/20230311-121205_80651e06-bfba-11ed-8592-d05099c20283/head_traj_controller_loose-traj_lock-56*.log
Unhandled exception in thread started by 
sys.excepthook is missing
lost sys.stderr
[r_arm_controller_loose/traj_lock-55] process has finished cleanly
log file: /var/log/ros/20230311-121205_80651e06-bfba-11ed-8592-d05099c20283/r_arm_controller_loose-traj_lock-55*.log
[ WARN] [1678504470.499911982] [/imu_node:ros.microstrain_3dmgx2_imu]: Gathering data took 113.519907 ms. Nominal is 10ms.
[ WARN] [1678504470.606545600] [/imu_node:ros.microstrain_3dmgx2_imu]: Exception thrown while trying to get the IMU reading. This sometimes happens due to a communication glitch, or if another process is trying to access the IMU port. You may try 'lsof|grep /etc/ros/sensors/imu' to see if other processes have the port open. invalid checksum.
 Make sure the IMU sensor is connected to this computer. (in microstrain_3dmgx2_imu::IMU:receive)
[right_endeffector/virtual_force_publisher-36] process has finished cleanly
log file: /var/log/ros/20230311-121205_80651e06-bfba-11ed-8592-d05099c20283/right_endeffector-virtual_force_publisher-36*.log
[left_endeffector/virtual_force_publisher-35] process has finished cleanly
log file: /var/log/ros/20230311-121205_80651e06-bfba-11ed-8592-d05099c20283/left_endeffector-virtual_force_publisher-35*.log
[ INFO] [1678504472.651980900] [/imu_node:ros.microstrain_3dmgx2_imu]: Connected to IMU [    Inertia-Link] model [            4200] s/n [            3039] options [       5g 300d/s]
[ INFO] [1678504472.652063641] [/imu_node:ros.microstrain_3dmgx2_imu]: Initializing IMU time with offset -0.040000.
[ INFO] [1678504472.662989959] [/imu_node:ros.microstrain_3dmgx2_imu]: Not calibrating the IMU sensor. Use the calibrate service to calibrate it before use.
[ INFO] [1678504472.663093308] [/imu_node:ros.microstrain_3dmgx2_imu]: IMU sensor initialized.
[ WARN] [1678504473.481219250] [/imu_node:ros.microstrain_3dmgx2_imu]: Gathering data took 114.671946 ms. Nominal is 10ms.
[ WARN] [1678504473.855132147] [/imu_node:ros.microstrain_3dmgx2_imu]: Exception thrown while trying to get the IMU reading. This sometimes happens due to a communication glitch, or if another process is trying to access the IMU port. You may try 'lsof|grep /etc/ros/sensors/imu' to see if other processes have the port open. invalid checksum.
 Make sure the IMU sensor is connected to this computer. (in microstrain_3dmgx2_imu::IMU:receive)
[ INFO] [1678504476.749409354] [/imu_node:ros.microstrain_3dmgx2_imu]: Connected to IMU [    Inertia-Link] model [            4200] s/n [            3039] options [       5g 300d/s]
[ INFO] [1678504476.749638695] [/imu_node:ros.microstrain_3dmgx2_imu]: Initializing IMU time with offset -0.040000.
[ INFO] [1678504476.754357008] [/imu_node:ros.microstrain_3dmgx2_imu]: Not calibrating the IMU sensor. Use the calibrate service to calibrate it before use.
[ INFO] [1678504476.754572012] [/imu_node:ros.microstrain_3dmgx2_imu]: IMU sensor initialized.
[... cut ...] rt. You may try 'lsof|grep /etc/ros/sensors/imu' to see if other processes have the port open. invalid checksum.
 Make sure the IMU sensor is connected to this computer. (in microstrain_3dmgx2_imu::IMU:receive)
[ INFO] [1678504480.798849282] [/robot_pose_ekf:ros.robot_pose_ekf]: Odom sensor not active any more
[ INFO] [1678504480.803129436] [/robot_pose_ekf:ros.robot_pose_ekf]: Initializing Odom sensor
[ INFO] [1678504480.803429271] [/robot_pose_ekf:ros.robot_pose_ekf]: Odom sensor activated
```

knorth55, 2023-03-15:
@k-okada  

> ただ，OSが問題だったら，特定のドライバだけで落ちるという事はないような気もするし，チェックしているfor j in joints:　の jointsに入っている順番で特定のだけ落ちるという事もあるのかな．．．

これだとすると，あるボードだけ増えていくというよりかは，各ボードにランダムで発生しそうな気がします．
ただMCBのハードウェアをうたがうまえに，ソフトウェアをチェックするべきかと思います．