jsk-ros-pkg/jsk_robot #1641: [pr1040] packet dropによりサーボが入らない. diagnosticからEthecat deviceから読めていない.

HiroIshida, 2022-10-04:
[image]

knorth55, 2022-10-05:
PowerCycleを信じるとすると接触不良かな
背中のあたりの接触を今度確認します

#0 Slaveが背中のEcatハブっていうのはどうやってわかりましたか？
Ecatハブはベースの方にもあるけど、どう区別をつけたんだろう？

HiroIshida, 2022-10-05:
すみません WG014が首根っこの方だと考えたのは勘違いでした..

knorth55, 2022-10-05:
> すみません WG014が首根っこの方だと考えたのは勘違いでした..

Slave #0 WG014がどのデバイスかは僕も確認してないのですが，どちらかわかっていないということですね．
ベースと背中にあるので，どちらもみてみましょう．
#1640 だとするとすべてのSlaveがダメになる気がするので，どうなんだろう．
今回のはSlave #0 WG014がPowerCycleしてしまっているから，原因は違いそうな気がします．

Kanazawanaoaki, 2022-11-03:
10/27(https://github.com/jsk-ros-pkg/jsk_robot/issues/1640#issuecomment-1283620642 の後)にも昼過ぎにもこちらと同じ問題が発生した可能性があるのでこちらで報告して起きます．  

pr2のサーボを入れようとしたらTimestamp jumpedのようなエラーが出てサーボ入らないという事がありました．パケットドロップも各デバイスでのrx errorも増えてはおらず，masterのRX Errorsは増えていました（[要素](http://wiki.ros.org/pr2_etherCAT/Tutorials/PacketLoss#Determining_which_problem_you_have )的にはCRC Errors）．  

ダッシュボード
[image]

ログ
[20221027-1323-robot.log](https://drive.google.com/file/d/1Y1BfISyOY-2V8S-g-YYcb640II5FM3DR/view?usp=share_link )
[20221027-1323-jsk-pr2-startup.log](https://drive.google.com/file/d/11fdyslXxSuwoz9uLZmDZxWcmDO1RiGhZ/view?usp=share_link )
rx_errorチェックの[csv](https://drive.google.com/file/d/1eKrSGUeADcxVz__B-BBOfZsF2eLI_G_D/view?usp=share_link )
packet dropのチェックの[log](https://drive.google.com/file/d/1GN-3HiANdxI18JPcU8o8KS9DDfmjwp5H/view?usp=share_link )
[それらのデータが入ったフォルダ](https://drive.google.com/drive/folders/1zuX54ChbLpxS5CcW9s5RSmTuhDrPmG-0?usp=share_link )

robotを立ち上げ直しましたが，エラーが出て上手く立ち上がらず，ブレーカーから立ち上げ直すとサーボが入るようになりました．

[robot立ち上げ直し後のログ](https://drive.google.com/file/d/1S2ugsODb_68dfvc-YOARNagM4aN5pGnb/view?usp=share_link )を見ると，`Perhaps you should power-cycle the MCBs`と`Unable to configure slave #0`がありました．
```
$ grep -A10 -B10 MCB 20221027-1331-robot.log 
low_level_txandrx: failed 10 times: Giving up
EC_AL::scan_slaves() Error reading Product code of slave 36
low_level_txandrx: failed 10 times: Giving up
EC_AL::scan_slaves() Error reading Revision of slave 36
low_level_txandrx: failed 10 times: Giving up
EC_AL::scan_slaves() Error reading Serial of slave 36
low_level_txandrx: failed 10 times: Giving up
Something went wrong while resetting slaves
EtherCAT_AL:: Can't init network
[FATAL] [1666844988.913292061] [/realtime_loop:ros.ethercat_hardware]: Note: 0xBADDBADD indicates that the value was not read correctly from device.
[FATAL] [1666844988.913338493] [/realtime_loop:ros.ethercat_hardware]: Perhaps you should power-cycle the MCBs
[FATAL] [1666844988.913359524] [/realtime_loop:ros.ethercat_hardware]: Unable to configure slave #0
[realtime_loop-2] process has died [pid 3428, exit code 1, cmd pr2-grant /opt/ros/indigo/lib/pr2_ethercat/pr2_ethercat -i ecat0 -x robot_description /diagnostics:=/diagnostics /diagnostics_agg:=/diagnostics_agg __name:=realtime_loop __log:=/var/log/ros/dd06cb98-55af-11ed-ae6d-d05099c29feb/realtime_loop-2.log].
log file: /var/log/ros/dd06cb98-55af-11ed-ae6d-d05099c29feb/realtime_loop-2*.log
[ERROR] [1666845035.223406784] [/pr2_run_stop_auto_restart:ros.pr2_run_stop_auto_restart]: Could not find reset motors service
[ERROR] [1666845108.222231781] [/pr2_run_stop_auto_restart:ros.pr2_run_stop_auto_restart]: Could not find reset motors service
```

HiroIshida, 2022-11-03:
先程, pr2のサーボをONにしたところ, 金沢くんが上で報告してくれたのと同様のエラーが出てサーボが落ちましたので, 報告致します. 

[image]

金沢くんscriptで各デバイスでの packet dropを見てみると, 今回もethercat masterだけでpackage dropがおきていることがわかります. [tmp.txt](https://github.com/jsk-ros-pkg/jsk_robot/files/9931187/tmp.txt)

jsk_pr2_startup.log とrobot.logを以下に添付しておきます. 
[jsk-pr2-startup.txt](https://github.com/jsk-ros-pkg/jsk_robot/files/9931253/jsk-pr2-startup.txt)
[robot.txt](https://github.com/jsk-ros-pkg/jsk_robot/files/9931254/robot.txt)

なお, 15分間ほどサーボをOFFにした後にもう一度ONにするとサーボが入るようになりました.

softyanija, 2022-11-15:
2022/11/15にPR1040のサーボを入れようとしたところ、ガタガタと腕が震えてその後サーボが入らなくなったので報告いたします。dashboardは以下のようになっていました。
[image]
robot.logを以下に添付します。
[robot.txt](https://drive.google.com/file/d/150_x8eHRqb1c_lUKAfCBOWl346f2gMqZ/view?usp=share_link)

sudo initctl stop robot -> sudo initctl start robot とするとハングしてしまったのでブレーカーを落として立ち上げ直したところ正常に起動しました。