jsk-ros-pkg/jsk_robot #1644: [PR1040] 2022.10.05 PR1040 C19コネクタ・プラグ焼損

708yamaguchi, 2022-10-05:
PR2のC19コンセント・プラグが焼損していることを発見
熱を持っていたので確認したら焼損していた by 北川
刺さった状態で熱をもっていたので接点の抵抗が大きくなっている
起きた可能性のある現象として
1. 何らかの原因により大電流が流れた -> 接点でのジュール熱が大きくなり焼損&接点抵抗の増大
2. HotPlugのように小さい接点で電流を流した -> 接点でのジュール熱が大きくなり焼損&接点抵抗の増大

1は起きたのであれば本体の電源系のどこかでショートが起きて動かなくなっているはず、起きていないっぽいので、2が起きた現象らしい

修理ログ
https://docs.google.com/presentation/d/1DYZBU0xUyvClG_n-8ttbGXEXjrm5U81kKoDpODuo5M8/edit#slide=id.p

knorth55, 2022-10-19:
ケーブル側はあたらしいものに交換した．
プラグ側も購入したので，交換する必要がある．今後やる．

Kanazawanaoaki, 2025-03-10:
この問題が2024年にも発生しており，2024.12.20に調査した際にPR2の台車裏のオス側のコネクタ端子（プログ側）が破損していて触れるとグラグラする状態になっていたことを発見しました．
この破損が原因でケーブルの温度がかなり上昇し，ケーブルが耐えられる以上の120℃近い高温になっていた可能性が高いです．

その日の内にプラグを新しいものに交換し，コネクタが高温になることも焼損することもなくなりました．
修理ログ：https://docs.google.com/presentation/d/16wheDLVARzwiiKowiKasskDDhdPviCLxcOMN36elM7k/edit#slide=id.p

closed 2025-03-10 by Kanazawanaoaki