jsk-ros-pkg/jsk_robot #1665: [Fetch1075] path-constraints を追加すると自己干渉するようになる

sktometometo, 2022-10-13:
詳細は後ほど記載

knorth55, 2022-10-13:
SRDFが少しおかしいので修正．

knorth55, 2022-10-13:
特異点周りで関節角度が大きく変更していて，MoveIt的には2つの姿勢を遷移したことになっているが，Euslispの補間で当たっている気がする．
Moveitをもっと細かく解く必要がある？
以下の例では，肩関節が大きく突然変化している．
これはMoveIt的には次の瞬間，肩関節が大きく遷移していることになり，補間は実行（Euslisp）に任されている．
その際にこの軌道だとベースにあたる．

コレが原因？

https://user-images.githubusercontent.com/9300063/195586651-3762d55b-2b08-4637-9c3f-982db71e8491.mp4

knorth55, 2022-10-13:
にたようなはなし
https://github.com/ros-planning/moveit/issues/3213

computeCartesianにはjump_thresholdというのがあるが，これがあったらやばい挙動は止められる？

knorth55, 2022-10-13:
これも関係しそう？
Orientation ConstraintsをあたえるとOMPLはCartesian Pathで計算するため，joint jumpがおきやすい？と書いてある．
https://github.com/ros-planning/moveit/issues/2250

knorth55, 2022-10-13:
https://github.com/ros-planning/moveit/pull/2273

これはすごくそれっぽい

knorth55, 2022-10-13:
https://github.com/ros-planning/moveit/issues/562

knorth55, 2022-10-13:
## 検証の仕方
Depends: https://github.com/jsk-ros-pkg/jsk_robot/pull/1666

```bash
roslaunch jsk_fetch_startup fetch_gazebo_bringup.launch
```

```lisp
#!/usr/bin/env roseus

(require "package://fetcheus/fetch-interface.l")
(ros::roseus "hoge")

(setq *pre-grasp-pose* (make-coords :pos #f(-64.85 360.9 823.818) :rpy #f(-1.926 0.181 0.85)))
(setq *grasp-pose* (make-coords :pos #f(-50.682 263.943 778.426) :rpy #f(-1.632 0.097 0.772)))
(setq *pro-grasp-pose* (make-coords :pos #f(-46.346 261.287 950) :rpy #f(-1.571 0 0)))
(setq *ready-pose* (make-coords :pos #f(363.334 -139.026 508.384) :rpy #f(0 0 0)))

(fetch-init)

(defun pre-grasp ()
  (send *fetch* :angle-vector (send *ri* :state :potentio-vector))
  (send *fetch* :inverse-kinematics *pre-grasp-pose* :use-torso nil)
  (send *ri* :angle-vector (send *fetch* :angle-vector) 5000)
  (send *ri* :wait-interpolation)
  )

(defun grasp ()
  (send *fetch* :inverse-kinematics *grasp-pose* :use-torso nil)
  (send *ri* :angle-vector (send *fetch* :angle-vector) 1000)
  (send *fetch* :inverse-kinematics *pro-grasp-pose* :use-torso nil)
  (send *ri* :angle-vector (send *fetch* :angle-vector) 2000)
  (send *ri* :wait-interpolation)
  )

(defun ready ()
  (setq path-constraints
        (instance moveit_msgs::constraints :init
                  :name ""
                  :orientation_constraints
                  (list (instance moveit_msgs::OrientationConstraint :init
                                  :header
                                  (instance std_msgs::Header :init :frame_id "base_link")
                                  :link_name "gripper_link"
                                  :orientation
                                  (instance geometry_msgs::Quaternion :init
                                            :x 0 :y 1.0 :z 0 :w 0)
                                  :absolute_x_axis_tolerance 3.14
                                  :absolute_y_axis_tolerance 3.14
                                  :absolute_z_axis_tolerance 3.14 
                                  ))
                  ))
  (send *fetch* :inverse-kinematics *ready-pose* :use-torso nil)
  (send *ri* :angle-vector (send *fetch* :angle-vector) 5000 :default-controller 0
        :use-torso t
        :path-constraints path-constraints
        )
  (send *ri* :wait-interpolation)
  )

(defun main ()
  (pre-grasp)
  (grasp)
  (ready)
  )
```

## Good case

https://user-images.githubusercontent.com/9300063/195620089-c3443451-4408-4593-8e54-cae00bc75e2e.mp4

## Bad case

https://user-images.githubusercontent.com/9300063/195620117-91db585f-c6ba-41d0-b555-481615c1a83c.mp4

knorth55, 2022-10-13:
```
diff --git a/fetch_moveit_config/config/ompl_planning.yaml b/fetch_moveit_config/config/ompl_planning
index 6910e9f..9af2709 100644
--- a/fetch_moveit_config/config/ompl_planning.yaml
+++ b/fetch_moveit_config/config/ompl_planning.yaml
@@ -36,6 +36,7 @@ arm:
     - PRMstarkConfigDefault
   projection_evaluator: joints(shoulder_pan_joint,shoulder_lift_joint)
   longest_valid_segment_fraction: 0.005
+  enforce_joint_model_state_space: true
 arm_with_torso:
   planner_configs:
     - SBLkConfigDefault
@@ -51,6 +52,7 @@ arm_with_torso:
     - PRMstarkConfigDefault
   projection_evaluator: joints(torso_lift_joint,shoulder_pan_joint)
   longest_valid_segment_fraction: 0.05
+  enforce_joint_model_state_space: true
 gripper:
   planner_configs:
     - SBLkConfigDefault
```
@sktometometo 
can you try `enforce_joint_model_state_space` in `ompl_planning.yaml ` `fetch_moveit_config`?
you can use `develop/fetch` branch of `knorth55/fetch_ros`.

knorth55, 2022-10-19:
`enforce_joint_model_state_space`で自己干渉しなくなった．
これと`longest_valid_segment_fraction`の値を小さくすることで，見ている限りは拘束をそれなりに守りながら動いているようである．

このIssueの解決のために必要なPRは以下の4つである．
https://github.com/jsk-ros-pkg/jsk_robot/pull/1664
https://github.com/ZebraDevs/fetch_ros/pull/162
https://github.com/ZebraDevs/fetch_ros/pull/163
https://github.com/ZebraDevs/fetch_ros/pull/164