jsk-ros-pkg/jsk_robot #1671: [pr1040] EtherCATのNICでrx errorが多くサーボが入らない

Kanazawanaoaki, 2022-10-15:
pr1040でサーボが入らなくなる(入ってもガクガクする)問題が発生し，原因を調べた所，https://github.com/jsk-ros-pkg/jsk_robot/issues/794 と同じようにrx errorが多くなっていました．  

## 原因特定の経緯
pr1040が暫くaudible warningでrealtime loopについて喋っていたのでサーボを入れてみたが入らなかった．（realtime loopのwarningは一度サーボを入れると無くなるという情報有り？要出典調査）

```
sudo tail /var/log/upstart/jsk-pr2-startup.log  -n 1000000 | grep warn
```
[のログ](https://gist.github.com/Kanazawanaoaki/8513fef5d783f1a11d99c4efa4d98adb#file-jsk-pr2-startup_grep-warn-log )をみると2022/10/15 14:36:54 あたりから
```
[INFO] [1665812214.234336] [/audible_warning:rosout]: audible warning error name "/Realtime Controllers/Realtime Control Loop"
[INFO] [1665812214.234614] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 701.6667 hz
```
のようなwarningを喋っていたよう．  

rqt_pr2_dashboardを見ると，モーターではパケットドロップしていないがMasterではドロップしている．
[image]
(時々 “Could not collect WG0X diagnostics” のようなメッセージも出ている)  

[robotのログ](https://gist.github.com/Kanazawanaoaki/8513fef5d783f1a11d99c4efa4d98adb#file-robot-20221015-log )を見ると
```
readMailboxInternal : [43mERROR[0m asking for read repeat after dropping 1 packets
```
が出ている．これを検索すると以下のissueがhit．
https://github.com/jsk-ros-pkg/jsk_robot/issues/794
issueを見るとこれ自体はサーボが入らなくなる様なエラーでは無い気がするが，issueで出てきている`ethtool -S ecat0`を試してみるとrx_errorsが多かった．  
ログが流れてしまったので，後ほど20:01時点で実行した結果， 
```
root@pr1040:~# ethtool -S ecat0
NIC statistics:
     rx_packets: 197312038
     tx_packets: 197434613
     rx_bytes: 168553767856
     tx_bytes: 168697398371
     rx_broadcast: 197312038
     tx_broadcast: 197434595
     rx_multicast: 0
     tx_multicast: 18
     rx_errors: 230816
     tx_errors: 0
     tx_dropped: 0
     multicast: 0
     collisions: 0
     rx_length_errors: 0
     rx_over_errors: 0
     rx_crc_errors: 115423
     rx_frame_errors: 115393
     rx_no_buffer_count: 0
     rx_missed_errors: 0
     tx_aborted_errors: 0
     tx_carrier_errors: 0
     tx_fifo_errors: 0
     tx_heartbeat_errors: 0
     tx_window_errors: 0
     tx_abort_late_coll: 0
     tx_deferred_ok: 0
     tx_single_coll_ok: 0
     tx_multi_coll_ok: 0
     tx_timeout_count: 0
     tx_restart_queue: 0
     rx_long_length_errors: 0
     rx_short_length_errors: 0
     rx_align_errors: 115393
     tx_tcp_seg_good: 0
     tx_tcp_seg_failed: 0
     rx_flow_control_xon: 0
     rx_flow_control_xoff: 0
     tx_flow_control_xon: 0
     tx_flow_control_xoff: 0
     rx_csum_offload_good: 0
     rx_csum_offload_errors: 0
     rx_header_split: 0
     alloc_rx_buff_failed: 0
     tx_smbus: 0
     rx_smbus: 0
     dropped_smbus: 0
     rx_dma_failed: 0
     tx_dma_failed: 0
     rx_hwtstamp_cleared: 0
     uncorr_ecc_errors: 0
     corr_ecc_errors: 0
     tx_hwtstamp_timeouts: 0
     tx_hwtstamp_skipped: 0
```
issueで報告されているpr1040の値は rx_errors: 4 なのでとても多くなっている．  

[rx_errorsをcountするscript](https://gist.github.com/kosuke55/4bbfca95a2f367b5b8ff8f9c6b3f8dfa#file-ethtool_rx_error_analyzer-py )で確認．[出力されたcsvファイル](https://drive.google.com/file/d/1Fo3FJkjNn8v11xCeaXkitNcDeBMhMao2/view?usp=sharing )を見ると，l_elbow_flex_motorのRX Error Port 1の値が大きく，上昇もしているので怪しい．
```
/Motors/EtherCAT Device (l_elbow_flex_motor) - RX Error Port 1
90738
90738
90739
90740
90742
90742
90743
90746
90747
```
rqt_pr2_dashboardでl_elbow_flex_motorのDropsを確認したが0だった．  

## 一時的な解決
upstartのrobotを立ち上げ直したが改善しなかった．左腕を少し動かしてサーボを何度か入れてみたが，サーボが上手くはいらずガクガクしている状態だったのでそのまま放置した．しばらく放置して確認作業をしている間にサーボが復活していた．  

[復活した後のrx_errorのcsv](https://drive.google.com/file/d/18R5daBvfFVASuVOVu35ov-HFBGiTcTk1/view?usp=sharing )を見ると，l_elbow_flex_motorのRX Error Port 1の値は固定値で上がっていない，やはりこれが今回の原因のよう．
```
/Motors/EtherCAT Device (l_elbow_flex_motor) - RX Error Port 1
42207
42207
42207
42207
42207
42207
```

## 今後の対策
- `l_elbow_flex_motor`周りのケーブル交換等が必要？

Kanazawanaoaki, 2022-10-16:
先程もpr1040でこの問題と同じようにサーボが入らない事態が発生しました．  少し調べてみて分かっていることを報告します．  
一時的な解決策としては，やはり(本当に効果があるかは未検証だが)pr2の腕を動かすなどして暫くするとサーボが入るようになるようです．

## 症状の整理
サーボが入らなくなる時の症状としては，
- `audible warning` で `realtime controllers realtime control loop realtime loop used too much time in the last 30 seconds.;` が出ている．
- `robot.log` に `[ERROR] [1664324590.399660822] [/realtime_loop:ros.ethercat_hardware]: Device 16 : previous port error counters less current values`のようなエラーが出ている
- EtherCATの 'rx error' が発生している

の3つの症状が発生しているようです．それぞれの関係性と原因についてはまだ良くわかっていません．

### `audible warning` で `realtime controllers realtime control loop realtime loop used too much time in the last 30 seconds.;` が出ている．
2022年の9月に入ってからpr1040でRealtime loopが遅くなり `audible warning` で以下のようなwarningが出ることが報告されるようになった．https://github.com/jsk-ros-pkg/jsk_robot/issues/1623
```
[INFO] [1664417978.378555] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop halting, realtime loop only ran at 740.0000 hz
```
このようなwarningの場合はサーボを入れるとwarningが出なくなる（しばらくすると再発することもある）ことが報告されています https://github.com/jsk-ros-pkg/jsk_robot/issues/1623#issuecomment-1267112863 ．  確かに[昨日のログ](https://gist.github.com/Kanazawanaoaki/8513fef5d783f1a11d99c4efa4d98adb#file-jsk-pr2-startup_grep-warn-log-L67-L76 )でも[今日のログ](https://gist.github.com/Kanazawanaoaki/206e1ef6452711b7d648b1e7f556bac5#file-jsk-pr2-startup-2022-1017-0119-grep-warn-log-L161-L171 )でもランストップを押してサーボを入れるとそれまで連続していたwarningが出なくなっている．   
これは，https://github.com/jsk-ros-pkg/jsk_robot/issues/1623#issuecomment-1267150609 で言われているように
- サーボOFF状態だと、このwarningが出てもHzが更新されず、同じHzのwarningが出続ける
- サーボON状態だと、Hzが更新されているようで、warningが出たり消えたりする

ということのようである．  
しかし，以下のように`realtime loop used too much time in the last 30 second`のwarningも出ている時にランストップを入れるとサーボが入らないようである．（[昨日の該当箇所のログ](https://gist.github.com/Kanazawanaoaki/8513fef5d783f1a11d99c4efa4d98adb#file-jsk-pr2-startup_grep-warn-log-L175-L186 )，[今日の該当箇所のログ](https://gist.github.com/Kanazawanaoaki/206e1ef6452711b7d648b1e7f556bac5#file-jsk-pr2-startup-2022-1017-0119-grep-warn-log-L176-L184 )）
```
[INFO] [1664309197.368809] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop realtime loop used too much time in the last 30 seconds.; halting, realtime loop only ran at 740.0000 hz
```
[これまでもhzのwarningとtoo muchのwarningが一緒に出ていたことはあるよう](https://github.com/jsk-ros-pkg/jsk_robot/issues/1623#issuecomment-1264591444 )だが，そのタイミングでサーボを入れていなかったからこの問題が発生しなかっただけ？  

#### 調査
この情報で検索してhitしたのは以下の2つ. 詳しくみる必要があるかも．
- https://groups.google.com/g/pr2_users/c/cqaRPAkeino
- https://github.com/tork-a/minas/blob/4ff346d334078f2958222024accb07499c6d66d9/minas_control/src/main.cpp#L203

### `robot.log` に `[ERROR] [1664324590.399660822] [/realtime_loop:ros.ethercat_hardware]: Device 16 : previous port error counters less current values`のようなエラーが出ている
よく調べていくと[昨日のログ](https://gist.github.com/Kanazawanaoaki/8513fef5d783f1a11d99c4efa4d98adb#file-robot-20221015-log )には
```
readMailboxInternal : [43mERROR[0m asking for read repeat after dropping 1 packets
```
だけではなく，以下のエラーログも出ていた．今日のログにもDevice の番号は違うが出ていたので，これがサーボがつかない直接的な原因かもしれない．
```
[31m[ERROR] [1665826174.272734993] [/realtime_loop:ros.ethercat_hardware]: Device 16 : previous port error counters less current values[0m
```

#### 調査
このエラーログは[PR2/pr2_ethercat_driversで](https://github.com/PR2/pr2_ethercat_drivers/blob/d93eb53a0a3a8e6706f7061e30e37a5d021eee1d/ethercat_hardware/src/ethercat_device.cpp#L245-L247 )出されているものだということは分かった．

### EtherCATの 'rx error' が発生している
https://github.com/jsk-ros-pkg/jsk_robot/issues/1671#issue-1410142269 でも報告した通り，rx errorが発生している時にサーボが入らなくなり，サーボが安定している時にはrx errorは増えていない．これが根本の原因の可能性もある．
```
$ rosrun jsk_pr2_startup diagnostics_rx_error_summarizer.py
/Motors/EtherCAT Device #00 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 1: 0
[... 61 lines cut ...]
[Max Device]
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 1: 208261
```
やはり Forwarded 以外の RX Error Port でみると `l_elbow_flex_motor` でエラーがおきているように思える．

#### 調査
`/Motors/EtherCAT Master RX Errors`が何を示しているか等は不明．RX Error Portの合計値という訳でも無い．しかし`ethtool -S ecat0`したときの`rx_crc_errors`と`rx_frame_errors`の値の合計が`rx_errors`になっている．  
```
 (rx_errors) = (rx_crc_errors) + (rx_frame_errors)
```
ちなみに関連していそうな話題として`EtherCAT Master`において[ros wiki に載っている](http://library.isr.ist.utl.pt/docs/roswiki/pr2_etherCAT(2f)Tutorials(2f)PacketLoss.html#Determining_which_problem_you_have )以下の関係式があるが RX Errorとの関連性は不明．値は微妙に一致していない．．．
```
(Dropped Packets) = (CRC Errors) + (RX Late Packet)
```
値の一覧
```
     rx_errors: 460752
     rx_crc_errors: 230412
     rx_frame_errors: 230340
```
```
Dropped Packets: 154664
RX Late Packet: 8
CRC Errors (Dropped Packets - RX Late Packet): 154656
```

Kanazawanaoaki, 2022-10-16:
### pr1012のrx errorを調べるとそこそこ発生している様子
```
/Motors/EtherCAT Device #10 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #10 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #10 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #10 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #11 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #35 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #36 (WG014) RX Error Port 3: 44
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 255
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 1: 8
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 83767
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_gripper_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_wrist_l_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_wrist_l_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_wrist_r_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_wrist_r_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (laser_tilt_mount_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (laser_tilt_mount_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (led_projector) RX Error Port 0: 0
/Motors/EtherCAT Device (led_projector) RX Error Port 1: 0
/Motors/EtherCAT Device (r_elbow_flex_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_elbow_flex_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_forearm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_forearm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_gripper_motor) RX Error Port 0: 0
[... 35 lines cut ...]

`realtime loop used too much time in the last 30 seconds.;`が発生していたlogは
```
jsk-pr2-startup.log.14
jsk-pr2-startup.log.13
jsk-pr2-startup.log.12
jsk-pr2-startup.log.9
jsk-pr2-startup.log.8
jsk-pr2-startup.log.7
jsk-pr2-startup.log.3
jsk-pr2-startup.log.2
jsk-pr2-startup.log.1
jsk-pr2-startup.log
```
`Device 27 : previous port error counters less current values`のエラーが発生していたlogは
```
robot.log.14
robot.log.13
robot.log.9
robot.log.8
robot.log.7
robot.log.1
robot.log
```
関連はしているが，必ず同時に起こる症状ではないのかもしれない．

Kanazawanaoaki, 2022-10-18:
こちらの問題ですが，`l_elbow_flex_motor`と`l_forearm_roll_motor`の間の通信経路に問題があるような気がしています．
```
/Motors/EtherCAT Device #00 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 1: 92621
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 362
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_gripper_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_wrist_l_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_wrist_l_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_wrist_r_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_wrist_r_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (laser_tilt_mount_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (laser_tilt_mount_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (led_projector) RX Error Port 0: 0
/Motors/EtherCAT Device (led_projector) RX Error Port 1: 0
/Motors/EtherCAT Device (r_elbow_flex_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_elbow_flex_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_forearm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_forearm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_gripper_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_shoulder_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_shoulder_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_upper_arm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_upper_arm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_wrist_l_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_wrist_l_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (r_wrist_r_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (torso_lift_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (torso_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Master RX Errors: 270586
[Max Device]
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 1: 92621
--------------------------------------------
```
実際に，今調べてみると，`l_elbow_flex_motor`と`l_forearm_roll_motor`でRX Errorが発生していることを確認しました．本格的に直すにはここの接続をバイパスしてチェックしてみるとかが必要になるかもしれません．
cc. @knorth55

Kanazawanaoaki, 2022-10-21:
先程も同じようなエラーが発生しました．  
```
[INFO] [1666323521.378856] [/audible_warning:rosout]: audible warning talking: error. realtime controllers realtime control loop realtime loop used too much time in the last 30 seconds.; halting, realtime loop only ran at 735.0000 hz
```
というaudible_warningを喋っていたのでRun Stopをonにした所やはりサーボが上手くつかずガクガクする状態になりました．  

`robot.log`を確認すると，
```
readMailboxInternal : ERROR asking for read repeat after dropping 1 packets
```
のログは出ていましたが，https://github.com/jsk-ros-pkg/jsk_robot/issues/1671#issuecomment-1280037795 で報告していた` [/realtime_loop:ros.ethercat_hardware]: Device 16 : previous port error counters less current values`のエラーは出ていませんでした．  

rx_errorを確認すると今度は，`l_forearm_roll_motor`でのみ発生していて，サーボが落ちている(or ガクガクしている)時には値が増えていきました．
```
/Motors/EtherCAT Device #00 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #00 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #01 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #15 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 0: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 1: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 2: 0
/Motors/EtherCAT Device #16 (WG014) RX Error Port 3: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (bl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (br_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fl_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_l_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_r_wheel_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (fr_caster_rotation_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (head_tilt_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 9120
/Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_gripper_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_lift_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_shoulder_pan_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 0: 0
/Motors/EtherCAT Device (l_upper_arm_roll_motor) RX Error Port 1: 0
/Motors/EtherCAT Device (l_wrist_l_motor) RX Error Port 0: 0
[... 46 lines cut ...]
EtherCAT Device (fr_caster_rotation_motor) Drops: 0
EtherCAT Device (torso_lift_motor) Drops: 0
EtherCAT Device (l_shoulder_pan_motor) Drops: 0
EtherCAT Device (l_shoulder_lift_motor) Drops: 0
EtherCAT Device (l_upper_arm_roll_motor) Drops: 0
EtherCAT Device (l_elbow_flex_motor) Drops: 0
EtherCAT Device (l_forearm_roll_motor) Drops: 0
EtherCAT Device (l_wrist_l_motor) Drops: 0
EtherCAT Device (l_wrist_r_motor) Drops: 0
EtherCAT Device (l_gripper_motor) Drops: 0
EtherCAT Device (head_pan_motor) Drops: 0
EtherCAT Device (r_shoulder_pan_motor) Drops: 0
EtherCAT Device (r_shoulder_lift_motor) Drops: 0
EtherCAT Device (r_upper_arm_roll_motor) Drops: 0
EtherCAT Device (r_elbow_flex_motor) Drops: 0
EtherCAT Device (r_forearm_roll_motor) Drops: 0
EtherCAT Device (r_wrist_l_motor) Drops: 0
EtherCAT Device (r_wrist_r_motor) Drops: 0
EtherCAT Device (r_gripper_motor) Drops: 0
EtherCAT Device (head_tilt_motor) Drops: 0
EtherCAT Device (laser_tilt_mount_motor) Drops: 0
--------------------------------------------
```

`l_elbow_flex_motor`と`l_forearm_roll_motor`の間の通信経路やスリップリングの接触不良がある可能性がさらに高くなりました．暫くするとrx_errorも発生せず，サーボが安定して入るようになりました．

Kanazawanaoaki, 2023-10-06:
同じような問題が現在も発生していて，
https://github.com/jsk-ros-pkg/jsk_robot/issues/794
と同じように，
```
clearReadMailbox : WARN read mbx contained garbage data
```
が出ていてサーボが入らないことがある．

```
--------------------------------------------
EtherCAT Master Dropped Packets: 4893961
EtherCAT Master RX Late Packet: 0
EtherCAT Device (br_caster_r_wheel_motor)Drops: 0
EtherCAT Device (br_caster_l_wheel_motor)Drops: 0
EtherCAT Device (br_caster_rotation_motor)Drops: 0
EtherCAT Device (fl_caster_r_wheel_motor)Drops: 0
EtherCAT Device (fl_caster_l_wheel_motor)Drops: 0
EtherCAT Device (fl_caster_rotation_motor)Drops: 0
EtherCAT Device (bl_caster_r_wheel_motor)Drops: 0
EtherCAT Device (bl_caster_l_wheel_motor)Drops: 0
EtherCAT Device (bl_caster_rotation_motor)Drops: 0
EtherCAT Device (fr_caster_r_wheel_motor)Drops: 0
EtherCAT Device (fr_caster_l_wheel_motor)Drops: 0
EtherCAT Device (fr_caster_rotation_motor)Drops: 0
EtherCAT Device (torso_lift_motor)Drops: 0
EtherCAT Device (l_shoulder_pan_motor)Drops: 0
EtherCAT Device (l_shoulder_lift_motor)Drops: 0
EtherCAT Device (l_upper_arm_roll_motor)Drops: 0
EtherCAT Device (l_elbow_flex_motor)Drops: 0
EtherCAT Device (l_forearm_roll_motor)Drops: 10153
EtherCAT Device (l_wrist_l_motor)Drops: 10153
EtherCAT Device (l_wrist_r_motor)Drops: 10153
EtherCAT Device (l_gripper_motor)Drops: 10153
EtherCAT Device (head_pan_motor)Drops: 0
EtherCAT Device (r_shoulder_pan_motor)Drops: 0
EtherCAT Device (r_shoulder_lift_motor)Drops: 0
EtherCAT Device (r_upper_arm_roll_motor)Drops: 0
EtherCAT Device (r_elbow_flex_motor)Drops: 0
EtherCAT Device (r_forearm_roll_motor)Drops: 0
EtherCAT Device (r_wrist_l_motor)Drops: 0
EtherCAT Device (r_wrist_r_motor)Drops: 0
EtherCAT Device (r_gripper_motor)Drops: 0
EtherCAT Device (head_tilt_motor)Drops: 0
EtherCAT Device (laser_tilt_mount_motor)Drops: 0
--------------------------------------------
```
```
[Devices with errors]
1696574170 : /Motors/EtherCAT Device (fl_caster_r_wheel_motor) RX Error Port 1: 2
1696574170 : /Motors/EtherCAT Device (l_elbow_flex_motor) RX Error Port 1: 167859
1696574170 : /Motors/EtherCAT Device (l_forearm_roll_motor) RX Error Port 0: 704
```
となっていてやはり左腕の`l_elbow_flex_motor`と`l_forearm_roll_motor`の辺りが怪しそう．

k-okada, 2023-10-06:
> l_elbow_flex_motorとl_forearm_roll_motorの間の通信経路やスリップリングの接触不良がある可能性がさらに高くなりました．

1年前にはケーブルとかスリップリングは交換したり、接点復活剤を掛けたり、スリップリングを飛ばして直接ボード同士をケーブルでつないだりしたのかな？Noだったらまずはそれを試して、Yesだったら新しく着たアームの基板と交換してみるのかな。

Kanazawanaoaki, 2023-10-06:
一年前にも同じような状態で緊急ではないということで何もしていないのでNoで，
スリップリングの換えは機械加工すれば使えるものがあるので交換するか，スリップリングを飛ばして直接ボード同士をケーブルを時期をみてやって様子をみることになるかと思います．

k-okada, 2023-10-06:
その2つだと

>  スリップリングを飛ばして直接ボード同士をケーブルを

が先にやって、動くならスリップリングが問題なので加工するのが正解じゃないでしょうか？
eus経由で動かしていれば、意図的にプログラム書かなければ無限回転しないはずなので、安全見て長めのケーブルでつないでみましょう。
--
◉ Kei Okada

mqcmd196, 2025-08-13:
2025/8/13 現在 同様の問題が発生

Kanazawanaoaki, 2025-08-14:
このl_elbow_flex_motorのrxエラーは時々起こっています．
https://gitlab.jsk.imi.i.u-tokyo.ac.jp/kanazawa/kanazawa_pr2_utils/-/blob/pr2-servo-debug/node_scripts/diagnostics_rx_error_summarizer.py?ref_type=heads
で記録すると例えば下のグラフのようになっているのですが，このrxエラーが起こっている時にサーボが入っているとサーボ落ちが発生してカクカクします．だいたいは暫くすると直るのですが，条件を変えて問題の特定するのをやってみるべきなのかもです．

[image]