jsk-ros-pkg/jsk_robot #1689: [jsk_pr2_startup] check_openni_node.pyのrestart_openni_nodeが機能していない？

Kanazawanaoaki, 2022-10-21:
pr1040が  
```
[INFO] [1666321846.687070] [/audible_warning:rosout]: audible warning error name "/Other/sanity_diagnostics:  kinect_head rgb image_raw compressed"
[INFO] [1666321846.687400] [/audible_warning:rosout]: audible warning talking: error. other sanity diagnostics colon kinect head rgb image raw compressed topic is not published
```
のaudible_warningを喋っていました．[jsk-pr2-startupのログ](https://gist.github.com/Kanazawanaoaki/61c934ed493edc2b8454a58d189baea8#file-20221021-1132-jsk-pr2-startup-log )を確認した所，
```
[INFO] [1666318789.372664] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666318789
[ERROR] [1666318789.372947] [/check_openni_node:rosout]: Restart openni_node1
[ERROR] [1666318789.373095] [/check_openni_node:rosout]: [CheckOpenNINode] resetting u s b
****
```
と`check_openni_node.py`のresetがされているのですが，うまくresetされていないみたいです．
対応する部分のログ：https://gist.github.com/Kanazawanaoaki/61c934ed493edc2b8454a58d189baea8#file-jsk-pr2-startup-kinect-error-log  
check_openni_node.pyのrestart_openni_node：https://github.com/jsk-ros-pkg/jsk_robot/blob/6a352a709b5964a419f0a4a33059be49e5c00a10/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors/check_openni_node.py#L57-L88

Kanazawanaoaki, 2022-10-21:
これと同じようなエラーがまた発生しました．  
```
[INFO] [1666336546.175142] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666336546
[ERROR] [1666336546.175773] [/check_openni_node:rosout]: Restart openni_node1
[ERROR] [1666336546.176217] [/check_openni_node:rosout]: [CheckOpenNINode] resetting u s b
```
となっていて，その後

```
[kinect_head/driver-42] restarting process
process[kinect_head/driver-42]: started with pid [32199]
[ INFO] [1666340210.549467781] [/kinect_head/driver:ros.nodelet]: Loading nodelet /kinect_head/driver of type openni_camera/driver to manager kinect_head_nodelet_manager with the following remappings:
[ INFO] [1666340210.549514979] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth -> /kinect_head/depth
[ INFO] [1666340210.549524898] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth_registered -> /kinect_head/depth_registered
[ INFO] [1666340210.549534075] [/kinect_head/driver:ros.nodelet]: /kinect_head/ir -> /kinect_head/ir
[ INFO] [1666340210.549540153] [/kinect_head/driver:ros.nodelet]: /kinect_head/projector -> /kinect_head/projector
[ INFO] [1666340210.549546833] [/kinect_head/driver:ros.nodelet]: /kinect_head/rgb -> /kinect_head/rgb
[ERROR] [1666340210.553549331] [/kinect_head/kinect_head_nodelet_manager:ros.nodelet]: Cannot load nodelet /kinect_head/driver for one exists with that name already
[FATAL] [1666340210.553621745] [/kinect_head/driver:ros.nodelet]: Failed to load nodelet '/kinect_head/driver` of type `openni_camera/driver` to manager `kinect_head_nodelet_manager'
remote[c2-0]: nt or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.411938999]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.444675371]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.476431241]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.511501267]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.543802222]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.576196638]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.613061381]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.644819140]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.676628867]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.713344954]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1666340210.744779092]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth im
[kinect_head/driver-42] process has died [pid 32199, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load openni_camera/driver kinect_head_nodelet_manager ir:=ir rgb:=rgb depth:=depth depth_registered:=depth_registered projector:=projector __name:=driver __log:=/home/applications/.ros/log/0ed64830-50ef-11ed-9b02-d05099c29feb/kinect_head-driver-42.log].
log file: /home/applications/.ros/log/0ed64830-50ef-11ed-9b02-d05099c29feb/kinect_head-driver-42*.log
```
や
```
[kinect_head/driver-42] restarting process
process[kinect_head/driver-42]: started with pid [17201]
[ INFO] [1666340359.451196696] [/kinect_head/driver:ros.nodelet]: Loading nodelet /kinect_head/driver of type openni_camera/driver to manager kinect_head_nodelet_manager with the following remappings:
[ INFO] [1666340359.451249785] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth -> /kinect_head/depth
[ INFO] [1666340359.451261896] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth_registered -> /kinect_head/depth_registered
[ INFO] [1666340359.451274888] [/kinect_head/driver:ros.nodelet]: /kinect_head/ir -> /kinect_head/ir
[ INFO] [1666340359.451289436] [/kinect_head/driver:ros.nodelet]: /kinect_head/projector -> /kinect_head/projector
[ INFO] [1666340359.451299497] [/kinect_head/driver:ros.nodelet]: /kinect_head/rgb -> /kinect_head/rgb
[ERROR] [1666340359.455716203] [/kinect_head/kinect_head_nodelet_manager:ros.nodelet]: Cannot load nodelet /kinect_head/driver for one exists with that name already
[FATAL] [1666340359.455800994] [/kinect_head/driver:ros.nodelet]: Failed to load nodelet '/kinect_head/driver` of type `openni_camera/driver` to manager `kinect_head_nodelet_manager'
[kinect_head/driver-42] process has died [pid 17201, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load openni_camera/driver kinect_head_nodelet_manager ir:=ir rgb:=rgb depth:=depth depth_registered:=depth_registered projector:=projector __name:=driver __log:=/home/applications/.ros/log/0ed64830-50ef-11ed-9b02-d05099c29feb/kinect_head-driver-42.log].
log file: /home/applications/.ros/log/0ed64830-50ef-11ed-9b02-d05099c29feb/kinect_head-driver-42*.log
```
のエラーが続いていました．エラーがたちあがり続けているのはcheck_openni_node.pyのrestart_openni_nodeでlaunchを立ち上げ直しているのと，[launchのrespawn](https://github.com/jsk-ros-pkg/jsk_robot/blob/e4ad004fbf175d2a9c2539d080a3aa3edcb33b5e/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors/kinect_head.launch#L3 )が被っているのが問題なのでは無いかと考えて https://github.com/jsk-ros-pkg/jsk_robot/pull/1692 を試してみています．

knorth55, 2022-10-21:
#1692 で修正を試みています

knorth55, 2022-10-21:
KinectのRGB画像が真っ黒になる問題が起きています
今日試したところ、冷蔵庫デモの冷蔵庫の中を見る同じ箇所で2回起きました
Kinectの接触不良ではないかと思っています

Kanazawanaoaki, 2022-10-21:
> 以前起きていたUSBデバイスとしてアクセスできない話

は https://github.com/jsk-ros-pkg/jsk_robot/issues/1511 ですね，こちらで報告されていた`Reason: Failed to send a USB control request!` のエラーは発生していませんでした．

knorth55, 2022-10-22:
昨日の19:40を最後にKinectの画像が真っ黒になるという現象は起きていません．
Kinectの接触不良なような気がします．PR1040を動かすと発生するかもしれません．

```bash
applications@pr1040:/var/log/upstart $ sudo zgrep -i "CheckOpenNINode" jsk-pr2-startup.log.1.gz | grep "sum of pixels is 0\|restart\|kill"                                                                  
[INFO] [1666313948.461089] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666313948
[ERROR] [1666313960.118114] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666313970.412286] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666313980.445417] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666314248.736698] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666314248
[ERROR] [1666314260.392322] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666314270.681820] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666314280.717756] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666314428.932793] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666314428
[ERROR] [1666314440.614258] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666314450.918062] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666314460.941700] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666314729.264473] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666314729
[ERROR] [1666314740.964195] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666314751.266702] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666314761.294704] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666318789.372664] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666318789
[ERROR] [1666318802.462699] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666318812.971065] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666318823.019106] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666322467.890779] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666322467
[ERROR] [1666322479.552331] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666322489.856121] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666322499.885935] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666324602.324992] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666324602
[ERROR] [1666324614.046068] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666324624.327796] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666324634.351896] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666336546.175142] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666336546
[ERROR] [1666336557.963648] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666336568.291845] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[ERROR] [1666336578.318078] [/check_openni_node:rosout]: [CheckOpenNINode] restarting processes
[INFO] [1666348835.524842] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666348835
[ERROR] [1666348847.344917] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666348857.677272] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
applications@pr1040:/var/log/upstart $ sudo zgrep -i "CheckOpenNINode" jsk-pr2-startup.log | grep "sum of pixels is 0\|restart\|kill"
applications@pr1040:/var/log/upstart $ 
```

knorth55, 2022-10-22:
KinectのRGB画像が真っ黒になる様子が動画に取れていたので，こちらに添付しておきます．
Full Videoの00:01:47あたりから，様子がおかしくなり00:01:59にはKinectのリセットがかかりました．
Clipped Videoは該当箇所だけClipしています．

### Clipped Video
https://user-images.githubusercontent.com/9300063/197340015-a6350cd3-5996-4949-aab8-83d21f9196e7.mp4

### Full video
https://user-images.githubusercontent.com/9300063/197340024-deff9193-d20e-4d29-83ad-7ca9e6de5c2f.mp4

Kanazawanaoaki, 2022-10-23:
先程確認した所，昨晩の22:40:11頃におそらくこのKinectの画像が真っ黒になるという現象が起き，その後も何度か起きているようです．またOpenNIは上手くrestart出来ていないようです． 
[jsk-pr2-startup.log.1](https://gist.github.com/Kanazawanaoaki/60732c6c9a89e5daf5bc2bb59aa0b2ed#file-jsk-pr2-startup-log-1 )  
[jsk-pr2-startup.log] (https://gist.github.com/Kanazawanaoaki/60732c6c9a89e5daf5bc2bb59aa0b2ed#file-jsk-pr2-startup-log )
```
applications@pr1040:/var/log/upstart $ sudo zgrep -i "CheckOpenNINode" jsk-pr2-startup.log.1.gz | grep "sum of pixels is 0\|restart\|kill"   
[INFO] [1666446011.254250] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666446011
[ERROR] [1666446022.878813] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666446033.254981] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[INFO] [1666446792.121355] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666446792
[ERROR] [1666446803.806606] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666446814.097019] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[INFO] [1666447092.439318] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666447092
[ERROR] [1666447104.085393] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666447114.369607] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[INFO] [1666447212.548989] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666447212
[ERROR] [1666447224.152517] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[ERROR] [1666447234.466805] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
```

knorth55, 2022-10-23:
以下の体内のBranchってどうなってますか？
もしかしたら`develop/pr2`にもどっているかも

Kanazawanaoaki, 2022-10-23:
そうですね，`develop/pr2`になっていました．とりあえず`develop/pr2-fix-check-openni-node`に変更しておきました．
```bash
applications@pr1040:~/ros/indigo/src/jsk-ros-pkg/jsk_robot/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors [develop/pr2]$ git branch 
  automate-logger
  deferred-load
  devel
  develop-bak
  develop-merge
* develop/pr2
  develop/pr2-fix-check-openni-node
  fetch-app-plugin
  fix-pr2-twitter
  fix-self-filter
  mannequin-test
  master
  nakane/develop
  persistent
  pr1040-audible-warning
  pr1040-test
  pr1040-with-audible-warning
  pr2-indigo
  pr2-indigo-bak
  rfcomm-bind-
  sktometometo/develop
  smach-to-mail-finish-success-failure
  throttle-action
  tweet-server
```

knorth55, 2022-10-23:
ありがとう！これで様子を見てみてください

knorth55, 2022-10-24:
```
[INFO] [1666619794.667012] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666619794
[31m[ERROR] [1666619794.667315] [/check_openni_node:rosout]: Restart openni_node1
[0m[31m[ERROR] [1666619794.667511] [/check_openni_node:rosout]: [CheckOpenNINode] resetting u s b
[0mResetting USB device /dev/bus/usb/001/044
Reset successful
Resetting USB device /dev/bus/usb/001/043
Reset successful
Resetting USB device /dev/bus/usb/001/006
Reset successful
[31m[ERROR] [1666619806.421436] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[0mkilling /kinect_head/kinect_head_nodelet_manager
[33m[ WARN] [1666619806.658834401] [/kinect_head/kinect_head_nodelet_manager:ros.roscpp]: Shutdown request received.[0m
[33m[ WARN] [1666619806.658868957] [/kinect_head/kinect_head_nodelet_manager:ros.roscpp]: Reason given for shutdown: [user request][0m
killed
[1m[kinect_head/kinect_head_nodelet_manager-41] process has finished cleanly
log file: /home/applications/.ros/log/96fc0f76-52af-11ed-b941-d05099c29feb/kinect_head-kinect_head_nodelet_manager-41*.log[0m
[kinect_head/kinect_head_nodelet_manager-41] restarting process
[1mprocess[kinect_head/kinect_head_nodelet_manager-41]: started with pid [11605][0m
[0m[ INFO] [1666619807.351663804] [/kinect_head/kinect_head_nodelet_manager:ros.nodelet]: Initializing nodelet with 8 worker threads.[0m
 Checking /kinect_head/rgb/image_raw/compressed for 10 seconds
[0m[ INFO] [1666619809.834236354] [/kinect_head/driver:ros.nodelet]: Bond broken, exiting[0m
[1m[kinect_head/driver-42] process has finished cleanly
log file: /home/applications/.ros/log/96fc0f76-52af-11ed-b941-d05099c29feb/kinect_head-driver-42*.log[0m
[kinect_head/driver-42] restarting process
[1mprocess[kinect_head/driver-42]: started with pid [13198][0m
[0m[ INFO] [1666619810.135899690] [/kinect_head/driver:ros.nodelet]: Loading nodelet /kinect_head/driver of type openni_camera/driver to manager kinect_head_nodelet_manager with the following remappings:[0m
[0m[ INFO] [1666619810.135954953] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth -> /kinect_head/depth[0m
[0m[ INFO] [1666619810.135970223] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth_registered -> /kinect_head/depth_registered[0m
[0m[ INFO] [1666619810.135986477] [/kinect_head/driver:ros.nodelet]: /kinect_head/ir -> /kinect_head/ir[0m
[0m[ INFO] [1666619810.135998031] [/kinect_head/driver:ros.nodelet]: /kinect_head/projector -> /kinect_head/projector[0m
[0m[ INFO] [1666619810.136008445] [/kinect_head/driver:ros.nodelet]: /kinect_head/rgb -> /kinect_head/rgb[0m
[0m[ INFO] [1666619810.284604874] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: Number devices connected: 1[0m
[0m[ INFO] [1666619810.284652174] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: 1. device on bus 001:46 is a SensorKinect (2ae) from PrimeSense (45e) with serial id 'A00362913844045A'[0m
[0m[ INFO] [1666619810.285214970] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: Searching for device with index = 1[0m
[0m[ INFO] [1666619810.466807868] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: Opened 'SensorKinect' on bus 1:46 with serial number 'A00362913844045A'[0m
[0m[ INFO] [1666619810.576151367] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: rgb_frame_id = 'head_mount_kinect_rgb_optical_frame' [0m
[0m[ INFO] [1666619810.576198310] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: depth_frame_id = 'head_mount_kinect_ir_optical_frame' [0m
[33m[ WARN] [1666619810.579793947] [/kinect_head/kinect_head_nodelet_manager:ros.camera_info_manager]: Camera calibration file /home/applications/.ros/camera_info/rgb_A00362913844045A.yaml not found.[0m
[33m[ WARN] [1666619810.579822398] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: Using default parameters for RGB camera calibration.[0m
[33m[ WARN] [1666619810.579838447] [/kinect_head/kinect_head_nodelet_manager:ros.camera_info_manager]: Camera calibration file /home/applications/.ros/camera_info/depth_A00362913844045A.yaml not found.[0m
[33m[ WARN] [1666619810.579853211] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: Using default parameters for IR camera calibration.[0m
[31m[ERROR] [1666619811.179747172] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: Cannot stream RGB and IR at the same time. Streaming RGB only.[0m
[31m[ERROR] [1666619816.779677] [/check_openni_node:rosout]: [CheckOpenNINode] killing child processes
[0m[31m[kinect_head/kinect_head_nodelet_manager-41] process has died [pid 11605, exit code -15, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect_head_nodelet_manager __log:=/home/applications/.ros/log/96fc0f76-52af-11ed-b941-d05099c29feb/kinect_head-kinect_head_nodelet_manager-41.log].
log file: /home/applications/.ros/log/96fc0f76-52af-11ed-b941-d05099c29feb/kinect_head-kinect_head_nodelet_manager-41*.log[0m
[31m[kinect_head/driver-42] process has died [pid 13198, exit code -15, cmd /opt/ros/indigo/lib/nodelet/nodelet load openni_camera/driver kinect_head_nodelet_manager ir:=ir rgb:=rgb depth:=depth depth_registered:=depth_registered projector:=projector __name:=driver __log:= [... cut ...]
[INFO] [1666619854.727960] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 23767706 at 1666619854
 Checking /kinect_head/rgb/image_raw/compressed for 10 seconds
 Checking /kinect_head/depth_registered/image_raw/compressedDepth for 10 seconds
 Checking /edgetpu_human_pose_estimator/output/image/compressed for 10 seconds
 Checking /edgetpu_object_detector/output/image/compressed for 10 seconds
 Checking /audio for 10 seconds
[INFO] [1666619914.769916] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 23731770 at 1666619914
```

knorth55, 2022-10-24:
さきほどまたブラックアウトが起きました．
問題なく`Kinect`にリセットがかかり，再起動に成功しました．

TODO:
- そもそも`/kinect_head/rgb/image_raw`や`/kinect_head/depth/image_raw`がでないときにリセットをかける．

```
[INFO] [1666624064.276813] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 0 at 1666624064
[31m[ERROR] [1666624064.277129] [/check_openni_node:rosout]: Restart openni_node1
[0m[31m[ERROR] [1666624064.277351] [/check_openni_node:rosout]: [CheckOpenNINode] resetting u s b
[0m[INFO] [1666624064.541941] [/audible_warning:rosout]: audible warning is enabled. speak [WARN, ERROR, STALE] levels
[INFO] [1666624064.542258] [/audible_warning:rosout]: audible warning's speak interval was set to 600
[INFO] [1666624064.606194] [/audible_warning:rosout]: Audible Warning: Runstop is enabled.
Resetting USB device /dev/bus/usb/001/046
Reset successful
 Checking /kinect_head/rgb/image_raw/compressed for 10 seconds
Resetting USB device /dev/bus/usb/001/045
Reset successful
Resetting USB device /dev/bus/usb/001/006
Reset successful
[31m[ERROR]  /kinect_head/rgb/image_raw/compressed is not published[39m
 Checking /kinect_head/depth_registered/image_raw/compressedDepth for 10 seconds
[31m[ERROR] [1666624075.907349] [/check_openni_node:rosout]: [CheckOpenNINode] something wrong with kinect, I'll restart it, killing nodelet manager
[0mkilling /kinect_head/kinect_head_nodelet_manager
[33m[ WARN] [1666624076.090348025] [/kinect_head/kinect_head_nodelet_manager:ros.roscpp]: Shutdown request received.[0m
[33m[ WARN] [1666624076.090383992] [/kinect_head/kinect_head_nodelet_manager:ros.roscpp]: Reason given for shutdown: [user request][0m
killed
[31m[kinect_head/kinect_head_nodelet_manager-41] process has died [pid 9325, exit code -15, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect_head_nodelet_manager __log:=/home/applications/.ros/log/67942e5e-53ad-11ed-8007-d05099c29feb/kinect_head-kinect_head_nodelet_manager-41.log].
log file: /home/applications/.ros/log/67942e5e-53ad-11ed-8007-d05099c29feb/kinect_head-kinect_head_nodelet_manager-41*.log[0m
[31m[kinect_head/driver-42] process has died [pid 9343, exit code -15, cmd /opt/ros/indigo/lib/nodelet/nodelet load openni_camera/driver kinect_head_nodelet_manager ir:=ir rgb:=rgb depth:=depth depth_registered:=depth_registered projector:=projector __name:=driver __log:=/home/applications/.ros/log/67942e5e-53ad-11ed-8007-d05099c29feb/kinect_head-driver-42.log].
log file: /home/applications/.ros/log/67942e5e-53ad-11ed-8007-d05099c29feb/kinect_head-driver-42*.log[0m
[kinect_head/kinect_head_nodelet_manager-41] restarting process
[1mprocess[kinect_head/kinect_head_nodelet_manager-41]: started with pid [8695][0m
[kinect_head/driver-42] restarting process
[1mprocess[kinect_head/driver-42]: started with pid [8700][0m
[0m[ INFO] [1666624076.239394310] [/kinect_head/driver:ros.nodelet]: Loading nodelet /kinect_head/driver of type openni_camera/driver to manager kinect_head_nodelet_manager with the following remappings:[0m
[0m[ INFO] [1666624076.239562983] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth -> /kinect_head/depth[0m
[0m[ INFO] [1666624076.239591521] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth_registered -> /kinect_head/depth_registered[0m
[0m[ INFO] [1666624076.239630395] [/kinect_head/driver:ros.nodelet]: /kinect_head/ir -> /kinect_head/ir[0m
[0m[ INFO] [1666624076.239652631] [/kinect_head/driver:ros.nodelet]: /kinect_head/projector -> /kinect_head/projector[0m
[0m[ INFO] [1666624076.239667364] [/kinect_head/driver:ros.nodelet]: /kinect_head/rgb -> /kinect_head/rgb[0m
[0m[ INFO] [1666624076.338971745] [/kinect_head/kinect_head_nodelet_manager:ros.nodelet]: Initializing nodelet with 8 worker threads.[0m
[31m[ERROR]  /kinect_head/depth_registered/image_raw/compressedDepth is not published[39m
 Checking /edgetpu_human_pose_estimator/output/image/compressed for 10 seconds
 Checking /edgetpu_object_detector/output/image/compressed for 10 seconds
[31m[FATAL] [1666624091.942327632] [/kinect_head/driver:ros.nodelet]: Failed to load nodelet '/kinect_head/driver` of type `openni_camera/driver` to manager `kinect_head_nodelet_manager'[0m
[31m[kinect_head/driver-42] process has died [pid 8700, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load openni_camera/driver kinect_head_nodelet_manager ir:=ir rgb:=rgb depth:=depth depth_registered:=depth_registered projector:=projector __name:=driver __log:=/home/applications/.ros/log/67942e5e-53ad-11ed-8007-d05099c29feb/kinect_head-driver-42.log].
log file: /home/applications/.ros/log/67942e5e-53ad-11ed-8007-d05099c29feb/kinect_head-driver-42*.log[0m
[kinect_head/driver-42] restarting process
[1mprocess[kinect_head/driver-42]: started with pid [9855][0m
[0m[ INFO] [1666624092.226443992] [/kinect_head/driver:ros.nodelet]: Loading nodelet /kinect_head/driver of type openni_camera/driver to manager kinect_head_nodelet_manager with the following remappings:[0m
[0m[ INFO] [1666624092.226557670] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth -> /kinect_head/depth[0m
[0m[ INFO] [1666624092.226578263] [/kinect_head/driver:ros.nodelet]: /kinect_head/depth_registered -> /kinect_head/depth_registered[0m
[0m[ INFO] [1666624092.226598007] [/kinect_head/driver:ros.nodelet]: /kinect_head/ir -> /kinect_head/ir[0m
[0m[ INFO] [1666624092.226611553] [/kinect_head/driver:ros.nodelet]: /kinect_head/projector -> /kinect_head/projector[0m
[... cut ...]
[33m[ WARN] [1666624092.629785222] [/kinect_head/kinect_head_nodelet_manager:ros.camera_info_manager]: Camera calibration file /home/applications/.ros/camera_info/depth_A00362913844045A.yaml not found.[0m
[33m[ WARN] [1666624092.629812383] [/kinect_head/kinect_head_nodelet_manager:ros.openni_camera./kinect_head/driver]: Using default parameters for IR camera calibration.[0m
 Checking /audio for 10 seconds
[31m[ERROR]  /audio is not published[39m
[INFO] [1666624124.306833] [/check_openni_node:rosout]: [CheckOpenNINode] sum of pixels is 23970715 at 1666624124
```

closed 2022-11-11 by knorth55