jsk-ros-pkg/jsk_robot #1706: mongodbの使い方

k-okada, 2022-10-25:
@knorth55 以下のlaunchファイルを立ち上げたら，とりあえずDBに何か貯まるのかな？と思ったんだけど，
```
[ERROR] [1666693339.502586]: Could not get message store services. Maybe the message store has not been started? Retrying..
[ERROR] [1666693339.502586]: Could not get message store services. Maybe the message store has not been started? Retrying..
```
というエラーがでる．
１）ローカルPCだけに保存する場合
２）リプリケーとする場合
のそれぞれのデフォルトの設定の方法を教えてください．
```
<launch>
  <!-- logging database -->
  <include file="$(find jsk_robot_startup)/lifelog/mongodb.launch">
    <arg name="use_daemon" value="true"/>
    <arg name="port" value="27017" />
    <arg name="repl_set_mode" value="false" />
  </include>

  <arg name="map_frame" default="odom" />
  <include file="$(find jsk_robot_startup)/lifelog/common_logger.launch">
    <arg name="save_rgb" value="false" />
    <arg name="save_depth" value="false" />
    <arg name="save_tf" value="true" />
    <arg name="save_joint_states" value="true" />
    <arg name="save_base_trajectory" value="false" />
    <arg name="save_object_detection" value="false" />
    <arg name="save_action" value="true" />
    <arg name="save_smach" value="true" />
    <arg name="save_app" value="true" />
    <arg name="enable_monitor" value="false" />
    <arg name="log_rate" value="1.0" />
    <arg name="launch_manager" value="true" />
    <arg name="map_frame_id" value="$(arg map_frame)" />
    <arg name="approximate_sync" value="true"/>
  </include>
</include>
```

k-okada, 2022-10-25:
thank you!
It sees mongodb is working
```
k-okada@p51s:~/catkin_ws/ws_hironx/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials$ cd
k-okada@p51s:~$ sudo systemctl status mongodb.service
[sudo] password for k-okada:
● mongodb.service - An object/document-oriented database
   Loaded: loaded (/lib/systemd/system/mongodb.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2022-10-25 19:16:32 JST; 2h 55min ago
     Docs: man:mongod(1)
 Main PID: 6030 (mongod)
    Tasks: 28 (limit: 4915)
   CGroup: /system.slice/mongodb.service
           └─6030 /usr/bin/mongod --unixSocketPrefix=/run/mongodb --config /etc/mongodb.conf

Oct 25 19:16:32 p51s systemd[1]: Started An object/document-oriented database.
```
but still fails with
```
  <!-- EDIT: rosparam for all robots -->
  <param name="robot/type" value="pepper"/>
  <param name="robot/name" value="basil"/>
  <param name="robot/database" value="jsk_robot_lifelog"/>

  <!-- logging database -->
  <include file="$(find jsk_robot_startup)/lifelog/mongodb.launch">
    <arg name="use_daemon" value="true"/>
    <arg name="port" value="27017" />
    <arg name="repl_set_mode" value="false" />
  </include>

  <arg name="map_frame" default="odom" />
  <include file="$(find jsk_robot_startup)/lifelog/common_logger.launch">
    <arg name="save_rgb" value="false" />
    <arg name="save_depth" value="false" />
    <arg name="save_tf" value="true" />
    <arg name="save_joint_states" value="true" />
    <arg name="save_base_trajectory" value="false" />
    <arg name="save_object_detection" value="false" />
    <arg name="save_action" value="true" />
    <arg name="save_smach" value="true" />
    <arg name="save_app" value="true" />
    <arg name="enable_monitor" value="false" />
    <arg name="log_rate" value="1.0" />
    <arg name="launch_manager" value="true" />
    <arg name="map_frame_id" value="$(arg map_frame)" />
    <arg name="approximate_sync" value="true"/>
  </include>
```

knorth55, 2022-10-25:
can you connect mongodb by `mongo` command?

```
mongo localhost:27017
```

if you can't, you may need to change `/etc/mongodb.conf`.
some PC needs to change `bind_ip` to `0.0.0.0` .
PR2 needs to change `bind_ip` to `0.0.0.0`.

```
bind_ip = 0.0.0.0
#bind_ip = 127.0.0.1
#port = 27017
```

k-okada, 2022-10-25:
seems working
```
$ mongo localhost:27017
MongoDB shell version v3.6.3
connecting to: mongodb://localhost:27017/test
MongoDB server version: 3.6.3
Welcome to the MongoDB shell.
For interactive help, type "help".
For more comprehensive documentation, see
        http://docs.mongodb.org/
Questions? Try the support group
        http://groups.google.com/group/mongodb-user
Server has startup warnings:
2022-10-25T19:16:32.228+0900 I STORAGE  [initandlisten]
2022-10-25T19:16:32.228+0900 I STORAGE  [initandlisten] ** WARNING: Using the XFS filesystem is strongly recommended with the WiredTiger storage engine
2022-10-25T19:16:32.228+0900 I STORAGE  [initandlisten] **          See http://dochub.mongodb.org/core/prodnotes-filesystem
2022-10-25T19:16:32.794+0900 I CONTROL  [initandlisten]
2022-10-25T19:16:32.794+0900 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
2022-10-25T19:16:32.794+0900 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
2022-10-25T19:16:32.794+0900 I CONTROL  [initandlisten]
>
```

k-okada, 2022-10-26:
```
$ apt show mongodb-server-core
Package: mongodb-server-core
Version: 1:3.6.3-0ubuntu1.4
Priority: optional
Section: universe/database
Source: mongodb
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian MongoDB Maintainers <pkg-mongodb-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 72.7 MB
Depends: libboost-filesystem1.65.1, libboost-program-options1.65.1, libboost-system1.65.1, libc6 (>= 2.15), libgcc1 (>= 1:3.4), libgoogle-perftools4, libpcrecpp0v5 (>= 7.7), libsnappy1v5, libssl1.1 (>= 1.1.0), libstdc++6 (>= 6), libstemmer0d (>= 0+svn527), libyaml-cpp0.5v5, zlib1g (>= 1:1.2.0)
Breaks: mongodb-server (<< 1:3.4.14-3ubuntu1)
Replaces: mongodb-server (<< 1:3.4.14-3ubuntu1)
Homepage: https://www.mongodb.org
Download-Size: 20.3 MB
APT-Manual-Installed: no
APT-Sources: http://jp.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages
Description: object/document-oriented database (server binaries package)
 MongoDB is a high-performance, open source, schema-free
 document-oriented data store that's easy to deploy, manage
 and use. It's network accessible, written in C++ and offers
 the following features:
 .
    * Collection oriented storage - easy storage of object-style data
    * Full index support, including on inner objects
    * Query profiling
    * Replication and fail-over support
    * Efficient storage of binary data including large objects (e.g. videos)
    * Auto-sharding for cloud-level scalability
 .
 High performance, scalability, and reasonable depth of
 functionality are the goals for the project.
 .
 This package contains the database server (mongod) and sharding
 server/load-balancer (mongos) binaries.

N: There is 1 additional record. Please use the '-a' switch to see it

$ apt search mongodb
Sorting... Done
Full Text Search... Done
collectd-core/bionic-updates 5.7.2-2ubuntu1.2 amd64
  statistics collection and monitoring daemon (core system)

golang-gopkg-mgo.v2-dev/bionic,bionic 2016.08.01-5 all
  Go library for MongoDB

jmeter-mongodb/bionic,bionic 2.13-3 all
  Load testing and performance measurement application (mongodb module)

kamailio-mongodb-modules/bionic 5.1.2-1ubuntu2 amd64
  mongodb modules for the Kamailio SIP server

libbson-1.0-0/bionic,now 1.9.2-1 amd64 [installed,automatic]
  Library to parse and generate BSON documents - runtime files

libbson-dev/bionic,now 1.9.2-1 amd64 [installed]
  Library to parse and generate BSON documents - dev files

libbson-doc/bionic,bionic 1.9.2-1 all
  Library to parse and generate BSON documents - documentation

libbson-perl/bionic,bionic 1.4.0-1 all
  BSON serialization and deserialization for Perl

libcatmandu-perl/bionic,bionic 1.0700-1 all
  metadata toolkit

[... 285 lines cut ...]
syslog-ng-mod-stardate/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (stardate plugin)

syslog-ng-mod-stomp/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (STOMP plugin)

syslog-ng-mod-tag-parser/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (tag parser plugin)

syslog-ng-mod-trigger/bionic 0.6.2-0.1 amd64
  Enhanced system logging daemon (trigger source)

syslog-ng-mod-xml-parser/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (xml parser plugin)

syslog-ng-mod-zmq/bionic 0.6.2-0.1 amd64
  Enhanced system logging daemon (ZeroMQ destination)

uwsgi-plugin-mongo/bionic 2.0.15+10+0.0.3 amd64
  MongoDB/GridFS plugins for uWSGI

w1retap-mongo/bionic 1.4.4-3 amd64
  Data logger for 1-Wire weather sensors (MongoDB plugin)

```

knorth55, 2022-10-26:
```
$ apt show mongodb-server-core
Package: mongodb-server-core
Version: 1:3.6.3-0ubuntu1.4
Priority: optional
Section: universe/database
Source: mongodb
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian MongoDB Maintainers <pkg-mongodb-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 72.7 MB
Depends: libboost-filesystem1.65.1, libboost-program-options1.65.1, libboost-system1.65.1, libc6 (>= 2.15), libgcc1 (>= 1:3.4), libgoogle-perftools4, libpcrecpp0v5 (>= 7.7), libsnappy1v5, libssl1.1 (>= 1.1.0), libstdc++6 (>= 6), libstemmer0d (>= 0+svn527), libyaml-cpp0.5v5, zlib1g (>= 1:1.2.0)
Breaks: mongodb-server (<< 1:3.4.14-3ubuntu1)
Replaces: mongodb-server (<< 1:3.4.14-3ubuntu1)
Homepage: https://www.mongodb.org
Download-Size: 20.3 MB
APT-Manual-Installed: no
APT-Sources: http://jp.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages
Description: object/document-oriented database (server binaries package)
 MongoDB is a high-performance, open source, schema-free
 document-oriented data store that's easy to deploy, manage
 and use. It's network accessible, written in C++ and offers
 the following features:
 .
    * Collection oriented storage - easy storage of object-style data
    * Full index support, including on inner objects
    * Query profiling
    * Replication and fail-over support
    * Efficient storage of binary data including large objects (e.g. videos)
    * Auto-sharding for cloud-level scalability
 .
 High performance, scalability, and reasonable depth of
 functionality are the goals for the project.
 .
 This package contains the database server (mongod) and sharding
 server/load-balancer (mongos) binaries.

N: 追加レコードが 1 件あります。表示するには '-a' スイッチを付けてください。
```

```
$ apt search mongodb
ソート中... 完了
全文検索... 完了  
collectd-core/bionic-updates 5.7.2-2ubuntu1.2 amd64
  statistics collection and monitoring daemon (core system)

golang-gopkg-mgo.v2-dev/bionic,bionic 2016.08.01-5 all
  Go library for MongoDB

jmeter-mongodb/bionic,bionic 2.13-3 all
  Load testing and performance measurement application (mongodb module)

kamailio-mongodb-modules/bionic 5.1.2-1ubuntu2 amd64
  mongodb modules for the Kamailio SIP server

libbson-1.0-0/bionic 1.9.2-1 amd64
  Library to parse and generate BSON documents - runtime files

libbson-dev/bionic 1.9.2-1 amd64
  Library to parse and generate BSON documents - dev files

libbson-doc/bionic,bionic 1.9.2-1 all
  Library to parse and generate BSON documents - documentation

libbson-perl/bionic,bionic 1.4.0-1 all
  BSON serialization and deserialization for Perl

libcatmandu-perl/bionic,bionic 1.0700-1 all
[... 289 lines cut ...]

syslog-ng-mod-stardate/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (stardate plugin)

syslog-ng-mod-stomp/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (STOMP plugin)

syslog-ng-mod-tag-parser/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (tag parser plugin)

syslog-ng-mod-trigger/bionic 0.6.2-0.1 amd64
  Enhanced system logging daemon (trigger source)

syslog-ng-mod-xml-parser/bionic 3.13.2-3 amd64
  Enhanced system logging daemon (xml parser plugin)

syslog-ng-mod-zmq/bionic 0.6.2-0.1 amd64
  Enhanced system logging daemon (ZeroMQ destination)

uwsgi-plugin-mongo/bionic 2.0.15+10+0.0.3 amd64
  MongoDB/GridFS plugins for uWSGI

w1retap-mongo/bionic 1.4.4-3 amd64
  Data logger for 1-Wire weather sensors (MongoDB plugin)
```

k-okada, 2022-10-26:
I see, 
even if we can connect by mongo client ` mongo localhost:27017`, we need to change `bind_ip` 
```
$ cat /etc/mongodb.conf | grep -v ^#
dbpath=/var/lib/mongodb
logpath=/var/log/mongodb/mongodb.log
logappend=true
bind_ip = 0.0.0.0
journal=true
```

knorth55, 2022-10-26:
my laptop doesn't need to change `bind_ip`, but `pr1040` needs to change `bind_ip`.
the `bind_ip` is related to `firewall` settings, so it depends on user's environment.