jsk-ros-pkg/jsk_robot #1713: [Fetch15] コントローラーが繋がらない

sktometometo, 2022-10-31:


sktometometo, 2022-10-31:
CC: @knorth55 @tkmtnt7000

closed 2022-10-31 by sktometometo

tkmtnt7000, 2022-11-02:
一応aptでなにかやっているっぽいログを貼っておきます．
```
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ history | grep apt
  631  2022/11/02 15:05:54 sudo apt install wireshark
  693  2022/11/02 15:05:54 apt search jsk-pcl
  694  2022/11/02 15:05:54 sudo apt install ros-melodic-jsk-pcl-ros
  695  2022/11/02 15:05:54 sudo apt update
  696  2022/11/02 15:05:54 sudo apt install ros-melodic-jsk-pcl-ros
 1865  2022/11/02 15:05:54 sudo apt purge at
 1915  2022/01/19 15:38:18 cd /var/log/apt
 1961  2022/01/19 15:28:25 cd /var/log/apt/
 2340  2022/04/08 19:28:31 apt list --installed |grep realsense
 2344  2022/04/11 10:05:33 apt search realsense
 2361  2022/04/11 10:07:15 apt --help
 2362  2022/04/11 10:07:26 apt search realsense
 2363  2022/04/11 10:07:38 apt show librealsense2
 2409  2022/11/02 15:05:54 fetch@fetch15:~/ros/melodic$   Intel(R) RealSense(tm) Camera Capture API - Network Bridge extension development files
 2414  2022/04/11 10:08:21 fetch@fetch15:~/ros/melodic$   Intel(R) RealSense(tm) Camera Capture API - udev rules
 2419  2022/11/02 15:05:54 fetch@fetch15:~/ros/melodic$   Intel(R) RealSense(tm) Camera Capture API - utils and demos
 2424  2022/11/02 15:05:54 fetch@fetch15:~/ros/melodic$   Library for capturing data from the Intel(R) RealSense(TM) SR300, D400 Depth cameras and T2xx Tracking devices.
 2431  2022/04/11 10:08:22 Try: sudo apt install <deb name>
 2437  2022/04/11 10:35:18 sudo apt-get install librealsense2-dev=2.45.0-0~realsense0.4551
 2438  2022/04/11 10:36:29 apt search librealsense2
 2439  2022/04/11 10:37:27 sudo apt install librealsense2-udev-rules=2.45.0-0~realsense0.4551
 2440  2022/04/11 10:37:58 sudo apt install librealsense2-gl=2.45.0-0~realsense0.4551
 2441  2022/04/11 10:38:16 sudo apt install librealsense2-gl-dbg=2.45.0-0~realsense0.4551
 2442  2022/04/11 10:38:27 sudo apt install librealsense2-gl-dev=2.45.0-0~realsense0.4551
 2443  2022/04/11 10:38:59 sudo apt install librealsense2-utils=2.45.0-0~realsense0.4551
 2444  2022/04/11 10:39:13 sudo apt install librealsense2-net=2.45.0-0~realsense0.4551
 2445  2022/04/11 10:39:28 sudo apt install librealsense2-net-dev=2.45.0-0~realsense0.4551
 2446  2022/04/11 10:39:38 sudo apt install librealsense2-net-dbg=2.45.0-0~realsense0.4551
 2447  2022/04/11 10:39:46 sudo apt install librealsense2-utils=2.45.0-0~realsense0.4551
 2448  2022/04/11 10:40:17 apt search librealsense
 2449  2022/04/11 10:40:37 sudo apt install librealsense2-dbg=2.45.0-0~realsense0.4551
 2450  2022/04/11 10:41:31 apt search librealsense
 2484  2022/04/11 10:28:02 sudo apt install ros-melodic-librealsense2
 2485  2022/04/11 10:28:12 sudo apt update
 2486  2022/04/11 10:28:59 sudo apt install ros-melodic-librealsense2=2.3.0
 2489  2022/04/11 10:30:58 ls | grep apt
 2490  2022/04/11 10:31:00 cd apt
 2493  2022/04/11 10:31:22 sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
 2494  2022/04/11 10:31:33 sudo apt udpate
 2495  2022/04/11 10:31:36 sudo apt update
 2496  2022/04/11 10:32:03 sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
 2497  2022/04/11 10:32:15 sudo apt update
 2498  2022/04/11 10:32:46 apt search librealsense2
 2499  2022/04/11 10:33:06 apt-cache policy librealsense2
 2500  2022/04/11 10:33:49 sudo apt install librealsense2=2.45.0-0~realsense0.4551
 2502  2022/04/11 10:20:45 apt search realsense2
 2503  2022/04/11 10:21:46 sudo apt install ros-melodic-realsense2-camera=2.3.0
 2504  2022/04/11 10:21:55 sudo apt install ros-melodic-realsense2-camera=2.3.0-1
 2505  2022/04/11 10:22:24 sudo apt install ros-melodic-realsense2-camera=2.3.0-1bionic.20211115.141014
 2506  2022/04/11 10:23:22 apt cache showpkg ros-melodic-realsense2-camera
 2507  2022/04/11 10:23:30 apt-cache showpkg ros-melodic-realsense2-camera
 2508  2022/04/11 10:24:01 apt-cache showpkg ros-melodic-librealsense2
 2509  2022/04/11 10:25:21 apt policy ros-melodic-librealsense2
 2514  2022/04/11 10:15:46 emacs -nw realsense_apt.txt
 2515  2022/04/11 10:34:05 less realsense_apt.txt 
 2517  2022/04/11 10:14:39 apt search realsense
 2518  2022/04/11 10:16:02 sudo apt purge librealsense2
 2519  2022/04/11 10:27:02 apt-cache policy ros-melodic-librealsense2
 2520  2022/04/11 11:13:25 apt search realsense2
 2521  2022/04/11 11:13:42 apt list | grep realsense
 2595  2022/04/11 12:28:41 apt search librealsense
 2641  2022/04/12 19:42:36 sudo apt install libbluetooth-dev libsbc libfdk-aac-dev 
 2643  2022/04/12 19:43:27 sudo apt install libbluetooth-dev libsbc libfdk-aac-dev 
 2644  2022/04/12 19:43:32 sudo apt install libbluetooth-dev libsbc-dev libfdk-aac-dev 
 2766  2022/04/12 22:58:13 roscd audio_capture/
 2770  2022/04/12 22:58:23 cp capture_wave.launch /tmp/
 2771  2022/04/12 22:58:26 emacs -nw /tmp/capture_wave.launch 
[... 44 lines cut ...]
10664  2022/08/24 15:18:21 apt search jsk-rosbag
10681  2022/08/25 13:49:30 apt search jsk-rosbag-tools
11297  2022/09/15 13:03:00 sudo apt purge atd
11298  2022/09/15 13:03:06 sudo apt install atd
11301  2022/09/15 13:04:15 apt-file find /usr/sbin/atd
11302  2022/09/15 13:04:19 sudo apt install apt-file
11303  2022/09/15 13:04:38 apt-file find /usr/sbin/atd
11304  2022/09/15 13:04:43 sudo apt update
11310  2022/09/15 13:43:37 apt-cache depends librealsense-udev-rules
11311  2022/09/15 13:43:43 apt search librealsense
[... cut ...]
13995  2022/10/25 00:13:19 sudo apt update
13997  2022/10/25 00:14:01 sudo apt install ros-melodic-rosbridge-suite ros-melodic-slam-mapping ros-melodic-robot-calibration
13998  2022/10/25 00:14:06 sudo apt install ros-melodic-rosbridge-suite ros-melodic-gslam-mapping ros-melodic-robot-calibration
13999  2022/10/25 00:14:18 sudo apt install ros-melodic-rosbridge-suite ros-melodic-slam-gmapping ros-melodic-robot-calibration
14143  2022/11/02 15:06:06 history | grep apt
[http://fetch15:11311][133.11.216.217] fetch@fetch15:~$ 

```