jsk-ros-pkg/jsk_robot #1716: ImportError: dynamic module does not define init function (init_tf2)

a-ichikura, 2022-11-09:
Spotの中で[aws_detect_faces](https://jsk-docs.readthedocs.io/projects/jsk_recognition/en/latest/jsk_perception/nodes/aws_detect_faces.html?highlight=detect%20face)を試そうとして、
内部でjsk_recognitionをbuildし、launchファイルを立ち上げたら`dynamic module does not define init function (init_tf2)`というImportErrorが出ました。
```
$ roslaunch jsk_spot_startup detect_faces.launch 
... logging to /home/a-ichikura/.ros/log/fb8dd2e2-5e6e-11ed-912e-a8a1599ff350/roslaunch-SpotCORE-8386.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://SpotCORE:45539/

SUMMARY
========

CLEAR PARAMETERS
 * /aws_detect_faces/

PARAMETERS
 * /aws_detect_faces/always_subscribe: True
 * /aws_detect_faces/attributes: ALL
 * /aws_detect_faces/aws_credentials_path: /home/a-ichikura/...
 * /aws_detect_faces/image: /spot/camera/hand...
 * /aws_detect_faces/use_window: True
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    aws_detect_faces (jsk_spot_startup/aws_detect_faces.py)

ROS_MASTER_URI=http://localhost:11311

process[aws_detect_faces-1]: started with pid [8531]
Traceback (most recent call last):
  File "/home/a-ichikura/spot_ws/src/jsk_robot/jsk_spot_robot/jsk_spot_startup/node_scripts/aws_detect_faces.py", line 23, in <module>
    from tf.transformations import quaternion_from_euler
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/home/spot/spot_driver_ws/devel/lib/python3/dist-packages/tf2_ros/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 38, in <module>
  File "/home/spot/spot_driver_ws/devel/lib/python3/dist-packages/tf2_py/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 38, in <module>
ImportError: dynamic module does not define init function (init_tf2)
[aws_detect_faces-1] process has died [pid 8531, exit code 1, cmd /home/a-ichikura/spot_ws/src/jsk_robot/jsk_spot_robot/jsk_spot_startup/node_scripts/aws_detect_faces.py __name:=aws_detect_faces __log:=/home/a-ichikura/.ros/log/fb8dd2e2-5e6e-11ed-912e-a8a1599ff350/aws_detect_faces-1.log].
log file: /home/a-ichikura/.ros/log/fb8dd2e2-5e6e-11ed-912e-a8a1599ff350/aws_detect_faces-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

a-ichikura, 2022-11-09:
解決策
a-ichikuraの`spot_ws`以下に` git clone https://github.com/jsk-ros-pkg/geometry2_python3.git` して再build

mqcmd196, 2022-11-09:
素晴らしい。ちなみにその後は問題なくコード動きましたか？

a-ichikura, 2022-11-09:
`aws_detect_face.py`がpython2のコードでspot_wsはpython3のワークスペースだったので、
`aws_detect_faces.py`で`#!/usr/bin/env python3`を明示しました。

k-okada, 2022-11-10:
メモ：`#!/usr/bin/env python3` ですが，workspaceに以下のソースを落としてきて，
```
rm /home/k-okada/catkin_ws/devel/.private/catkin/cmake.lock /home/k-okada/catkin_ws/devel/./cmake.lock
```
をして，jsk_recognition はhttps://github.com/jsk-ros-pkg/jsk_recognition/pull/2743
をつかって，もう一回 コンパイルすると，`rosrun jsk_perception aws_detect_faces.py` すると，python3で動くようになるはず．．．
pythonコードの` /usr/bin/env python `を無限に変更し続ける感じになってきたら思い出してください．

```
# to avtivate python_catkin_install, we need newer catkin
- git:
    local-name: catkin
    uri:     https://github.com/ros/catkin.git
    version: 0.8.10
```